fwupd/fwupd #4014: `fwupd-self-test` failed if no `fwupd` has been previously installed

piyueh, 2021-11-22:
**Describe the bug**

When running the tests, `fwupd-self-test` failed if no `fwupd` has been previously installed, i.e., it failed due to not being able to find running `fwupd` service. I believe this is the same as #64. The error message is:

```
Fwupd:ERROR:../fwupd-git/libfwupd/fwupd-self-test.c:597:fwupd_client_remotes_func: assertion failed (error == NULL): The name org.freedesktop.fwupd was not provided by any .service files (FwupdError, 10)
Bail out! Fwupd:ERROR:../fwupd-git/libfwupd/fwupd-self-test.c:597:fwupd_client_remotes_func: assertion failed (error == NULL): The name org.freedesktop.fwupd was not provided by any .service files (FwupdError, 10)
```

I tried to add the following code snippet to before [line 543](https://github.com/fwupd/fwupd/blob/c92f352393f534b34d0632290dcf65df8ac7bfa8/libfwupd/fwupd-self-test.c#L543) and [line 597](https://github.com/fwupd/fwupd/blob/c92f352393f534b34d0632290dcf65df8ac7bfa8/libfwupd/fwupd-self-test.c#L597), and the error was gone.
```c
if (array == NULL && g_error_matches(error, FWUPD_ERROR, FWUPD_ERROR_NOT_SUPPORTED)) {
    g_test_skip("no available fwupd daemon");
    return;
}
```
But I'm not sure my workaround was correct or not.

**Steps to Reproduce**

- step 1: configuration
  ```
  meson setup ${source_dir} ${build_dir} \
      --prefix /usr \
      --libexecdir lib \
      --sbindir bin \
      --buildtype plain \
      --auto-features enabled \
      --wrap-mode nodownload \
      -D b_pie=true \
      -D plugin_flashrom=true \
      -D plugin_modem_manager=true \
      -D plugin_intel_spi=true \
      -D b_lto=false \
      -D lzma=true \
      -D docs=gtkdoc \
      -D supported_build=true \
      -D efi_binary=false
  ```

- step 2: build
  ```
  meson compile -C ${build_dir} -j 0
  ```

- step 3: test
  ```
  CACHE_DIRECTORY=/tmp meson test -C ${build_dir}
  ```

**Expected behavior**

All tests pass.

**fwupd version information**

Git commit https://github.com/fwupd/fwupd/commit/c92f352393f534b34d0632290dcf65df8ac7bfa8

**fwupd device information**

See [devices.txt](https://github.com/fwupd/fwupd/files/7578797/devices.txt).

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes.
- Is this a regression? Don't know.

hughsie, 2021-11-22:
@piyueh hey, sorry about that. Can you try again with the wip/hughsie/no-daemon-self-tests branch please?

piyueh, 2021-11-22:
Hi @hughsie , I still got the same error for `fwupd-self-test` with `wip/hughsie/no-daemon-self-tests` branch.

hughsie, 2021-11-23:
@piyueh I've force pushed a new more complete fix. Can you try that please?

closed 2021-11-23 by hughsie

piyueh, 2021-11-23:
Yes, it works! Thanks!