fwupd/fwupd #4022: Fwupd accept several components in metainfo XML

d4s, 2021-11-23:
**Describe the bug**

The CAB file containing several firmware components parsed well by `fwupdtool` and components could be used for local upgrade.

**Steps to Reproduce**
1. Prepare CAB file with the single `firmware.metainfo.xml` describing 2 separate firmwares with different ID's:
```
<?xml version='1.0' encoding='utf-8'?>
<component type="firmware">
  <id>com.example.0.firmware</id>
...
</component>
<component type="firmware">
  <id>com.example.1.firmware</id>
...
</component>
```
2. Use prepared CAB file for upgrade:
```
fwupdtool install example.cab
```

**Expected behavior**

CAB file shouldn't be used for upgrade.

**fwupd version information**

```shell
fwupdmgr --version
daemon version: 1.7.2-13-g01071a6c
```

**Additional**
LVFS do not accept that format, throwing error if I try to upload the CAB:
```
The metadata file could not be parsed as valid XML on line 42: Extra content at the end of the document 
```

superm1, 2021-11-23:
This should be opened here:
https://gitlab.com/fwupd/lvfs-website

Also if you can post the cab or attach it to the bug there it will be helpful for fixing it.

d4s, 2021-11-23:
@superm1 the issue is not related to LVFS. Works only for locally prepared CAB and fwupd

d4s, 2021-11-23:
Argh! Gitlab! let me to create there

superm1, 2021-11-23:
> @superm1 the issue is not related to LVFS. Works only for locally prepared CAB and fwupd

Right so it fails to parse on LVFS side from your description but works with local tools.

d4s, 2021-11-23:
> > @superm1 the issue is not related to LVFS. Works only for locally prepared CAB and fwupd
> 
> Right so it fails to parse on LVFS side from your description but works with local tools.

correct, after the discussion with @hughsie -- it shouldn't happened for local tools

hughsie, 2021-11-23:
I think that fwupd probably ought to fail in the same way.

hughsie, 2021-11-24:
A note for my future self: `fu_cabinet_build_silo_file()`

hughsie, 2021-11-24:
Future self said thanks: https://github.com/hughsie/libxmlb/pull/99

closed 2021-11-24 by hughsie