fwupd/fwupd #4028: Failed update to System76 Launch

ahoneybun, 2021-11-25:
**Describe the bug**
A clear and concise description of what the bug is.

Not able to update the firmware on the System76 Launch

**Steps to Reproduce**
Steps to reproduce the behavior.

1. Connect Launch
2. Run 'fwupd update' or 'fwupd upgrade'
3. Make sure the device is unlocked with Fn+ESC

Errors:

```shell
fwupdmgr update
Devices with no available firmware updates:
 • SSD 980 PRO 2TB
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • Unifying Receiver
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Launch Configurable Keyboard (launch 1) from 0.7.103-4007-ga49720    ║
║ to 0.7.103-4011-g708ed5?                                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Add system76_ec command to disable input events. This mainly allows for the  ║
║ System76 production line to perform quality assurance tests faster than the  ║
║ previous method of setting and resetting the entire keymap.                  ║
║                                                                              ║
║ Launch Configurable Keyboard (launch 1) and all connected devices may not    ║
║ be usable while updating.                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Writing…                 [                                       ]not supported
[sbarcomb@perseus ~]$ 
```

**Expected behavior**
A clear and concise description of what you expected to happen.

Firmware is updated to the latest release

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
1.7.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
A320M-S2H
│
├─AMD Ryzen 7 2700X Eight-Core Processor         :
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Vendor:             AMD
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         997104da-ec11-5e04-9d94-ac5501fbf609
│                         c0a70727-4a05-5a83-9473-d3e164a2873d
│                         bd304824-d1b3-52b5-982b-a995d4ed0e6c
│     Device Flags:       • Internal device
│   
├─CT250MX500SSD4:
│     Device ID:          185b59feeff9d1bfb0713198bf1a495f5db119ab
│     Summary:            ATA Drive
│     Current version:    M3CR022
│     Vendor:             Micron (ATA:0x1344, OUI:00a075)
│     GUIDs:              1ee73633-a3d5-5d8c-ba46-6341f96945ef
│                         6b963677-f2db-5800-95bc-26bb07dedd68
│                         acf130fc-ff0e-5250-920d-61964039f9e4
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Ellesmere [Radeon RX 470/480/570/570X/580/580X/590] (Radeon RX 570 Pulse 4GB):
│     Device ID:          bcd9d972af6e51b856983257c010bc28d37914ea
│     Current version:    e7
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              67572e30-d9de-5f7a-815f-4496ab9cc731
│                         8af086a1-724f-5866-83b8-70cd0426a63a
│                         68958f93-be94-5a98-a06d-a4da21a38135
│                         55746bd7-4196-594a-9d39-0808222765d1
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Launch Configurable Keyboard (launch 1):
│     Device ID:          83521e02a973e5f6a55b12eb4a95c73c309acbcb
│     Current version:    0.7.103-4007-ga49720
│     Vendor:             System76 (USB:0x3384)
│     GUIDs:              b2cfb86e-dacd-5663-83e6-58059831ab6d
│                         7549de75-189b-5fa6-911a-c9f5d127dc28
│                         af6272df-30f2-5067-ae05-201e69cd08a5
│     Device Flags:       • Updatable
│                         • Supported on remote server
│   
├─Mass storage controller:
│     Device ID:          7d29f2075dcafb4488b40c73f199cf46bb76bddb
│     Current version:    02
│     Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              8757fa0e-b2e3-51fe-b3a0-66fecfbc7b5e
│                         20bd68d1-c2db-580c-a0cc-d623f2f48939
│                         00be5727-5627-5f10-bf1b-3f1dcbf085a5
│                         303b5758-c2d9-569c-8e42-3332807046d1
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Samsung SSD 860 EVO 500GB:
│     Device ID:          55220ca6004d0cfc5f5067ff3b4e2b4923bf355c
│     Summary:            ATA Drive
│     Current version:    RVT02B6Q
│     Vendor:             Samsung (ATA:0x144D, OUI:002538)
│     GUIDs:              7cb4c12d-a507-548c-ab33-33ca06126875
│                         3831476a-add3-5bc8-88ff-31739e99ad07
│                         bd25ba7e-8dd0-5aed-90dd-db6a84712c8f
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─WDC WD20EZRZ-00Z5HB0:
      Device ID:          9afcd7db8643d511ee389321038ae4c56c7a45f3
      Summary:            ATA Drive
      Current version:    80.00A80
      Vendor:             Western Digital (ATA:0x101C, OUI:0014ee)
      GUIDs:              25536fd8-9a31-5a6e-9036-5455ed6b66fa
                          181ebd30-a13a-50f2-960f-0f2569cc360d
                          3ef82fd4-98a0-5009-b817-2fc40343b986
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```

**Additional questions**

- Operating system and version: Fedora 35
- Have you tried rebooting? Yes
- Is this a regression?

ahoneybun, 2021-11-25:
Sorry if I created this on the wrong example but I updated it to match my issue.

ahoneybun, 2021-11-25:
I am able to update the firmware on Pop!_OS 21.10 with fwupd version 1.5.11.

```shell
fwupdmgr update
Devices with no available firmware updates: 
 • CT250MX500SSD4
 • Samsung SSD 860 EVO 500GB
 • WDC WD20EZRZ-00Z5HB0
Upgrade available for Launch Configurable Keyboard (launch 1) from 0.7.103-4007-ga49720 to 0.7.103-4011-g708ed5
Launch Configurable Keyboard (launch 1) and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Updating Launch Configurable Keyboard (launch 1)…****************]
Erasing…                 [***************************************]
Writing…                 [***************************************]
Verifying…               [***************************************]
Restarting device…       [***************************************]
Successfully installed firmware
```

superm1, 2021-11-25:
Since you already upgraded, can you still reproduce it on 1.7.2?  If so, would you please re-launch the daemon in verbose mode (either add --verbose to the systemd unit or launch it manually) and reproduce again with those logs enabled. 

@jackpot51 can you take a look?

ahoneybun, 2021-11-25:
I downgraded to an older version of the firmware to recreate the issue so I did recreate it on version 1.7.2 which is in Fedora 35. I'll grab the logs in a min.

ahoneybun, 2021-11-25:
Here are some logs.

[fwupd-1.7.2-debug.txt](https://github.com/fwupd/fwupd/files/7599444/fwupd-1.7.2-debug.txt)

ahoneybun, 2021-11-25:
I'm also attaching what it looks like when it downgrades the device. 

[fwupd-1.5.11-downgrade-debug.txt](https://github.com/fwupd/fwupd/files/7599672/fwupd-1.5.11-downgrade-debug.txt)

hughsie, 2021-11-25:
@jackpot51 is that offer of a launch keyboard PCB still available? then we can catch regressions like this before we do tarball releases with a device-test: https://github.com/fwupd/fwupd/tree/main/data/device-tests

hughsie, 2021-11-25:
@ahoneybun does https://github.com/fwupd/fwupd/pull/4030 fix things for you? 100% untested.

jackpot51, 2021-11-25:
Yes, we could send a launch. It requires a manual step to flash, will that be okay?

hughsie, 2021-11-25:
Yes, that's fine, thanks.

jackpot51, 2021-11-25:
It is a holiday in the US, I can look into this next Monday

hughsie, 2021-11-25:
Dude, no panic at all, have a great thanksgiving!

ahoneybun, 2021-11-25:
> @ahoneybun does https://github.com/fwupd/fwupd/pull/4030 fix things for you? 100% untested.

I'll check it in a few!

closed 2021-11-27 by hughsie

ahoneybun, 2022-05-25:
Not sure about using the flatpak, does the other packaging work?

superm1, 2022-05-26:
> Im noticing after updating to Ubuntu 22.04 LTS, im noticing an update as well that is failing. Pretty much just as described above.
> 
> fwupdmgr --version runtime org.freedesktop.fwupd 1.7.5 runtime com.dell.libsmbios 2.4 compile org.freedesktop.gusb 0.3.10 runtime org.kernel 5.15.0-33-generic compile org.freedesktop.fwupd 1.7.5 runtime org.freedesktop.gusb 0.3.10

Can you please open a new bug and supply everything requested in the template and refer back to this bug?  It's supposed to be fixed in your build, but there might be a different regression.