fwupd/fwupd #4029: Flatpak version 1.7.2: it complains of root and of non-root and gives other errors

LinuxOnTheDesktop, 2021-11-25:
**Describe the question**
This version of fwupd says it needs root but also complains when I run it with root. Moreover, whether or not I run it as root, other messages - which are hard to understand - appear. Am I doing something wrong? For more about the problems, please see the 'Further information' section, below.

**fwupd version information**
```
# flatpak run org.freedesktop.fwupd --version

Note that the directory 

'/root/.local/share/flatpak/exports/share'

is not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

client version:	1.7.2
compile-time dependency versions
	gusb:	0.3.8
```

**Further information**

```
$ flatpak run org.freedesktop.fwupd refresh
This program may only work correctly as root
Loading…                 [-                                      ]03:09:47:0289 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: offset bigger than size 9
03:09:47:0298 FuPluginTpm          failed to load eventlog: Failed to open file “/sys/kernel/security/tpm0/binary_bios_measurements”: Permission denied
03:09:47:0341 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
03:09:47:0341 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
03:09:47:0341 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
03:09:47:0341 FuPluginUefiCapsule  skipping device that failed coldplug: failed to read fw_class
03:09:47:0346 FuEngine             failed to probe device usb:02:00:03: USB error on device 0bda:0328 : Entity not found [-5]
03:09:47:0389 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:16.0: failed to add device using on pci_mei: could not read HFS1: failed to read from port 0x0040: No such file or directory
03:09:47:0445 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1f.0: failed to add device using on pci_bcr: could not read BCR: failed to read from port 0x00dc: No such file or directory
03:09:47:0478 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/drm_dp_aux1: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux1: Permission denied
03:09:47:0479 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/drm_dp_aux2: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux2: Permission denied
03:09:47:0480 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-eDP-1/drm_dp_aux0: failed to add device using on synaptics_mst: failed to open /dev/drm_dp_aux0: Permission denied
03:09:47:0498 FuEngine             failed to add device /sys/devices/virtual/msr/msr0: failed to add device using on msr: failed to open /dev/cpu/0/msr: Permission denied
03:09:47:0503 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1c.4/0000:04:00.0/nvme/nvme0: failed to open /dev/nvme0: Permission denied
03:09:47:0506 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:046D:C52B.0003/hidraw/hidraw0: failed to open /dev/hidraw0
ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a) 
03:09:47:0517 FuEngine             failed to add device /sys/devices/platform/STM7112:00/tpm/tpm0: failed to initialize TPM library
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
$ sudo flatpak run org.freedesktop.fwupd refresh
[sudo] password for [redacted]: 
error: "flatpak run" is not intended to be ran with sudo
$ su
Password: 
# flatpak run org.freedesktop.fwupd refresh

Note that the directory 

'/root/.local/share/flatpak/exports/share'

is not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

Loading…                 [-                                      ]03:10:38:0885 FuEngine             failed to probe device usb:02:00:03: USB error on device 0bda:0328 : Entity not found [-5]
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
```

LinuxOnTheDesktop, 2021-11-25:
Oh, and also:

```
# flatpak run org.freedesktop.fwupd update
[. . .]
An update requires a reboot to complete. Restart now? [y|N]: y
No supported backend compiled in to perform the operation.
```

LinuxOnTheDesktop, 2021-11-25:
I find that the 'device' errors disappear if I unplug my Logitech Unified Receiver. (Cf. #155.) This seems a regression: I did not have those problem on fwupd 1.5.11-0ubuntu1~20.04.2, which is what apt gives me.

The strange stuff about root persists. So does a message saying 'FuPluginUefiCapsule missing or invalid embedded capsule header'. I take it that the latter problem is a Lenovo problem. (For, the update in question, for 'TPM', owes to Lenovo and, well, look.)

I suspect that I should open a proper but report for some part(s) of this - please advise.

hughsie, 2021-11-25:
You need `-i` -- can you try `sudo -i flatpak run org.freedesktop.fwupd --verbose get-devices`

LinuxOnTheDesktop, 2021-11-26:
Thanks. Ah. Can you get the flatpak to tell the user about the `-i`, when the user tries other sudo variants? Otherwise, the user is liable to flail around as I did.

```
$ sudo -i flatpak run org.freedesktop.fwupd --verbose get-devices
00:18:00:0356 FuDebug              Verbose debugging enabled (on console 1)
00:18:00:0356 FuCommon             creating path /run/lock
00:18:00:0356 FuMain               locked /run/lock/fwupdtool
00:18:00:0356 FuConfig             loading config values from /app/etc/fwupd/daemon.conf
00:18:00:0356 FuConfig             using autodetected max archive size 18.4 EB
00:18:00:0357 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
00:18:00:0357 FuRemoteList         loading /app/share/fwupd/metainfo
00:18:00:0357 XbSilo               attempting to load /var/data/cache/fwupd/metainfo.xmlb
00:18:00:0357 XbSilo               file: b7cd88e1-4626-712a-daa5-bfa49dc7b995, current:b7cd88e1-4626-712a-daa5-bfa49dc7b995, cached: (null)
00:18:00:0357 XbSilo               loading silo with file contents
00:18:00:0357 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/lvfs-testing.conf
00:18:00:0358 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/lvfs.conf
00:18:00:0358 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/vendor-directory.conf
00:18:00:0358 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/vendor.conf
00:18:00:0358 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/dell-esrt.conf
00:18:00:0358 FuRemoteList         skipping invalid file /var/data/lib/fwupd/remotes.d/lvfs
00:18:00:0358 FuRemoteList         ordering lvfs-testing=lvfs+1
00:18:00:0358 FuRemoteList         ignoring unfound remote fwupd
00:18:00:0358 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
00:18:00:0358 Jcat                 trying to load certificate from /app/etc/pki/fwupd/LVFS-CA.pem
00:18:00:0358 Jcat                 reading /app/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
00:18:00:0358 Jcat                 loaded 1 certificates
00:18:00:0358 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
00:18:00:0358 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
00:18:00:0358 Jcat                 trying to load certificate from /app/etc/pki/fwupd-metadata/LVFS-CA.pem
00:18:00:0358 Jcat                 reading /app/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
00:18:00:0358 Jcat                 loaded 1 certificates
00:18:00:0358 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
00:18:00:0358 Jcat                 reading /var/data/lib/fwupd/pki/secret.key with 2459 bytes
00:18:00:0358 Jcat                 reading /var/data/lib/fwupd/pki/client.pem with 1383 bytes
00:18:00:0365 FuEngine             client certificate exists and working
00:18:00:0365 FuHistory            trying to open database '/var/data/lib/fwupd/pending.db'
00:18:00:0365 FuHistory            got schema version of 7
00:18:00:0365 FuSmbios             trying to read /sys/class/dmi/id
00:18:00:0365 FuHwids              loading HwId overrides from /app/etc/fwupd/daemon.conf
00:18:00:0366 FuHwids              not loading HwId overrides from /var/data/lib/fwupd/daemon.conf
00:18:00:0366 FuHwids              smbios property Manufacturer=LENOVO
00:18:00:0366 FuHwids              smbios property EnclosureKind=a
00:18:00:0366 FuHwids              smbios property Family=ThinkPad X1 Carbon 6th
00:18:00:0366 FuHwids              smbios property ProductName=20KH001UKP
00:18:00:0366 FuHwids              smbios property ProductSku=LENOVO_MT_20KH_BU_Think_FM_ThinkPad X1 Carbon 6th
00:18:00:0366 FuHwids              smbios property BiosVendor=LENOVO
00:18:00:0366 FuHwids              smbios property BiosVersion=N23ET79W (1.54 )
00:18:00:0366 FuHwids              smbios property BiosMajorRelease=01
00:18:00:0366 FuHwids              smbios property BiosMinorRelease=36
00:18:00:0366 FuHwids              smbios property FirmwareMajorRelease=01
00:18:00:0366 FuHwids              smbios property FirmwareMinorRelease=16
00:18:00:0366 FuHwids              smbios property BaseboardManufacturer=LENOVO
00:18:00:0366 FuHwids              smbios property BaseboardProduct=20KH001UKP
00:18:00:0369 XbSilo               attempting to load /var/data/cache/fwupd/quirks.xmlb
00:18:00:0370 XbSilo               file: 3224a026-bd16-2832-4790-b12d56cf648f, current:3224a026-bd16-2832-4790-b12d56cf648f, cached: (null)
00:18:00:0370 XbSilo               loading silo with file contents
00:18:00:0374 XbSilo               attempting to load /var/data/cache/fwupd/metadata.xmlb
00:18:00:0374 XbSilo               file: 172591b9-5ff1-9191-d02b-615f66f5a357, current:172591b9-5ff1-9191-d02b-615f66f5a357, cached: (null)
00:18:00:0374 XbSilo               loading silo with file contents
00:18:00:0375 FuEngine             1176 components now in silo
00:18:00:0380 FuPlugin             init(/app/lib/fwupd-plugins-5/libfu_plugin_optionrom.so)
00:18:00:0380 FuContext            added udev subsystem watch of pci
00:18:00:0380 FuPlugin             init(/app/lib/fwupd-plugins-5/libfu_plugin_linux_lockdown.so)
00:18:00:0380 FuPlugin             init(/app/lib/fwupd-plugins-5/libfu_plugin_elanfp.so)
00:18:00:0381 FuPlugin             init(/app/lib/fwupd-plugins-5/libfu_plugin_colorhug.so)
[... cut ...]
00:18:01:0625 FuPlugin             destroy(upower)
00:18:01:0626 FuPlugin             destroy(modem_manager)
00:18:01:0626 FuPlugin             destroy(linux_lockdown)
00:18:01:0626 FuPlugin             destroy(linux_swap)
00:18:01:0626 FuPlugin             destroy(dell)
00:18:01:0626 FuPlugin             destroy(powerd)
00:18:01:0626 FuPlugin             destroy(linux_tainted)
00:18:01:0626 FuPlugin             destroy(tpm)
00:18:01:0626 FuPlugin             destroy(redfish)
00:18:01:0626 FuPlugin             destroy(uefi_capsule)
```

LinuxOnTheDesktop, 2021-11-26:
Also (though I will move this question, if you like): surely the user is left unclear, upon running `get-updates', about the difference between

`[1] Devices with no available firmware updates`

and

`[2] Devices with the latest available firmware version`

Does _1_ mean that the devices at issue are unknown to fwupd?

hughsie, 2021-11-26:
> Does 1 mean that the devices at issue are unknown to fwupd?

Exactly that. i.e. they *could* have updates, but we don't know of any.

hughsie, 2021-11-26:
> Can you get the flatpak to tell the user about the -i, when the user tries other sudo variants

I guess we could -- if somehow we know if we're not interactive. Is the environment set up any differently in those two cases?

LinuxOnTheDesktop, 2021-11-26:
> Exactly that. i.e. they could have updates, but we don't know of any.

OK; but it is not _obvious_ that that is what is meant.

> Is the environment set up any differently in those two cases?

Well, in shell, one can do the following.

```
#!/usr/bin/env sh
isInteractiveTerminal=1 # i.e., false
if xset q > /dev/null 2>&1 ; then
	# Have DISPLAY SERVER
	if tty -s ; then
		isUnset=${TERM:-}
		[ -z "${isUnset}" ] || isInteractiveTerminal=0 # i.e. true
	fi
fi
```

hughsie, 2021-11-26:
> OK; but it is not obvious that that is what is meant.

Better words welcome dude.

> Well, in shell, one can do the following.

So TERM is set when using `sudo -i`, but is not set just using `sudo`?

LinuxOnTheDesktop, 2021-11-26:
> Better words welcome

I propose the following. Have the message that I numbered '1' read thusly: `Devices not currently supported by fwupd`. Have the message that I numbered '2' read thusly: `Devices that have the latest available firmware`.

> So TERM is set when using `sudo -i`, but is not set just using `sudo`?

No. I misunderstood your earlier question. I thought that by 'interactive' you meant: running in a terminal into which the user can type. That is what my shell snippet checks for. I think that whether TERM is set when using `sudo` is a matter solely of whether the command is run interactively in my sense of 'interactive' and not a function of the presence or absence of `-i`.

superm1, 2021-11-26:
> > Better words welcome
> 
> I propose the following. Have the message that I numbered '1' read thusly: `Devices not currently supported by fwupd`. 

This is wrong though. Fwupd supports them but no updates are available according to the current metadata. This means either:
1. refresh wasn't run recently (stale metadata)
2. nothing is available on supported remotes for this device
3. The remote providing updates is disabled.

> Have the message that I numbered '2' read thusly: `Devices that have the latest available firmware`.
>
>
> No. I misunderstood your earlier question. I thought that by 'interactive' you meant: running [...]

I suspect it's not the TERM variable that matters but something else in the environment copied when you use -i.

hughsie, 2021-11-26:
> but something else in the environment

Hmm, I just tried this and got:

    [hughsie@fedora ~]$ sudo flatpak run --command=sh --devel org.freedesktop.fwupd 
error: "flatpak run" is not intended to be run as `sudo flatpak run`, use `sudo -i` or `su -l` instead and invoke "flatpak run" from inside the new shell

LinuxOnTheDesktop, 2021-11-26:
**On the matter of wording**

At present, the`get-updates` can produce output of this format:

> Devices with no available firmware updates: 
> [. . .]
> Devices with the latest available firmware version:
> [. . .]
> No updates available for remaining devices

As I stuggled to think of a more intelligible version of the first string, I realised that, in order to do so, I would have to know the difference between what that string means to convey and what the third  string means to convey (that third string being, 'Devices with no available firmware updates'). I do not know what that difference is. What is it, please? **TL;DR: what does 'No updates available for remaining devices' mean, if it does not mean any of the things at issue in, 'Devices with no available firmware updates'?**

hughsie, 2022-01-19:
> No updates available for remaining devices

This is confusing, but it's a new issue please.

closed 2022-01-19 by hughsie

LinuxOnTheDesktop, 2022-01-19:
[Done](https://github.com/fwupd/fwupd/issues/4194).