fwupd/fwupd #4047: MOTD doesn't show on Fedora

hughsie, 2021-12-01:
I'm assuming Fedora isn't looking at /run/motd.d/fwupd/85-fwupd -- there's no update-motd like on Ubuntu. @superm1 perhaps we need to install the 85-fwupd.motd to /etc/motd.d/ in this instance?

superm1, 2021-12-01:
Is there pam_motd` by default?  Or anything else that can be used for this purpose?

hughsie, 2021-12-01:
@mezcalero got any insight here? What are we supposed to do?

superm1, 2021-12-01:
I see I left a comment linking back to https://bugs.launchpad.net/ubuntu/+source/pam/+bug/399071 and also mentioning that we don't install 85-fwupd by default outside of Debian/Ubuntu.  
There we install it to `/etc/update-motd.d`.  It essentially just sources `/run/motd.d/fwupd/85-fwupd` which is generated by the daemon.

I would _expect_ Fedora to just get stuff from `/run/motd.d` using `pam_motd`

closed 2021-12-01 by hughsie