fwupd/fwupd #4048: Enable Snap package lzo compression

the-spyke, 2021-12-01:
**Is your feature request related to a problem? Please describe.**

Current snap cold start time is too long.

**Describe the solution you'd like**

Part of the issue is that by default Snap uses `xz` compression. There's a setting to enable much more faster `lzo` compression.

1. https://snapcraft.io/blog/snap-speed-improvements-with-new-compression-algorithm
2. https://snapcraft.io/blog/why-lzo-was-chosen-as-the-new-compression-method

Hoe to enable it: https://forum.snapcraft.io/t/how-to-switch-your-snap-to-use-lzo-compression/21714

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

hughsie, 2021-12-01:
@superm1 do you want to add "compression: lzo" ?? it seems the snap package would get bigger, but start faster -- so a tradeoff to consider. I'm okay with a smaller snap and slower commands if that helps.

superm1, 2021-12-01:
No concerns to me. Seems like a reasonable tradeoff.

hughsie, 2021-12-01:
@the-spyke would you be okay to do a pull request to add the extra line please?

closed 2021-12-02 by the-spyke