fwupd/fwupd #4063: Load firehose bootlader via sysfs API instead of modifying the firmware loadpath

t-8ch, 2021-12-09:
Currently the plugin for ModemManager provides the firehose bootloader firmware to the kernel by modifying the kernels firmware loadpath.
(See `fu_mm_setup_firmware_dir()` and `fu_common_set_firmware_search_path()`)

This is modifying global configuration which can lead to race-conditions with other parts of the system loading firmware at the same time and could corrupt global state if fwupd where to exit without restoring the firmware load path.

In addition to the standard path-based firmware-loading Linux has another, sysfs-based interface which can be used by userspace to explicitly supply firmware files.
See https://docs.kernel.org/driver-api/firmware/fallback-mechanisms.html#firmware-sysfs-loading-facility

This API would work without any modificatoin to global state.

Using this API would require changes to the kernel however.
Unfortunately I cannot test/submit this patch, as the firmware updates files for the devices I have access to are not yet public.
Maybe someone else has the hardware and firmware package to test this.

[0001-bus-mhi-core-Load-firmware-asynchronous.txt](https://github.com/fwupd/fwupd/files/7688258/0001-bus-mhi-core-Load-firmware-asynchronous.txt)

Cc people from #3422 @hughsie @ivanmikh @superm1 @aleksander0m

hughsie, 2021-12-10:
> This is modifying global configuration which can lead to race-conditions

Yes, I was also worried about this.

> Using this API would require changes to the kernel however.

Okay, awesome. Writing to LKML is not good for my blood pressure, but I'm happy to test with whatever I have here. If the concept gets ack'd I'll dig out hardware and test.

t-8ch, 2021-12-10:
> Okay, awesome. Writing to LKML is not good for my blood pressure, but I'm happy to test with whatever I have here. If the concept gets ack'd I'll dig out hardware and test.

Ok, I'll send the patch as RFC to LKML and Cc you.

superm1, 2021-12-10:
> > Okay, awesome. Writing to LKML is not good for my blood pressure, but I'm happy to test with whatever I have here. If the concept gets ack'd I'll dig out hardware and test.
> 
> Ok, I'll send the patch as RFC to LKML and Cc you.

Please cc me too.

t-8ch, 2021-12-10:
@superm1 Will do!

t-8ch, 2021-12-10:
Submitted at https://lore.kernel.org/lkml/20211210161645.10925-1-linux@weissschuh.net/

ivanmikh, 2021-12-11:
Using sysfs API definitely sounds like a good idea. I have hardware and firmware readily available, I am going to compile mhi-next with the new sysfs API and try to test it

t-8ch, 2021-12-11:
I would recommend to first test the current API with my patch. That should still work and give a first hint whether my code is correct.

For the sysfs-API it may be necessary to write at most PAGE_SIZE per write to the `data`-file.

t-8ch, 2021-12-12:
FYI I fixed a few bugs in v1 of the patch. You can find the current version of the patch here: https://git.sr.ht/~t-8ch/linux/log/mhi-firmware

closed 2024-11-11 by hughsie