fwupd/fwupd #4075: Use fwupdtool without root permissions

amfern, 2021-12-12:
From recent conversation, I understood you are in a process of phasing out `dfu-tool` in favor of `fwupdtool`. 
I use `dfu-tool` to flash external microcontrollers, and I don't want to use `sudo` for flashing, Is there a way I can use `fwupdtool` without root?

**Describe the solution you'd like**
Run `fwupdtool` without `sudo`

**Describe alternatives you've considered**
continuing using `dfu-tool`

**Additional context**
Add any other context or screenshots about the feature request here.

hughsie, 2021-12-13:
> and I don't want to use sudo for flashing

Back a bit; can you explain this a bit? If we don't run as UID 0 then we can't interface with devices that do not have some kind of permissive ACL set upon them (e.g. HID devices). Does dfu-tool and dfu-util always work correctly when non-root?

hughsie, 2022-01-06:
Can you try with master now please?

amfern, 2022-01-07:
yes, i tried with master

amfern, 2022-01-07:
Yes, I tried master.
```
$ fwupdmgr --version
client version:	1.7.3-98-g40dad2238
compile-time dependency versions
	gusb:	0.3.8

daemon version:	1.7.3-98-g40dad2238
```
It doesn't work for me, fwupdtool can't run as non-root.

```
$ fwupdtool install-blob --plugins dfu --no-reboot-check obj/fcos_v3.6.0-a1_HUNTER_V0_DIGITAL_KAKUTEF7.dfu
This program may only work correctly as root
Failed to lock: failed to open /run/lock/fwupdtool
```

But if i change the permission of the file `sudo chmod 777 /run/lock/fwupdtool` i am able to flash as expected
```
I expected the fwupdtool to be able to flash the microcontroller without having to be ran as root

superm1, 2022-01-12:
https://github.com/fwupd/fwupd/commit/62a64afda5f2ac3c15aab22c57fb24101e891f41 should hopefully fix this.  Can you please try?

superm1, 2022-01-12:
And now with https://github.com/fwupd/fwupd/commit/e92a286ce847549ef9b2dbc305f089a7689ceedd it shouldn't show the root warning either unless there is a failure.

So please check the latest main and see if we can close this now.

amfern, 2022-01-15:
I get request for root password from polkit when I run the  fwupdtool, although I am still able to flash successfully if I don't type the password.

[image]

The command I use `fwupdtool install-blob --plugins dfu --no-reboot-check obj/fcos_v3.6.0-a1_HUNTER_V0_DIGITAL_KAKUTEF7.dfu`

```
$ fwupdmgr --version
client version:	1.7.4-7-g64e7ea25b
compile-time dependency versions
	gusb:	0.3.10

daemon version:	1.7.4-7-g64e7ea25b
```

hughsie, 2022-01-15:
@superm1 maybe https://github.com/fwupd/fwupd/pull/4178 makes sense for this?

amfern, 2022-01-15:
With the latest main, i no longer get the polkit request and the flashing works without root privileges

amfern, 2022-01-15:
I appreciate your work on this issue for me, I think it can be closed

closed 2022-01-15 by superm1