fwupd/fwupd #4080: Contribution setup breaks Fedora 35 boot

DylanVanAssche, 2021-12-14:
**Describe the bug**

Last night I installed the contribution tools as mentioned in #4076 to fix some indentation in my patches on Fedora 35.
I didn't noticed anything until I powered on my desktop again this morning, Fedora couldn't get into the GNOME desktop environment anymore.

After some emergency mode and dnf history voodoo I figured out the reason: fwupd's contribution setup was causing this.
Steps I did to fix my desktop:

1. I removed all the installed packages with a DNF rollback. Didn't help as expected.
2. I listed the failed systemd units in emergency mode and the Kernel Load Modules one was failing so I had no kernel modules loaded. I removed the modules from the load list that where listed there as failed. No luck either.
3. I digged into the setup script and executed this part manually: https://github.com/fwupd/fwupd/blob/main/contrib/prepare-system#L9-L12 Since then I have a working Fedora 35 Workstation environment again.

**Steps to Reproduce**
1. Run Fedora 35 Workstation
2. Install the contribution scripts
3. Reboot
4. Party!

**Expected behavior**

Install normally without killing the desktop's boot process.

**fwupd version information**

1.7.3

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): git (as I was developing for this)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

Not Applicable, it is about the fwupd contrib scripts

**Additional questions**

- Operating system and version: Fedora 35 Workstation
- Have you tried rebooting? Yes
- Is this a regression? No idea

hughsie, 2021-12-14:
I can't reproduce this here; could you try using `setup` again and deleting the files referenced in cleanup one at a time? Then we can work out which bit is causing the issues. Thanks.

hughsie, 2021-12-14:
Ohh, and in the general case you don't need to install those files -- I install in a prefix and just use that to avoid "polluting" my system versions of stuff.

DylanVanAssche, 2021-12-14:
This seems to be the issue:

```
● dbus-broker.service - D-Bus System Message Bus
     Loaded: loaded (/usr/lib/systemd/system/dbus-broker.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2021-12-14 18:05:54 CET; 5min ago
TriggeredBy: ● dbus.socket
       Docs: man:dbus-broker-launch(1)
    Process: 18260 ExecReload=/usr/bin/busctl call org.freedesktop.DBus /org/freedesktop/DBus org.freedesktop.DBus ReloadConfig (code=exited, status=1/FAILURE)
   Main PID: 1199 (dbus-broker-lau)
      Tasks: 2 (limit: 9337)
     Memory: 7.5M
        CPU: 329ms
     CGroup: /system.slice/dbus-broker.service
             ├─1199 /usr/bin/dbus-broker-launch --scope system --audit
             └─1234 dbus-broker --log 4 --controller 9 --machine-id 83aa80f322964a608d8714ea9fb464ce --max-bytes 536870912 --max-fds 4096 --max-matches 131072 --audit

dec 14 18:10:58 desktop systemd[1]: Reloading D-Bus System Message Bus...
dec 14 18:10:58 desktop dbus-broker-launch[1199]: Access denied in /etc/dbus-1/system-local.conf +1: /etc/dbus-1/system-local.conf
dec 14 18:10:58 desktop busctl[14849]: Call failed: Config invalid. Reload ignored.
dec 14 18:10:58 desktop systemd[1]: dbus-broker.service: Control process exited, code=exited, status=1/FAILURE
dec 14 18:10:58 desktop systemd[1]: Reload failed for D-Bus System Message Bus.
dec 14 18:11:31 desktop systemd[1]: Reloading D-Bus System Message Bus...
dec 14 18:11:31 desktop dbus-broker-launch[1199]: Access denied in /etc/dbus-1/system-local.conf +1: /etc/dbus-1/system-local.conf
dec 14 18:11:31 desktop busctl[18260]: Call failed: Config invalid. Reload ignored.
dec 14 18:11:31 desktop systemd[1]: dbus-broker.service: Control process exited, code=exited, status=1/FAILURE
dec 14 18:11:31 desktop systemd[1]: Reload failed for D-Bus System Message Bus.
```

That's what I noticed then as well, the dbus-broker service failed to start --> everything depending on DBus fails --> no desktop environment.

superm1, 2021-12-14:
> Access denied in /etc/dbus-1/system-local.conf

Could this be a failure from apparmor or selinux?

DylanVanAssche, 2021-12-14:
AFAIK, SELinux is enabled on Fedora, but I'm just a regular user of Fedora, I don't really have knowledge of the inner workings of Fedora TBH. Anyway, the issue is here, maybe this is just happening to me. If anyone has this issue they will probably find it through the search with a possible solution to unbrick their install.

superm1, 2021-12-14:
> AFAIK, SELinux is enabled on Fedora, but I'm just a regular user of Fedora, I don't really have knowledge of the inner workings of Fedora TBH. Anyway, the issue is here, maybe this is just happening [...]

It would be nice to figure out how widespread it is.  Maybe we need the defaults for the contrib script different for Fedora vs others or different by default.

We could also set up the contrib script not to use `/usr/local` but use a prefix path, but I think the same problem will happen from this.

DylanVanAssche, 2021-12-14:
Are there other contributors here running Fedora 35? Maybe they can shine a light on how widespread this is.
I haven't applied any customization to my Fedora setup, just installed with the normal UI installer with full disk encryption.
So no hacks or custom configs ;)

hughsie, 2021-12-14:
If you `setenforce 0` then try to start the service does it work? I'll admit I tested it with selinux turned off.

DylanVanAssche, 2021-12-19:
I did some more experiments and the culprit is `/etc/dbus-1/system-local.conf`.
With or without `setenforce` enabled, the same thing.
I also noticed that the permissions are different from the other config files in there.
This file has 0664 with owner `$USER:$USER`, while the others have 0644 `root:root`.
I tried to match the permissions to see if this would help but not really...

This is what happens if you retry to restart the `dbus-broker` service:

[image]

memily, 2022-03-10:
>Are there other contributors here running Fedora 35? Maybe they can shine a light on how widespread this is.

We, the ardent fans of ./setup, also encountered this issue when running ./setup shortly after installing bare metal Fedora 34/35 on four machines. As discussed above, this is indeed an issue related to selinux with ./setup overwriting labels.

>We could also set up the contrib script not to use /usr/local but use a prefix path, but I think the same problem will happen from this.

Confirmed, same problem. The log for all labels changed is very long.

---
Here are the steps I took to fix the issue for our machines (with reference at end):
1.	After running ./setup we opened terminal on root path: `sudo vim /etc/selinux/config`
2.	Changed enforcing to permissive: `SELINUX=permissive` 
3.	Save file, quit vim, and reboot. 
4.	Open terminal, run command with path set to root: `sudo restorecon -Rv /`
5.	Reboot.
6.	Re-open selinux/config: `sudo vim /etc/selinux/config` 
7.	Changed permissive to enforcing: `SELINUX=enforcing` 
8.	Save file, quit vim, and reboot (we rebooted again, but is optional).

See if this also works for you!
Reference: https://ask.fedoraproject.org/t/fedora-30-selinux-enabled-things-fail-status-226-namespace/1516

chlee75, 2022-03-23:
I have dbus-broker failure when I run ./contrib/setup. And linux do not boot after I reboot system due to dbus error.

To fix this issue, I tried below according to memily's guide, then issue has been solved.
 1. Run ./contrib/setup and remove /etc/dbus-1/system-local.conf if dbus-broker failure happen.
   $cd ./fwupd 
   $./contrib/setup
   [image]
   $sudo rm -f /etc/dbus-1/system-local.conf
 2. Change enforcing to permissive: SELINUX=permissive in /etc/selinux/config
   $sudo vi /etc/selinux/config
 3. Reboot 
 4. Run ./contrib/setup again 
  $cd ./fwupd 
  $./contrib/setup 

If I change permissive to enforcing again and reboot, system do not boot due to dbus error.
It seems that dbus has some trouble with SELINUX=enforcing.

hughsie, 2022-03-23:
@superm1 would you be okay with skipping the /usr/local bits if selinux=enforcing? This is the 2nd person bumping into this.

chlee75, 2022-03-23:
@hughsie
There is no issue with SELINUX=permissive.

superm1, 2022-03-23:
> @superm1 would you be okay with skipping the /usr/local bits if selinux=enforcing? This is the 2nd person bumping into this.

sure, go ahead and make a PR to the setup script to make some noise in this configuration.

closed 2022-03-25 by hughsie