fwupd/fwupd #4098: TPM PCR0 differs from reconstruction

PhilippWendler, 2021-12-16:
**Describe the bug**
fwupd reports that the TPM PCR0 differs from reconstruction and an issue should be created with this [output of fwupdtpmevlog](https://github.com/fwupd/fwupd/files/7729107/fwupdtpmevlog.txt).

Btw.: The [wiki page](https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction#analysis) has a typo in the tool name: `fwupdtpmevelog` should be `fwupdtpmevlog`.

**Steps to Reproduce**
`fwupdmgr security --force`

**Expected behavior**
No such warning, because this is a brand-new system with a fresh installation of Ubuntu 20.04, so neither a hardware failure nor malware are likely. Furthermore, I have several identical machines and it occurs on both machines where I checked it.

**fwupd version information**
1.5.11 (from Ubuntu 20.04 apt package)

**fwupd device information**
This is a Lenovo ThinkCentre M70s with Intel Core i7-10700.

<details>
<summary>Output of "fwupdmgr get-devices --show-all-devices"</summary>

```
11DBSAPP00
│
├─Display controller:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    05
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              768306a9-1c54-5ad2-866c-106768410c94 ← PCI\VEN_8086&DEV_9BC5&SUBSYS_17AA3168&REV_05
│                         93ddcacf-bf57-5d67-9293-ac7c93455565 ← PCI\VEN_8086&DEV_9BC5&SUBSYS_17AA3168
│                         f677e5ac-41ac-5587-b2ef-1322dc207d18 ← PCI\VEN_8086&DEV_9BC5&REV_05
│                         a8dc77a9-e1b4-5ff1-9b0e-72dd2a64d3df ← PCI\VEN_8086&DEV_9BC5
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Embedded Controller:
│     Device ID:          230c9f798927716eeb2c5a2fc8e9304cc17cc68d
│     Current version:    65547
│     Minimum Version:    65547
│     Vendor:             DMI:LENOVO
│     GUIDs:              a7f3a1a9-fd4e-4df9-8c9b-0ce37d1a7a6e
│                         0bbacb87-7824-54ac-b4f1-5422a26aa041 ← UEFI\RES_{A7F3A1A9-FD4E-4DF9-8C9B-0CE37D1A7A6E}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          1b283e8c07f679415e2fe0172ce86d5f233aad3b
│     Current version:    1612776549
│     Minimum Version:    1612776549
│     Vendor:             DMI:LENOVO
│     GUIDs:              6159fb9c-e80b-4d13-9018-bb00bd2f353a
│                         acb3838e-98f7-5a8f-aacc-b10362ce23b5 ← UEFI\RES_{6159FB9C-E80B-4D13-9018-BB00BD2F353A}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel(R) Core™ i7-10700 CPU @ 2.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ec
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         a45b0522-5722-54bd-b802-86cd044262df ← CPUID\PRO_0&FAM_06&MOD_A5
│                         538440cb-7cac-5185-bc01-d93a1e474058 ← CPUID\PRO_0&FAM_06&MOD_A5&STP_5
[... 108 lines cut ...]
✔ Intel BootGuard verified boot: Valid    
✔ Intel DCI debugger:            Locked                                              
✘ IOMMU:                         Not found
✘ TPM PCR0 reconstruction:       Invalid                                             
                                          
HSI-3                                                                                
✔ Intel BootGuard error policy:  Valid                                               
✘ Intel CET Enabled:             Not supported
✘ Pre-boot DMA protection:       Disabled                                            
✘ Suspend-to-idle:               Disabled 
✘ Suspend-to-ram:                Enabled                                             
                                          
HSI-4                                                                                
✔ Intel SMAP:                    Enabled                                             
✘ Encrypted RAM:                 Not supported
                                          
Runtime Suffix -!                         
✔ Linux kernel:                  Untainted                                           
✔ Linux kernel lockdown:         Enabled  
✔ Linux swap:                    Disabled                                            
✔ fwupd plugins:                 Untainted                                                                                                                                
                                                                                     
The TPM PCR0 differs from reconstruction.                                                                                                                                 
 » https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
```

hughsie, 2021-12-16:
Is this fixed in something newer, e.g. 1.7.3?

PhilippWendler, 2021-12-17:
I hadn't checked because the wiki page only mentions that I should use at least 1.5.0. But now I checked with 1.7.3-31-g9f94dc97 from the `--edge` snap and it still reports the same (thanks for fixing #4099 so quickly which allowed me to do this now).

hughsie, 2021-12-17:
If you do `sudo fwupdtool security --force -v` do you get a line like:

    FuPluginTpm          comparing TPM FOO and EVT BARBAZ

Can you also share the output of `sudo ./plugins/tpm/fwupdtpmevlog` from the built tree please.

Thanks!

PhilippWendler, 2021-12-17:
> If you do `sudo fwupdtool security --force -v` do you get a line like:
> 
> ```
> FuPluginTpm          comparing TPM FOO and EVT BARBAZ
> ```

With `1.7.3-31-g9f94dc97` yes, it is
```
FuPluginTpm          comparing TPM 750c738b99a9258e935f0bf00a5ff901bacef8a36f7968691662998559cea54d and EVT 51c3ff2e82c72fe474468de34c7bb6cd8095db69237b1f56d5096ab8cbcd7f8a
```

Here is its full output: [fwupdtool_security_force_v.log](https://github.com/fwupd/fwupd/files/7734355/fwupdtool_security_force_v.log)

> Can you also share the output of `sudo ./plugins/tpm/fwupdtpmevlog -p 0` from the built tree please.

Hm, do you happen to have a binary of that somewhere, e.g., as part of a snap or a Docker container or as a job artifact of CI? I am not sure whether I will find the time to build fwupd on my own soon.

hughsie, 2021-12-17:
> comparing TPM 750c738b99a9258e935f0bf00a5ff901bacef8a36f7968691662998559cea54d...

Interesting; that's a SHA256 hash. Can you share that whole log please?

PhilippWendler, 2021-12-17:
The full output of `fwupdtool security --force -v` is linked above, do you need anything else?

hughsie, 2021-12-17:
In this case I think it might actually be a firmware bug. SeeL

    sudo tpm2_pcrread
    sudo tpm2_eventlog /sys/kernel/security/tpm0/binary_bios_measurements

I bet those pcr0's are different.

PhilippWendler, 2021-12-17:
`tpm2_pcrread` outputs
```
sha1:
sha256:
  0 : 0x750C738B99A9258E935F0BF00A5FF901BACEF8A36F7968691662998559CEA54D
  1 : 0x18583B10341D0ED9F4CF071139CC173BB102061ED6609688E78D84B7644189AE
  2 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  4 : 0x7D5D10F5D05300B1C93F4DC5A9F38BAB21795477A52C17FA88B2A8A20180A6EF
  5 : 0xE3E0F27D492EBD25C6C112C5CC751ADB957A4ACA531DE8B7F7C01DAEDD21BF51
  6 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  7 : 0x1F337CC4973E5D0126C4C3DDE270865C7FCCFD850BFE52B76A498EFD50A7DC23
  8 : 0x4B6ABEE74C0EB7B38BC5ED1016D477AE755BB24DFA5A6EAF3E8843B25E7DC821
  9 : 0xA47BB3AE816178D640E794B720A56D0CE5DE40524760A08904A51E0CD3231E04
  10: 0x98412ACE523391DFAAC78E5CD2B2941980D0D8D390CDAE9E61AE9F2253AFD3E3
  11: 0x0000000000000000000000000000000000000000000000000000000000000000
  12: 0x0000000000000000000000000000000000000000000000000000000000000000
  13: 0x0000000000000000000000000000000000000000000000000000000000000000
  14: 0xE3991B7DDD47BE7E92726A832D6874C5349B52B789FA0DB8B558C69FEA29574E
  15: 0x0000000000000000000000000000000000000000000000000000000000000000
  16: 0x0000000000000000000000000000000000000000000000000000000000000000
  17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  23: 0x0000000000000000000000000000000000000000000000000000000000000000
```

`tpm2_eventlog` outputs this (actually, I copied `binary_bios_measurements` to a different machine and passed it to `tpm2_eventlog` there, I think this should be ok):
```
---
events:
  - EventNum: 0
    PCRIndex: 0
    EventType: EV_NO_ACTION
    Digest: "0000000000000000000000000000000000000000"
    EventSize: 33
    SpecID:
      - Signature: Spec ID Event03
        platformClass: 0
        specVersionMinor: 0
        specVersionMajor: 2
        specErrata: 0
        uintnSize: 2
        numberOfAlgorithms: 1
        Algorithms:
          - Algorithm[0]:
            algorithmId: sha256
            digestSize: 32
        vendorInfoSize: 0
  - EventNum: 1
    PCRIndex: 0
    EventType: EV_NO_ACTION
    DigestCount: 1
    Digests:
      - AlgorithmId: sha256
        Digest: "0000000000000000000000000000000000000000000000000000000000000000"
    EventSize: 17
    Event: "537461727475704c6f63616c6974790000"
  - EventNum: 2
    PCRIndex: 0
    EventType: EV_S_CRTM_VERSION
    DigestCount: 1
    Digests:
      - AlgorithmId: sha256
        Digest: "d4720b4009438213b803568017f903093f6bea8ab47d283db32b6eabedbbf155"
    EventSize: 16
    Event: "1efb6b540c1d5540a4ad4ef4bf17b83a"
[... 1089 lines cut ...]
    EventType: EV_IPL
    DigestCount: 1
    Digests:
      - AlgorithmId: sha256
        Digest: "0de426f490886807d6bec7f7792d81585c2848279b707497f9e68df1bf8e3a9e"
    EventSize: 34
    Event: "2f626f6f742f696e697472642e696d672d352e342e302d39312d67656e6572696300"
pcrs:
  sha256:
    0  : 0x51c3ff2e82c72fe474468de34c7bb6cd8095db69237b1f56d5096ab8cbcd7f8a
    1  : 0x18583b10341d0ed9f4cf071139cc173bb102061ed6609688e78d84b7644189ae
    2  : 0x3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969
    3  : 0x3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969
    4  : 0x7d5d10f5d05300b1c93f4dc5a9f38bab21795477a52c17fa88b2a8a20180a6ef
    5  : 0xe3e0f27d492ebd25c6c112c5cc751adb957a4aca531de8b7f7c01daedd21bf51
    6  : 0x3d458cfe55cc03ea1f443f1562beec8df51c75e14a9fcf9a7234a13f198e7969
    7  : 0x1f337cc4973e5d0126c4c3dde270865c7fccfd850bfe52b76a498efd50a7dc23
    8  : 0x4b6abee74c0eb7b38bc5ed1016d477ae755bb24dfa5a6eaf3e8843b25e7dc821
    9  : 0xa47bb3ae816178d640e794b720a56d0ce5de40524760a08904a51e0cd3231e04
    14 : 0xe3991b7ddd47be7e92726a832d6874c5349b52b789fa0db8b558c69fea29574e
```

PCR0 differs. So that seems to confirm your suspicion. Thanks for the investigation!

Now, because I know that firmware issues are discussed in fwupd-related GitHub repositories if the issues are related to firmware upgrades or LVFS, and because there are contacts of the vendor of this system active here: Is there a way to pursue this further here and maybe ping the vendor's people? But I could understand if you do not want this and I should open a support request (though actually I fear that this would not reach the respective people).

hughsie, 2021-12-17:
If you file it here https://github.com/fwupd/firmware-lenovo-thinkpad/issues/ I can try and find the right person. @mrhpearson might also know a better way to file this internally.