fwupd/fwupd #4103: Udisks required for detecting ESP at /boot/efi?

PhilippWendler, 2021-12-17:
**Describe the question**

Should fwupd be able to recognize an ESP that is mounted as `/boot/efi` without manually specifying it with `OverrideESPMountPoint` even if my system does not have udisks installed? Or do I always need to set `OverrideESPMountPoint` if udisks is not installed?

The reason why this is not clear to me is that both the [docs in the default config](https://github.com/fwupd/fwupd/blob/9f94dc97c4c005dff919402e35b322a9ae12acbf/plugins/uefi-capsule/uefi_capsule.conf#L10-L12) as well as the [plugin docs](https://github.com/fwupd/fwupd/blob/9f94dc97c4c005dff919402e35b322a9ae12acbf/plugins/uefi-capsule/README.md#custom-efi-system-partition) do not mention any requirement for udisks. Instead they just claim that fwupd will detect ESPs at `/boot/efi` (but that doesn't work for me, so I wondered whether something is wrong with my ESP). The name **`Override`**`ESPMountPoint` also does not really sound like a fallback if udisks is not installed. The [wiki page](https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found) of course mentions udisks, but is not really explicit about it being required even if the ESP is mounted in a standard way. Only from https://github.com/fwupd/fwupd/issues/2553#issuecomment-728222721 I started to get the understanding that one of udisks or `OverrideESPMountPoint` seem to be always required. So I would like to ask for clarification.

And if the answer is that one of udisks or `OverrideESPMountPoint` is always required, I would like to suggest to update both `uefi_capsule.conf` and `uefi-capsule/README.md`.

**fwupd version information**
1.7.3-31-g9f94dc97

hughsie, 2021-12-17:
> Or do I always need to set OverrideESPMountPoint if udisks is not installed?

I think this. I don't do any testing without udisks unfortunate. Without udisks detecting the ESP turns into a mad bad of hacks.

hughsie, 2021-12-17:
> would just be nice to make this more explicit in the docs

hughsie, 2021-12-17:
> would just be nice to make this more explicit in the docs

Any chance for a hand? I'm really bad at technical prose like this. Thanks!

PhilippWendler, 2021-12-17:
Let me see, I think I could do this.

PR #4109 (2021-12-17): uefi-capsule: clarify UDisks requirement for ESP autodetection
The ESP is autodetected only if UDisks is available, but this was not explicitly documented so far.

Fixes #4103

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ ] Feature
- [x] Documentation
commit 935e92d0 (2021-12-17): uefi-capsule: clarify UDisks requirement for ESP autodetection

The ESP is autodetected only if UDisks is available,
but this was not explicitly documented so far.

Fixes #4103
closed 2021-12-17 by superm1