fwupd/fwupd #4105: apt purge then snap install presents errors and a roadblock to complete the snap install

OsirianOne, 2021-12-17:
**Describe the bug**
Following the wiki to remove the apt version and install the snap version the following error is presented and there still dbus and other services referencing the system, creating other error messages when apt version of fwupd is reinstalled.

**Steps to Reproduce**
Use the version and installation I have for the OS configuration at the bottom.
sudo apt install fwupd
E: fwupd is not found

snap install fwupd
Please install with... 'list of commands' they all fail with the sane error

**Expected behavior**
When I run an update on fwupd to update, or if I follow the steps on the wiki it would install to take advantage that git is offering in the install. 

**fwupd version information**
Please provide the version of the daemon and client.

Folliowing the steps this is the current condition for both APT and SNAP.  I will then include when I try to reinstall and I will attach the syslog as well

~$ sudo apt purge fwupd
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'fwupd' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

~$ snap remove fwupd
snap "fwupd" is not installed

fwupdmgr --version
Command 'fwupdmgr' not found, but can be installed with:
sudo snap install fwupd  # version 1.7.2, or
sudo apt  install fwupd  # version 1.5.11-0ubuntu1~21.04.1
See 'snap info fwupd' for additional versions.

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
~$: sudo apt -v
apt 2.2.4ubuntu0.1 (amd64)

~$: sudo apt edit-sources

deb-src http://archive.ubuntu.com/ubuntu hirsute main restricted
deb http://us.archive.ubuntu.com/ubuntu/ hirsute main restricted
deb-src http://us.archive.ubuntu.com/ubuntu/ hirsute multiverse main universe
deb http://us.archive.ubuntu.com/ubuntu/ hirsute-updates main restricted
deb-src http://us.archive.ubuntu.com/ubuntu/ hirsute-updates multiverse main un>

~$ snap --version
snap    2.53.4
snapd   2.53.4
series  16
ubuntu  21.04
kernel  5.11.0-41-generic

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

       Can not showing as not installed

Reboot before continuing and sudo apt update

**Attempt to continue with install with snap** 

~$ snap install fwupd --stable --classic
error: cannot perform the following tasks:
- Run install hook of "fwupd" snap if present (run hook "install": install: cannot stat '/snap/fwupd/3664/share/dbus-1/interfaces/org.freedesktop.fwupd.xml': No such file or directory)

**Attempt to install with APT installs the application again**
Ran the following as the first install did not update to the latest in repo

~$ sudo apt reinstall fwupd # 1.5.8-0ubuntu1 amd64
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 1,836 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu hirsute-updates/main amd64 fwupd amd64 1.5.11-0ubuntu1~21.04.1 [1,836 kB]
Fetched 1,836 kB in 1s (2,980 kB/s)
(Reading database ... 185715 files and directories currently installed.)
Preparing to unpack .../fwupd_1.5.11-0ubuntu1~21.04.1_amd64.deb ...
Unpacking fwupd (1.5.11-0ubuntu1~21.04.1) over (1.5.11-0ubuntu1~21.04.1) ...
Setting up fwupd (1.5.11-0ubuntu1~21.04.1) ...
fwupd-offline-update.service is a disabled or a static unit not running, not sta
rting it.
fwupd-refresh.service is a disabled or a static unit not running, not starting i
t.
fwupd.service is a disabled or a static unit not running, not starting it.
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for dbus (1.12.20-1ubuntu3) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...

~$: reboot

**back where I started; no wiser and removing fwupd in this version impish indri, groovy, gorilla, 

**Additional questions**

- Operating system and version:
   -- Linux abydos 5.11.0-41-generic #45-Ubuntu SMP Fri Nov 5 11:37:01 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
   -- Ubuntu 21.10 Hirsute Hippo - after installing the desktop version and ironing out the issues with gnome-shell, Geoclue-2, network issues with DNS, and getting all the timing adjusted.  Allowed Snap Store to install the upgrade that moved part of the OS to 21.04, some updates come from 20.04 (Focal Fossa) repository and a few come now from 21.04.  I stopped it from updating to 21.10 (Impish Indri) which it tried to do twice in the last day..
   -- snap.store seemed to always have issues until today all of the timing and networks were blessed and got a good almost no error boot and then thought I would go ahead and try to enhance the firmware and put it over to the snap store version to get some more control over the version of firmware that gets loaded.

- Have you tried rebooting?  Yes - especially after following the instructions on this wiki that instructs to do 'sudo apt purge fwupd'  which did uninstall but would not install the snap version.  Thus be bug being filed above.

- Is this a regression?
   -- This seems to have happened before with other users, but seems to have been closed and directive in the defect was to file a new bug as to investigate the issue separate.  Additionally since this involves errors in freedesktop.systemd.dbus  and the fwupd still not removed i may have something to do with the error I have.

hughsie, 2021-12-17:
I've pulled fde47a4159d6cccaa06e39d59ff2cb3439943961 into main; does that help? CI will take a little while to regenerate the snap package.

superm1, 2021-12-17:
> I've pulled [fde47a4](https://github.com/fwupd/fwupd/commit/fde47a4159d6cccaa06e39d59ff2cb3439943961) into main; does that help? CI will take a little while to regenerate the snap package.

It should for this problem.

it will be on the edge channel, please verify from that.

superm1, 2022-01-14:
I verified that 1.7.4 release works now.

closed 2022-01-14 by superm1