fwupd/fwupd #4114: Branches are not remembered after switch

DylanVanAssche, 2021-12-20:
**Describe the bug**

Branches are not remembered when using `fwupmgr switch branch $GUID $BRANCH`.
As soon as the switch is complete and the alternative firmware is flashed, the branch seems to be forgotten according to the fwupd logs:

```
17:06:51:0297 FuEngine             ignoring EG25GGBR07A08M2G - 05010820 == EG25GGBR07A08M2G - 05010820
17:06:51:0297 FuEngine             ignoring release PINEPHONE_MODEM_SDK_0.4.10-002 as branch biktorgj-002, and device is (null)
```

I used `switch-branch` to move to the `biktorgj-002` branch, but that doesn't seem to be picked up.
The firmware version `PINEPHONE_MODEM_SDK_0.4.10-002` is different from (and higher than) the one reported by the modem `EG25GGBR07A08M2G - 05010820`, so an update should be advertised.

Is there a way to see on which branch a device currently is? I don't see it with `get-devices`.

**Steps to Reproduce**
1. Switch to a different branch
2. Finish the upgrade
3. Put a newer firmware in the repos
4. No updates are advertised

**Expected behavior**

The newer firmware is detected, processed and shown as an update.

**fwupd version information**
f6833f96c309e74da7f537bc0fa11d7ad52dfb3a

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): git

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
Pine64 PinePhone (1.2)
│
├─DA4032:
│     Device ID:          0e0c93c8b4bb222157feedbde8f863e23bd1a8f7
│     Current version:    0x3034313430363139
│     Vendor:             EMMC:0x000045
│     Serial Number:      0x67ecec14
│     GUIDs:              8c6f96d4-f406-53fa-84ca-a367cd368dfc ← EMMC\DA4032
│                         add792f2-7053-5638-80e7-20987fd16769 ← EMMC\0069&0256
│                         5500994c-1b73-52b4-acf4-bb574b6f3029 ← EMMC\0069&0256&DA4032
│     Device Flags:       • Internal device
│                         • Updatable
│   
└─QUECTEL Mobile Broadband Module:
      Device ID:          976c4a39e87f61e6940ea6a8d39c583cfa99615f
      Summary:            Quectel EG25-G modem
      Current version:    EG25GGBR07A08M2G - 05010820
      Vendor:             QUALCOMM INCORPORATED (USB:0x2C7C)
      GUIDs:              db379a33-254f-5140-b37e-d36ae7e5c039
                          c33a4560-8681-55b6-bbb6-85258f2de149 ← USB\VID_2C7C&PID_0125&REV_0318&CARRIER_ROW_GENERIC_3GPP
                          1a2996cb-f86e-5583-a464-e1b96e1c6ae9 ← USB\VID_2C7C&PID_0125&REV_0318
                          587bf468-6859-5522-93a7-6cce552a0aa3 ← USB\VID_2C7C&PID_0125
                          22ae45db-f68e-5c55-9c02-4557dca238ec ← USB\VID_2C7C
      Device Flags:       • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Device supports switching to a different branch of firmware

```

**Additional questions**

- Operating system and version: postmarketOS edge, PinePhone
- Have you tried rebooting? Yes
- Is this a regression? Yes, it should work as it is supported for firmware like coreboot: https://fwupd.org/lvfs/devices/com.starlabs.L3.coreboot.flashrom

hughsie, 2021-12-20:
> the branch seems to be forgotten

The way we implemented this was that the device would "know" what branch it's on and set it as a property. Is there any way to "ask" the modem and find out what variant of firmware is running?

DylanVanAssche, 2021-12-20:
> The way we implemented this was that the device would "know" what branch it's on and set it as a property. Is there any way to "ask" the modem and find out what variant of firmware is running?

Oh that might explain it...
The alternative firmware has a different bootloader/kernel/userspace than the original firmware, but the Hexagon firmware (the baseband firmware of the modem) is the same as that is closed source. Because the modem versions are reported to fwupd, they are the same as the modem baseband firmware is the same. 

Currently, there's no universal way to ask this question to the modem I'm afraid.
I thought fwupd would know this somehow by saving the branch somewhere, but it seems to fetch it from the device...

hughsie, 2021-12-20:
> The alternative firmware has a different bootloader/kernel/userspace than the original firmware

Is there any way of detecting it (even as a boolean) from userspace?

DylanVanAssche, 2021-12-20:
> Is there any way of detecting it (even as a boolean) from userspace?

The alternative firmware has custom AT commands available, but that's specific for this modem.
Maybe if we have a custom AT command to identify the alternative firmware we can add it as a quirk? 
Like: if the modem supports the command, it reports a different branch?

hughsie, 2021-12-20:
>  Maybe if we have a custom AT command to identify the alternative firmware we can add it as a quirk?

Could we have a custom AT query for the new firmware to return the branch name? The original firmware will ignore it, and the branch will be `NULL` and that's probably exactly what we want.

DylanVanAssche, 2021-12-20:
> Could we have a custom AT query for the new firmware to return the branch name? The original firmware will ignore it, and the branch will be NULL and that's probably exactly what we want.

Probably yes, I asked the developer if that's possible.
We would need 3 different branches:
- default one = Quectel
- alternative 002 = alternative with Hexagon firmware 002
- alternative 003 = alternative with Hexagon firmware 003

The differences between 002 and 003 are:
- Some users have better stability with 002 than 003, others the other way around
- 003 provides more VoLTE profiles

If such command is possible, we can return the branch name with the hexagon firmware name included as well :)

DylanVanAssche, 2021-12-28:
@hughsie I have now locally this working, but the AT command is hardcoded, I suppose this needs to be a quirk.
Any suggestions for the name of this quirk?

What about `ModemManagerBranchATCommand`?
```
ModemManagerBranchATCommand = "AT+GETFWBRANCH"
```

hughsie, 2021-12-28:
I guess snake case 'At' but everything else seems good, thanks.