fwupd/fwupd #4127: fwupd downloads always time out

aFineBoi, 2021-12-24:
```
# fwupdmgr refresh --verbose --force
(fwupdmgr:3325): GLib-DEBUG: 15:28:43.866: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:3325): Fwupd-DEBUG: 15:28:43.887: Emitting ::status-changed() [idle]
Updating lvfs
(fwupdmgr:3325): Fwupd-DEBUG: 15:28:43.893: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
(fwupdmgr:3325): Fwupd-DEBUG: 15:28:43.893: Emitting ::status-changed() [downloading]
Downloading…             [ -                                     ](fwupdmgr:3325): Fwupd-DEBUG: 15:29:43.897: Emitting ::status-changed() [idle]
(fwupdmgr:3325): Fwupd-DEBUG: 15:29:43.897: status-code was 0
Downloading…             [***************************************]
failed to download file: Operation timed out after 60000 milliseconds with 0 out of 0 bytes received
```
I've purged and reinstalled fwupd, and rebooted the PC afterwards, but the problem persists.

aFineBoi, 2021-12-24:
OS is Ubuntu LTS 20.04.3 with KDE, and I've purged Plasma Discover Store and Snap Store for troubleshooting. The both depend on fwupd.

Also, when updating with Snap Store, everything works fine except fwupd part. It says connection terminated for fwupd.

aFineBoi, 2021-12-24:
I'm in a working environment, and I need to connect to Internet with a proxy.

Config files for proxy were created for a workaround, one is /etc/systemd/system/fwupd.service.d/http-proxy.conf, another is /etc/systemd/system/fwupd-refresh.service.d/http-proxy.conf
```
[Service]
Environment="http_proxy=http://1.2.3.4:5678/"
```

After reboot, I verified the proxy with these:
```
# systemctl status fwupd
● fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static; vendor preset: enabled)
    Drop-In: /etc/systemd/system/fwupd.service.d
             └─http-proxy.conf
     Active: active (running) since Thu 2021-12-23 20:56:18 AWST; 21min ago
       Docs: https://fwupd.org/
   Main PID: 2879 (fwupd)
      Tasks: 5 (limit: 18943)
     Memory: 15.9M
     CGroup: /system.slice/fwupd.service
             └─2879 /usr/libexec/fwupd/fwupd

Dec 23 20:56:17 dell-g3-3590 systemd[1]: Starting Firmware update daemon...
Dec 23 20:56:18 dell-g3-3590 systemd[1]: Started Firmware update daemon.
```
```
# systemctl status fwupd-refresh
● fwupd-refresh.service - Refresh fwupd metadata and update motd
     Loaded: loaded (/lib/systemd/system/fwupd-refresh.service; static; vendor preset: disabled)
    Drop-In: /etc/systemd/system/fwupd-refresh.service.d
             └─http-proxy.conf
     Active: inactive (dead)
TriggeredBy: ● fwupd-refresh.timer
       Docs: man:fwupdmgr(1)
```
```
# systemctl show fwupd-refresh | grep proxy
Environment=http_proxy=http://1.2.3.4:5678/
DropInPaths=/etc/systemd/system/fwupd-refresh.service.d/http-proxy.conf
```

I believe I've got this right. My Snap Store has similar proxy settings and it works just fine. However, fwupd is just not working:
```
# fwupdmgr refresh --force
Updating lvfs
Downloading…             [ -                                     ]failed to download file: Operation timed out after 60001 milliseconds with 0 out of 0 bytes received
```

aFineBoi, 2021-12-25:
Network status:
```
# snap install snap-store
# sudo netstat -tupn
Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.1:34122         1.2.3.4:5678          ESTABLISHED 4096/snapd
```
```
# fwupdmgr refresh --verbose
# sudo netstat -tupn
Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name              
tcp        0    370 192.168.1.2:37210       151.101.194.49:443      ESTABLISHED 3666/fwupdmgr
```

aFineBoi, 2021-12-26:
Got the proxy working though a program named `proxychains-ng`, by tweaking its config file `/etc/proxychains4.conf`, and running following command:
```
# proxychains4 fwupdmgr refresh --verbose
[proxychains] config file found: /etc/proxychains4.conf
[proxychains] preloading /usr/lib/x86_64-linux-gnu/libproxychains.so.4
[proxychains] DLL init: proxychains-ng 4.14
(fwupdmgr:13791): GLib-DEBUG: 20:46:10.941: setenv()/putenv() are not thread-safe and should not be used after threads are created
[proxychains] DLL init: proxychains-ng 4.14
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:12.206: Emitting ::status-changed() [idle]
Updating lvfs
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:12.212: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:12.212: Emitting ::status-changed() [downloading]
Downloading…             [ -                                     ][proxychains] Strict chain  ...  1.2.3.4:5678  ...  cdn.fwupd.org:443  ...  OK
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:13.447: download progress: 48%
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:13.447: download progress: 48%
Downloading…             [******************                     ](fwupdmgr:13791): Fwupd-DEBUG: 20:46:13.489: download progress: 100%
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:13.489: download progress: 100%
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:13.489: download progress: 100%
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:13.489: download progress: 100%
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:13.489: Emitting ::status-changed() [idle]
Downloading…             [***************************************]
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:13.489: changing metadata URI from https://cdn.fwupd.org/downloads/firmware.xml.gz to https://cdn.fwupd.org/downloads/firmware-02765-stable.xml.gz
(fwupdmgr:13791): Fwupd-DEBUG: 20:46:13.489: metadata signature of lvfs is unchanged, skipping
Successfully downloaded new metadata: 1 local device supported
```

superm1, 2021-12-29:
proxychains-ng feels like a hack to me still TBH.  I am surprised that the environment variable via drop in configuration isn't working.  Can you please confirm if manually running this fails (with proxychains-ng disabled):

`http_proxy=http://1.2.3.4:5678/ fwupdmgr refresh --verbose`

aFineBoi, 2021-12-30:
Yeah, that looks about right

**Before**
```
# fwupdmgr refresh --verbose --force
(fwupdmgr:3207): GLib-DEBUG: 11:43:30.345: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:3207): Fwupd-DEBUG: 11:43:30.354: Emitting ::status-changed() [idle]
Updating lvfs
(fwupdmgr:3207): Fwupd-DEBUG: 11:43:30.356: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
(fwupdmgr:3207): Fwupd-DEBUG: 11:43:30.356: Emitting ::status-changed() [downloading]
Downloading…             [ -                                     ](fwupdmgr:3207): Fwupd-DEBUG: 11:44:30.359: Emitting ::status-changed() [idle]
(fwupdmgr:3207): Fwupd-DEBUG: 11:44:30.359: status-code was 0
Downloading…             [***************************************]
failed to download file: Operation timed out after 60001 milliseconds with 0 out of 0 bytes received
```
**After**
```
# http_proxy="http://1.2.3.4:5678/" fwupdmgr refresh --verbose --force
(fwupdmgr:3332): GLib-DEBUG: 11:45:24.141: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:24.150: Emitting ::status-changed() [idle]
Updating lvfs
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:24.153: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.jcat
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:24.153: Emitting ::status-changed() [downloading]
Downloading…             [ -                                     ](fwupdmgr:3332): Fwupd-DEBUG: 11:45:32.733: download progress: 0%
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:32.733: download progress: 0%
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:33.735: download progress: 0%
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:33.735: download progress: 0%
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:34.615: download progress: 100%
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:34.615: download progress: 100%
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:34.615: download progress: 100%
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:34.615: download progress: 100%
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:34.615: Emitting ::status-changed() [idle]
Downloading…             [***************************************]
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:34.617: changing metadata URI from https://cdn.fwupd.org/downloads/firmware.xml.gz to https://cdn.fwupd.org/downloads/firmware-02775-stable.xml.gz
(fwupdmgr:3332): Fwupd-DEBUG: 11:45:34.617: metadata signature of lvfs is unchanged, skipping
Successfully downloaded new metadata: 1 local device supported
```

superm1, 2021-12-30:
Seems like it's behaving properly then.

Are you sure you did the reload of systemd units properly when you did this with systemd?

aFineBoi, 2021-12-30:
I did a `daemon-reload` right after creating that proxy config file, and `fwupd` is just not going through the proxy, despite `systemctl` is telling me that the environment variable is dropped in through that particular file. Rebooting didn't solved that problem either.

Also, I don't think it's systemd's problem, since `snapd` works properly with the same method.

superm1, 2021-12-30:
Wait a minute - I just re-read your first post.  Is the refresh unit actually failing or are you just mentioning that the configuration file drop in doesn't apply to the manually launched process?

The manually launched process needs it set in the environment of the user who launched it.

aFineBoi, 2021-12-30:
Well, that explains a lot of things. Guess I'm just inexperienced. Thank for your help anyway

closed 2021-12-30 by aFineBoi