fwupd/fwupd #4128: modem-manager: fwupd crashes when modem is reprobed by ModemManager

DylanVanAssche, 2021-12-26:
**Describe the bug**

In some rare cases, modems may disconnect from the system causing ModemManager to reprobe the modem and expose it under a new DBus object. When this happens when fwupd tries to detach the modem using the modem-manager plugin, fwupd crashes.
Fwupd never asks ModemManager to inhibit the modem, instead it crashes and is restarted by the supervisor or DBus.

I truncated the logs right before starting an upgrade so I don't have the full logs, the logs became too big otherwise to find this needle in the haystack.

**Logs for a crash**

*fwupd*

fwupd doesn't seem to noticed that the modem ID is here incremented to `1` as a port hangup happened.

```
10:36:50:0909 FuPlugin             detach(modem_manager)
10:36:50:0983 FuPluginMm           No response expected for AT command: 'AT+QFASTBOOT', assuming succeed
10:36:50:0985 FuMain               Emitting PropertyChanged('Status'='device-write')
10:36:50:0985 FuMain               Emitting PropertyChanged('Percentage'='2%')
10:36:50:0986 FuDeviceList         waiting 20000ms for replug
--------- fwupd segfaults and restarts ----------
10:36:51:0218 FuDebug              Verbose debugging enabled (on console 0)
10:36:51:0220 FuConfig             loading config values from /etc/fwupd/daemon.conf
10:36:51:0220 FuConfig             using autodetected max archive size 18.4 EB
10:36:51:0225 FuRemoteList         loading /usr/share/fwupd/metainfo
```

*ModemManager*

MM reprobed the modem when sending 'AT' and after response a port hangup happened.

```
[25098]: <debug> [1640169410.989355] [modem1/ttyUSB2/at] <-- '<CR><LF>OK<CR><LF>'
[25098]: <debug> [1640169411.072749] [modem1/ttyUSB2/at] unexpected port hangup!
[25098]: <debug> [1640169411.073094] [modem1/ttyUSB2/at] forced to close port
[25098]: <debug> [1640169411.073172] [modem1/ttyUSB2/at] device open count is 0 (close)
[25098]: <debug> [1640169411.073283] [modem1/ttyUSB2/at] closing serial port...
[25098]: <debug> [1640169411.073479] [modem1/ttyUSB2/at] serial port closed
[25098]: <info>  [1640169411.076516] [base-manager] port ttyUSB0 released by device '/sys/devices/platform/soc/1c1b000.usb/usb2/2-1'
[25098]: <info>  [1640169411.080653] [base-manager] port ttyUSB1 released by device '/sys/devices/platform/soc/1c1b000.usb/usb2/2-1'
[25098]: Cannot read from istream: connection broken
[25098]: [/dev/cdc-wdm0] QMI endpoint hangup: removed
[25098]: <info>  [1640169411.088265] [modem1] connection to qmi-proxy for /dev/cdc-wdm0 lost, reprobing
[25098]: <debug> [1640169411.089395] [device /sys/devices/platform/soc/1c1b000.usb/usb2/2-1] unexported modem from path '/org/freedesktop/ModemManager1/Modem/1'
[25098]: <debug> [1640169411.089725] [cdc-wdm0/qmi] Releasing client for service 'voice'...
```

**Logs without a crash**

*fwupd*

I added some debug statements inside `fu_plugin_mm_uninhibit_device` to see if it gets called.

```
10:43:57:0464 FuPlugin             detach(modem_manager)
10:43:57:0465 FuPluginMm           -------------------- fu_plugin_mm_uninhibit_device BEGIN ------------------
10:43:57:0465 FuPluginMm           g_clear_object OK
10:43:57:0465 FuPluginMm           fu_plugin_mm_udev_device_removed OK
10:43:57:0465 FuPluginMm           -------------------- fu_plugin_mm_uninhibit_device END ------------------
10:43:57:0465 FuPluginMm           inhibit modemmanager device with uid /sys/devices/platform/soc/1c1b000.usb/usb2/2-1
10:43:57:0564 FuPluginMm           No response expected for AT command: 'AT+QFASTBOOT', assuming succeed
10:43:57:0580 FuMain               Emitting PropertyChanged('Status'='device-write')
10:43:57:0581 FuMain               Emitting PropertyChanged('Percentage'='2%')
10:43:57:0582 FuDeviceList         waiting 20000ms for replug
10:43:57:0584 FuPluginMm           removed modem: /org/freedesktop/ModemManager1/Modem/0
10:43:57:0584 FuPlugin             emit removed from modem_manager: 976c4a39e87f61e6940ea6a8d39c583cfa99615f
10:43:57:0586 FuDeviceList         waiting 20000ms for QUECTEL Mobile Broadband Module device removal
```

*ModemManager*

The modem ID is still `0` here as no disconnect happened.

```
[2075]: <debug> [1640169837.485402] [device /sys/devices/platform/soc/1c1b000.usb/usb2/2-1] unexported modem from path '/org/freedesktop/ModemManager1/Modem/0'
[2075]: <debug> [1640169837.485815] [cdc-wdm0/qmi] Releasing client for service 'voice'...
[2075]: [/dev/cdc-wdm0] Releasing 'voice' client with flags 'release-cid'...
```

**Steps to Reproduce**

This is rather hard to reproduce as it depends on the modem disconnecting from the system during the detach.
If you run alternative firmware you can reproduce this issue way more reliable by forcing an USB reset with `AT+RESETUSB` and start the upgrade directly after that.

1. Start upgrade
2. Wait for detach
3. Update fails since fwupd disconnects and the client reports 'message did not receive a reply ...' since fwupd is crashed

**Expected behavior**

fwupd doesn't crashes, uses the new modem object correctly.

**fwupd version information**

```
1.7.4
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): git

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```
Quectel EG25-G modem
```

**Additional questions**

- Operating system and version: postmarketOS edge, musl, Pine64 PinePhone
- Have you tried rebooting? Yes
- Is this a regression? No

superm1, 2021-12-26:
Could you turn on abrt or a crash dump handler to catch it?

If not maybe can you launch the daemon under gdb?

DylanVanAssche, 2021-12-26:
> Could you turn on abrt or a crash dump handler to catch it?

I had a look at the meson options and the args of `fwupd` but I don't see how I should enable this mode.
Is there something I missed?

> If not maybe can you launch the daemon under gdb?

Trying to do that, but of course it doesn't reproduce now.
This issue comes up once in a while and is hard to reproduce.

superm1, 2021-12-26:
> > Could you turn on abrt or a crash dump handler to catch it?
> 
> I had a look at the meson options and the args of `fwupd` but I don't see how I should enable this mode.
> Is there something I missed?

It's an OS level thing for catching crashes. Different distributions do it differently. I'm not familiar with postmarket OS so I don't know the common thing used there.

Basically when a process crashes the crash handler will get a crash dump that you can tear apart and use the backtrace to figure out what happened.

> 
> > If not maybe can you launch the daemon under gdb?
> 
> Trying to do that, but of course it doesn't reproduce now.
> This issue comes up once in a while and is hard to reproduce.

Ah too bad

t-8ch, 2022-01-21:
I have probably the same problem.
Stack trace:
```
#0  0x00007f4bad424a7f in fu_plugin_mm_device_add (plugin=plugin@entry=0x55ebc8c289f0, modem=modem@entry=0x55ebc8bf9b90) at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:231
#1  0x00007f4bad424bf0 in fu_plugin_mm_setup_manager (plugin=0x55ebc8c289f0) at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:313
#2  fu_plugin_mm_name_owner_updated (plugin=0x55ebc8c289f0) at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:329
#3  0x00007f4bad424d2f in fu_plugin_mm_coldplug (plugin=0x55ebc8c289f0, error=<optimized out>) at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:342
#4  0x00007f4bb11aae74 in fu_plugin_runner_coldplug (self=self@entry=0x55ebc8c289f0, error=error@entry=0x7ffecc5dd5b8) at ../fwupd-1.7.4/libfwupdplugin/fu-plugin.c:1023
#5  0x000055ebc84969c2 in fu_engine_plugins_coldplug (self=0x55ebc8b59030) at ../fwupd-1.7.4/src/fu-engine.c:5591
#6  fu_engine_load (self=<optimized out>, flags=flags@entry=14, error=error@entry=0x7ffecc5dd648) at ../fwupd-1.7.4/src/fu-engine.c:6993
#7  0x000055ebc847c36e in main (argc=<optimized out>, argv=<optimized out>) at ../fwupd-1.7.4/src/fu-main.c:2112
```

Culprit:
```c
          if (!fu_device_setup(FU_DEVICE(dev), &error)) {
                  g_warning("failed to probe MM device: %s", error->message);
                  return;
          }
```
It seems `fu_device_setup()` fails without allocating an error object and then the logging call derefs the NULL pointer.

The original error return comes from `fu_mm_device_setup()` -> `fu_mm_device_io_open()` -> `fu_io_channel_new_file(port_at)` because `port_at` is `NULL`.

```
11:55:17:0765 FuIOChannel          fu_io_channel_new_file: assertion 'filename != NULL' failed
```

hughsie, 2022-01-21:
@t-8ch you're seeing https://github.com/fwupd/fwupd/commit/0264c29bdaa2d96308052c1fd0d9ff8537501a5b

t-8ch, 2022-01-21:
@hughsie Indeed, thanks!

closed 2023-02-20 by hughsie