fwupd/fwupd #4131: Distinguish build-depencies from nice-to-have in fwupd_setup_helpers.py

kerneis-anssi, 2021-12-28:
`contrib/ci/fwupd_setup_helpers.py` has a method `get-dependencies` that will output a superset of dependencies strictly required to build. The Build-Depends for Debian amount to 900MB (transitively) whereas the total for the output of `get-dependencies` is 1.3GB. If I understand the scripts correctly (in particular `generate_debian.py`), the issue is that it restricts the content of `dependencies.xml` to entries of type `build` but does not further restrict to those containing a `<control />` element, which is what is used to generate build-depends on Debian (because the file listing them is called `control`). clang-format is an example of a non-essential dependency.

This should be a low-priority request because I now have a workaround, but it may help simplify your Python scripts by consolidating some of the logic between fwupd_setup_helpers.py and generate_debian.py. I haven't looked at what is done for other distributions.

kerneis-anssi, 2021-12-28:
For the record, to list the size of each transitive build dependency (you'll need the `apt-rdepends` package):
```
apt-cache showsrc fwupd | grep ^Build-Depends|grep -o " [-a-zA-Z0-9]\+[ ,]"|sed 's/[ ,]//g'| \
  xargs apt-rdepends|grep -v '^ '|sort -u|xargs dpkg-query -Wf '${Installed-Size} ${Package}\n'|sort -n
```
Replace the first line with `python3 ./contrib/ci/fwupd_setup_helpers.py get-dependencies --os debian` to get the other side of the coin.

superm1, 2021-12-29:
Can you better describe how you are wanting to use these scripts?  They're mostly intended for two purposes:
1. CI
2. Getting a foolproof environment for contributors setup effectively (IE `contrib/setup`)

You're absolutely right that `clang-format` for example isn't needed to build, but we absolutely want everyone contributing to fwupd to have it installed.  Precommit will skip it if it's missing, and we don't want to nitpik contributors whitespace and style problems.  It's a waste of everyone's time.

kerneis-anssi, 2021-12-29:
The context is https://github.com/ANSSI-FR/chipsec-check, a script to build a Debian-based bootable USB key to run firmware diagnostic tools. Right now, the released version installs fwupd from Debian repositories, but I wanted to add an option to switch it to fwupd git repository. I could of course compile build a Debian package on the host and install that on the key, but experience with chipsec (which is already included on the key as a git repo) has shown that it is often valuable to be able to try a quick patch directly from the key and immediately recompile on the system under test. That's why I wanted a simple way to add all build-depends while keeping things small if possible (because we try and release a 2GB version of the key as a github release).

Anyway, I complained on github and @hughsie suggested I filed a bug report but really, this is such a corner-case made of odd requirements that I'd be fine with a won't-fix.

hughsie, 2021-12-29:
> but I wanted to add an option to switch it to fwupd git repository

Back a bit; couldn't you install the git .deb or snap package generated by CI? Is there any specific reason you _want_ to build everything by source?

superm1, 2021-12-29:
> > but I wanted to add an option to switch it to fwupd git repository
> 
> Back a bit; couldn't you install the git .deb or snap package generated by CI? Is there any specific reason you _want_ to build everything by source?

Snap seems pretty easy to do and well contained - we do publish it to edge channel and all dependencies are installed in the snap package.

Deb we don't save any artifacts, I think this is a harder thing to accomplish.

superm1, 2021-12-29:
> The context is https://github.com/ANSSI-FR/chipsec-check, a script to build a Debian-based bootable USB key to run firmware diagnostic tools. Right now, the released version installs fwupd from Debian repositories, but I wanted to add an option to switch it to fwupd git repository. I could of course compile build a Debian package on the host and install that on the key, but experience with chipsec (which is already included on the key as a git repo) has shown that it is often valuable to be able to try a quick patch directly from the key and immediately recompile on the system under test. That's why I wanted a simple way to add all build-depends while keeping things small if possible (because we try and release a 2GB version of the key as a github release).
> 
> Anyway, I ranted on twitter and @hughsie suggested I filed a bug report but really, this is such a corner-case made of odd requirements that I'd be fine with a won't-fix.

If you need a step to install a git debian package - the "proper" way is to run pbuilder/pdebuild on the source package. Then your build dependencies won't end up in the chroot used to build the USB key, only runtime ones.

kerneis-anssi, 2021-12-29:
> If you need a step to install a git debian package - the "proper" way is to run pbuilder/pdebuild on the source package.

I'll go that route, thanks for the discussion. As I said, there are cases when iterating over a patch where being able to rebuild from the key is convenient, but it's a rare enough event that I can rebuild on a separate computer and transfer the .deb.

closed 2021-12-29 by kerneis-anssi