fwupd/fwupd #4140: Failing tests fu-self-test on GNU Guix

phodina, 2022-01-03:
**Describe the bug**
I'm porting the fwupd to GNU Guix. After the package is built the tests are run. There is one test which fails.

**Steps to Reproduce**
Apply package definition in the [patches](https://issues.guix.gnu.org/46278) and build the package.

**Expected behavior**
Tests pass.

**fwupd version information**
Version 1.55.0

```shell
./pre-inst-env guix build fwupd -K
...
phase `build' succeeded after 14.9 seconds
starting phase `check'
ninja: Entering directory `/tmp/guix-build-fwupd-1.5.5.drv-0/build'
ninja: no work to do.
 1/27 fwupd-exported-api             OK              0.00s
 2/27 fwupd-self-test                OK              0.01s
 3/27 fwupd-thread-test              OK              0.01s
 4/27 fwupd-context-test             OK              0.01s
 5/27 fwupdplugin-exported-api       OK              0.00s
 6/27 fwupdplugin-self-test          OK              2.13s
 7/27 fu-self-test                   FAIL            0.39s   killed by signal 6 SIGABRT
>>> MALLOC_PERTURB_=10 /tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test
????????????????????????????????????? ?  ?????????????????????????????????????
stdout:
# random seed: R02Scbd0a6868f86473cd36d557c8a7f2941
# FuCommon-DEBUG: removing /tmp/fwupd-self-test
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuCommon-DEBUG: 22:15:28.512: removing /tmp/fwupd-self-test
# FuCommon-DEBUG: removing /tmp/fwupd-self-test/var
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuCommon-DEBUG: 22:15:28.513: removing /tmp/fwupd-self-test/var
# FuCommon-DEBUG: removing /tmp/fwupd-self-test/var/cache
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuCommon-DEBUG: 22:15:28.513: removing /tmp/fwupd-self-test/var/cache
# FuCommon-DEBUG: removing /tmp/fwupd-self-test/var/cache/fwupd
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuCommon-DEBUG: 22:15:28.513: removing /tmp/fwupd-self-test/var/cache/fwupd
# FuPlugin-DEBUG: init(/tmp/guix-build-fwupd-1.5.5.drv-0/build/plugins/test/libfu_plugin_test.so)
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuPlugin-DEBUG: 22:15:28.513: init(/tmp/guix-build-fwupd-1.5.5.drv-0/build/plugins/test/libfu_plugin_test.so)
# FuPluginTest-DEBUG: init
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuPluginTest-DEBUG: 22:15:28.513: init
1..42
# Start of fwupd tests
# FuEngine-DEBUG: failed to build machine-id: The machine-id is not present
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuEngine-DEBUG: 22:15:28.514: failed to build machine-id: The machine-id is not present
# FuConfig-DEBUG: loading config values from /tmp/guix-build-fwupd-1.5.5.drv-0/source/data/tests/fwupd/daemon.conf
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuConfig-DEBUG: 22:15:28.514: loading config values from /tmp/guix-build-fwupd-1.5.5.drv-0/source/data/tests/fwupd/daemon.conf
# FuConfig-DEBUG: failed to read UpdateMotd key: Key file does not have key ?UpdateMotd? in group ?fwupd?
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuConfig-DEBUG: 22:15:28.514: failed to read UpdateMotd key: Key file does not have key ?UpdateMotd? in group ?fwupd?
# FuConfig-DEBUG: failed to read EnumerateAllDevices key: Key file does not have key ?EnumerateAllDevices? in group ?fwupd?
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuConfig-DEBUG: 22:15:28.514: failed to read EnumerateAllDevices key: Key file does not have key ?EnumerateAllDevices? in group ?fwupd?
# GLib-GIO-DEBUG: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): GLib-GIO-DEBUG: 22:15:28.514: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
# Jcat-DEBUG: generating a 3072 bit RSA private key...
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): Jcat-DEBUG: 22:15:28.515: generating a 3072 bit RSA private key...
# Jcat-DEBUG: writing /tmp/fwupd-self-test/var/lib/fwupd/pki/secret.key with 2455 bytes
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): Jcat-DEBUG: 22:15:28.782: writing /tmp/fwupd-self-test/var/lib/fwupd/pki/secret.key with 2455 bytes
# Jcat-DEBUG: writing /tmp/fwupd-self-test/var/lib/fwupd/pki/client.pem with 1383 bytes
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): Jcat-DEBUG: 22:15:28.788: writing /tmp/fwupd-self-test/var/lib/fwupd/pki/client.pem with 1383 bytes
# FuEngine-DEBUG: client certificate exists and working
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuEngine-DEBUG: 22:15:28.794: client certificate exists and working
# FuHistory-DEBUG: trying to open database '/tmp/fwupd-self-test/var/lib/fwupd/pending.db'
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuHistory-DEBUG: 22:15:28.794: trying to open database '/tmp/fwupd-self-test/var/lib/fwupd/pending.db'
# FuHistory-DEBUG: no schema version: no such table: schema
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuHistory-DEBUG: 22:15:28.795: no schema version: no such table: schema
# FuHistory-DEBUG: got schema version of 0
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuHistory-DEBUG: 22:15:28.795: got schema version of 0
# FuHistory-DEBUG: building initial database
(/tmp/guix-build-fwupd-1.5.5.drv-0/build/src/fu-self-test:4118): FuHistory-DEBUG: 22:15:28.795: building initial database
[... 31 lines cut ...]
15/27 wacom-usb-self-test            OK              0.01s
16/27 vli-self-test                  OK              0.00s
17/27 ata-self-test                  OK              0.01s
18/27 logitech-hidpp-self-test       OK              0.00s
19/27 optionrom-self-test            OK              0.01s
20/27 tpm-eventlog-self-test         OK              0.01s
21/27 nvme-self-test                 OK              0.01s
22/27 thunderbolt-self-test          OK              3.24s
23/27 redfish-self-test              OK              0.01s
24/27 dell-self-test                 OK              0.01s
25/27 synaptics-mst-self-test        OK              0.01s
26/27 uefi-self-test                 OK              0.01s
27/27 uefi-dbx-self-test             OK              0.01s

Summary of Failures:

 7/27 fu-self-test                   FAIL            0.39s   killed by signal 6 SIGABRT

Ok:                 26
Expected Fail:      0
Fail:               1
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```

superm1, 2022-01-03:
Since you're porting - suggest you start with the latest release instead - 1.7.x..  If it's still present in 1.7.x we would want to fix it there first anyway before backporting to 1.5.x branch.

superm1, 2022-01-03:
BTW regarding your downstream patches, I only glanced through them but wanted to give you a note about it.

> gnu/packages/patches/fwupd-remove-nonfree-lvfs-repo.patch

Rather than fully remove the remotes, suggest that you instead just `s/Enabled/Disabled` in the conffile.  You'll have the same result that they're not used by default but it's **a lot easier** to re-enable should your users want to do so.

phodina, 2022-01-04:
Ok, I'll update to the latest 1.7.3 version and post the update here.

The patch has unfortunately it's reason as the GNU Guix won't accept nonfree firmware sources.

FIY I'll add another package which inherits this one and will add the nonfree sources. This will be upstreamed to [NonGuix channel](https://gitlab.com/nonguix/nonguix).

hughsie, 2022-01-04:
> it's reason as the GNU Guix won't accept nonfree firmware sources

It's probably better to do `-Dlvfs=disabled` in this case

phodina, 2022-01-04:
Thanks @hughsie for the tip! That's definitely better alternative than patch/substitution.

superm1, 2022-01-17:
leaving as needinfo pending checking if 1.7.4 or main affected

closed 2022-01-21 by hughsie