fwupd/fwupd #4142: dfu-tool replace regression 1.5.7 -> 1.7.3

mobilinkd, 2022-01-04:
**Describe the bug**
`dfu-tool replace-data` no longer works. Plus dfu-tool no longer reads ihex files, which makes creating bug reports more tedious (see #903 for an example where ihex format was helpful).

This is broken after an upgrade from Fedora 33 to Fedora 35, fwupd-1.5.7-1.fc33 -> fwupd-1.7.3-1.fc35.x86_64

```shell
$ dfu-tool replace-data test.dfu 0e2f 0e30
search string was not found
```

I use dfu-tool to maintain a file used for OTP data.

**Steps to Reproduce**
1. Create a test.dfu file with a specific hex value, such as 0xdeadbeef
2. Run `dfu-tool replace-data test.dfu 0xdeadbeef 0xd0d0d0d0`
3. Read error message "search string was not found"

**Expected behavior**
The behavior in 1.5.7 was to replace the string in the DFU file.

**fwupd version information**
fwupd-1.7.3-1.fc35.x86_64

```shell
$ fwupdmgr --version
client version:	1.7.3
compile-time dependency versions
	gusb:	0.3.9

daemon version:	1.7.3
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

*This is irrelevant to this bug report.*

**Additional questions**

- Operating system and version: Fedora 35
- Have you tried rebooting? No
- Is this a regression? Yes

Please also note the following when attempting to replace data in an ihex file.

```shell
$ dfu-tool replace-data test.ihex 00000001 00000002
no DFU signature

Usage:
...
```
 Per example in #903 this is a regression in behavior as well.  It has been a long time since I tried this with an ihex file.  I have no idea when this broke.

hughsie, 2022-01-04:
> I use dfu-tool to maintain a file used for OTP data.

Okay; lets try to understand the usecase in parallel to fixing the reported bug. Is the replacement always at the same offset, or it can occur anywhere in the file? I've always been hesitant to move the `replace-data` command to fwupdtool as it's a bit "greedy" in operation, especially for short bytestrings like `0e2f`.

mobilinkd, 2022-01-05:
The format is fixed, so it would be trivial to specify a fixed offset.  It is generally a 24- or 32-bit field. I used a 16-bit value as an example for the defect report.

mobilinkd, 2022-01-05:
Also , in `fwupdtool`, please omit the `This program may only work correctly as root`  warning when doing things like `firmware-convert` which do not require root.  It grinds my gears. :)

hughsie, 2022-01-05:
> please omit the This program may only work correctly as root

Does https://github.com/fwupd/fwupd/pull/4149 work as you expect?

closed 2022-01-06 by hughsie