fwupd/fwupd #4169: USB4 retimer nvm updates failed to install

krankun, 2022-01-12:
**Describe the bug**
retimer nvm updates could not be installed 

**Steps to Reproduce**

fwupdmgr install /home/root/retimer-2.21.cab --allow-reinstall --force --verbose

**Expected behavior**
nvm updates install fine

**fwupd version information**
1.7.2

```shell
fwupdmgr --version
client version:	1.7.2
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.7.2

- Is this a regression?
NO

krankun, 2022-01-12:
[debug_retimer_nvm_update_4169.txt](https://github.com/fwupd/fwupd/files/7856372/debug_retimer_nvm_update_4169.txt)

krankun, 2022-01-12:
export FWUPD_PROBE_VERBOSE=1 && export FU_UDEV_DEVICE_DEBUG=1
fwupdtool --verbose --plugins thunderbolt get-devices

[retimer_debug_bug_4169_verbose.txt](https://github.com/fwupd/fwupd/files/7856497/retimer_debug_bug_4169_verbose.txt)

krankun, 2022-01-12:
re: https://github.com/fwupd/fwupd/pull/4135#discussion_r781620029
@hughsie 
17:18:40:0025 FuPluginThunderbolt  missing nvm_version attribute
17:18:40:0025 FuPluginThunderbolt  failed to get Vendor ID
17:18:40:0025 FuPluginThunderbolt  failed to get Device ID
17:18:40:0025 FuPluginThunderbolt  Could not find non-volatile memory location
17:18:43:0577 FuPlugin             emit added from thunderbolt: 84301540cc951663d7c5dc652853e56ec4095925
is above for retimer it failed to get the vendorid etc ? the enumeration would take few seconds i guess after we echo 1 > usb4_port1/offline and rescan , once the retimer is enumerated in sysfs the nvmem would show up to0 in the retimer sysfs dir.

krankun, 2022-01-13:
[debug_log_fwupd_tool_retimer.txt](https://github.com/fwupd/fwupd/files/7860537/debug_log_fwupd_tool_retimer.txt)
after debugging the code it looks like the when "fwupdtool install <.cab> --allow-reinstall --plugins=thunderbolt --verbose" is launched

we have properly set up the retimer from the code flow below and the device is enumerated at sysfs
06:40:00:0933 FuPluginThunderbolt  fu_thunderbolt_controller_setup
06:40:00:0933 FuPluginThunderbolt  fu_thunderbolt_controller_setup_usb4
06:40:00:0933 FuPluginThunderbolt  fu_thunderbolt_udev_set_port_offline
06:40:04:0525 FuPluginThunderbolt  fu_thunderbolt_controller_setup PASSED....

however we are facing "first pass requirement on 2de6f8080e10007518d002394b68f0d1cf214890:com.retimer.firmware.test failed: No supported devices found" 
I have no clue what is wrong in here so far  CC : @hughsie

hughsie, 2022-01-13:
> I have no clue what is wrong in here 

This would be much easier to debug with actual hardware, or even SSH access into hardware. I think you need to use fwupd, i.e. the daemon rather than fwupdtool as a 5 second timeout is going to be expected *after* the fwupdtool process has already exited.

krankun, 2022-01-13:
> > I have no clue what is wrong in here
> 
> This would be much easier to debug with actual hardware, or even SSH access into hardware. I think you need to use fwupd, i.e. the daemon rather than fwupdtool as a 5 second timeout is going to be expected _after_ the fwupdtool process has already exited.
[demon_log_debug_fwupdmgr_4169.txt](https://github.com/fwupd/fwupd/files/7864216/demon_log_debug_fwupdmgr_4169.txt)
the demon log attached the retimer is probed and the port is back online from log I see retimer is auto-remove this should not be the case right ? I will be out few hours in first half today, I will arrange ssh connection tommrow morning time pst if needed. Thank you!

16:14:30:0734 FuPluginThunderbolt  fu_thunderbolt_controller_setup
16:14:30:0734 FuPluginThunderbolt  fu_thunderbolt_controller_setup_usb4
16:14:30:0734 FuPluginThunderbolt  fu_thunderbolt_udev_set_port_offline
16:14:34:0354 FuPluginThunderbolt  fu_thunderbolt_controller_setup PASSED....
16:14:34:0355 FuPlugin             emit added from thunderbolt: f54fdf34d09686ded67fa9b2a3d380060f0dc92e
16:14:34:0355 FuPlugin             fu_plugin_device_registered(dell_dock)
16:14:34:0355 FuPlugin             fu_plugin_device_registered(pci_bcr)
16:14:34:0355 FuPlugin             fu_plugin_device_registered(thunderbolt)
16:14:34:0356 FuPlugin             fu_plugin_device_registered(usi_dock)
16:14:34:0356 FuPlugin             fu_plugin_device_registered(msr)
16:14:34:0356 FuDeviceList         ::added f54fdf34d09686ded67fa9b2a3d380060f0dc92e
16:14:34:0356 FuEngine             Emitting PropertyChanged('Status'='idle')
16:14:34:0356 FuMain               Emitting PropertyChanged('Status'='idle')
16:14:34:0357 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
16:14:34:0357 GLib-GIO             Failed to initialize portal (GMemoryMonitorPortal) for gio-memory-monitor: Not using portals
16:14:34:0357 GLib-GIO             _g_io_module_get_default: Found default implementation dbus (GMemoryMonitorDBus) for ?gio-memory-monitor?
16:14:34:0360 FuMain               Daemon ready for requests (locale (NULL))
16:14:34:0363 FuPlugin             fu_plugin_device_created(thunderbolt)
16:14:34:0363 FuIdle               inhibiting: thunderbolt requires device wakeup
16:14:34:0363 FuIdle               inhibiting: thunderbolt requires device wakeup
16:14:34:0363 FuPluginThunderbolt  probe is called fu_thunderbolt_retimer_probe
16:14:34:0365 FuDevice             using 16d35c510618116e7899ae45e62605572872e713 for 0-0:1.1
16:14:34:0365 FuPlugin             emit added from thunderbolt: 16d35c510618116e7899ae45e62605572872e713
16:14:34:0366 FuPlugin             fu_plugin_device_registered(dell_dock)
16:14:34:0366 FuPlugin             fu_plugin_device_registered(pci_bcr)
16:14:34:0366 FuPlugin             fu_plugin_device_registered(thunderbolt)
16:14:34:0366 FuPlugin             fu_plugin_device_registered(usi_dock)
16:14:34:0366 FuPlugin             fu_plugin_device_registered(msr)
16:14:34:0366 FuDeviceList         ::added 16d35c510618116e7899ae45e62605572872e713
16:14:34:0366 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
16:14:34:0367 FuMain               acquired name: org.freedesktop.fwupd
16:14:39:0770 FuPluginThunderbolt  fu_thunderbolt_controller_set_port_online_cb
16:14:39:0770 FuPluginThunderbolt  fu_thunderbolt_udev_set_port_online
16:14:39:0871 FuContext            battery state now discharging
16:14:39:0890 FuEngine             auto-removing backend device USB4 Retimer [16d35c510618116e7899ae45e62605572872e713]
16:14:39:0890 FuDeviceList         ::removed 16d35c510618116e7899ae45e62605572872e713
16:14:39:0890 FuPlugin             backend_device_removed(dell_dock)

krankun, 2022-01-14:
the PR has fixed the issue https://github.com/fwupd/fwupd/pull/4171

krankun, 2022-01-15:
[91-fwupdtool-retimer.txt](https://github.com/fwupd/fwupd/files/7875838/91-fwupdtool-retimer.txt)

Auto update looks like still an issue , lets reopen this bug  Please to continue debug tracking 

on chrome:

What steps will reproduce the problem?
1.place attached signed .cab files in /usr/share/fwupd/remotes.d/vendor/firmware
2.place the attached 91-fwupdtool-retimer.rules for redrix in /lib/udev/rules.d/

example at boot process update command :
Process '/sbin/initctl emit fwupdtool-update 'PLUGIN=thunderbolt' GUID=<guid of retimer>

krankun, 2022-01-15:
ccing @djcampello  to explain auto update usecase better, Daniel is there a way we can simulate auto update with fwupdtool post update for debugging purposes , understanding the flow for auto update would help debug the issue in right way 

currently 
1) get-devices works
2) run demon / run fwupdmgr install works
3) auto update fails 
Any other use cases to validate ?

hughsie, 2022-01-15:
I don't think this *can* work with fwupdtool; it relies on a idle runout and udev hotplug.

djcampello, 2022-01-19:
Copy/pasting here from other post:

Auto-update is handled by a script `fwupd-at-boot.sh` https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/third_party/chromiumos-overlay/sys-apps/fwupd/files/fwupd-at-boot.sh

The above script basically looks for pending updates marked from the previous boot and for each of them it emits a fwupdtool-update event (legacy name - we are using fwupdmgr for thunderbolt updates) which is handled by this script: https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/third_party/chromiumos-overlay/sys-apps/fwupd/files/fwupdtool-update.conf

> At the moment of the execution of the script we expect the daemon to be already running as it is started during boot via a different upstart script https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/third_party/chromiumos-overlay/sys-apps/fwupd/files/fwupd.conf but if that is not the case we relay on dbus activation for the daemon to start. 
In the specific case of the auto-update the fwupd-at-boot.sh might be running with fwupd.conf to start the daemon during boot so it might more likely than not that it is activated via dbus.