fwupd/fwupd #4176: TSME check on Ryzen CPU returns wrong result

hardfalcon, 2022-01-15:
On a system with a Ryzen 5 PRO 3600, `fwupdmgr security --force` claims that `Encrypted RAM` is `Enabled`, while [AMD's own TSME test kernel module](https://github.com/AMDESE/mem-encryption-tests) shows TSME is disabled.

Sadly I currently don't have access the BIOS of this system, but I know that the motherboard used in this system has TSME disabled by default, but I recall having tested this on an identical machine a few weeks ago, and there `fwupdmgr security --force` would also always claim that memory encryption was enabled regardless of whether TSME was enabled or disabled in the BIOS setup.

hughsie, 2022-01-15:
@superm1 can you throw any light on this please?

superm1, 2022-01-19:
That test that we have in there today - I don't know where it actually came from.  I do think it's better to get this type of information from the kernel..

superm1, 2022-01-19:
And FWIW I have a a 4750U in front of me that it does show disabled.  I guess we should cross reference the stuff the test we have now does against the BKDG or PPR for the 3600 and see if it makes sense.  If not, then we should set some minimum CPU families.

superm1, 2022-01-20:
Here is the relevant documentation: https://developer.amd.com/sev/

superm1, 2022-01-20:
Looking through [the kernel documentation](https://www.kernel.org/doc/html/v5.16/x86/amd-memory-encryption.html) on SME I think some of the MSR plugin code could use a little clarification (SME and SEV are intermingled). 

I believe that fwupd is doing the right thing though for reading if SME is active.  Now the part that I think probably needs clarification is whether TSME or OS activated SME are active.  Because TSME can be turned off, but bit 23 can be set by BIOS and if you have `CONFIG_AMD_MEM_ENCRYPT_ACTIVE_BY_DEFAULT` enabled then the Linux kernel will have turned on SME.

That might be what's going on for you.

superm1, 2022-01-20:
Can you please have a check with main again?

superm1, 2022-01-20:
I had some discussion on this - that kernel module you linked also can't detect it unless OEM has set the SME enable bit in the MSR, which at least on client parts isn't very commonly done..
So the check that's in fwupd is for **SME**, _not_ **TSME**.  The check is renamed to SME as a result.  TSME detection is not currently possible.

hardfalcon, 2022-01-22:
Thanks for implementing this, even though I'm now even *more* confused than before. :D

Until two or three days ago, I assumed Ryzen Pro CPUs only had support for TSME but not for "normal" SME. On the machine in question, with Archlinux kernel 5.16.2-arch1-1, I see the following behaviour:
- when booting with `mem_encrypt=on`:
  - `dmesg`:
    - `Unknown kernel command line parameters "BOOT_IMAGE=/vmlinuz-linux mem_encrypt=on", will be passed to user space.`
    - `AMD Memory Encryption Features active: SME`
  - `fwupdmgr security --force`:
    - `Encrypted RAM: Unencrypted`
- when booting with `mem_encrypt=off`:
  - `dmesg`:
    - `Unknown kernel command line parameters "BOOT_IMAGE=/vmlinuz-linux mem_encrypt=off", will be passed to user space.`
    - No message about `AMD Memory Encryption Features` like above
  - `fwupdmgr security --force`:
    - `Encrypted RAM: Unencrypted`

superm1, 2022-01-23:
TSME and SME can both be active at the same time.  There's not a lot of a point to turning them both on simultaneously.  There is an impact to the amount of memory available to the system because of the C bit for SME though.  Like userspace, the kernel also doesn't know when TSME is active.  So if the OEM has set the bit that allows the kernel to turn on SME then `mem_encrypt=on` or `CONFIG_AMD_MEM_ENCRYPT_ACTIVE_BY_DEFAULT` will enable SME (as you observed in your `dmesg`).

With `Encrypted RAM: Unencrypted` in the SME case - is that fwupd 1.7.4 or main?

hardfalcon, 2022-01-23:
I don't have access to the BIOS setup of the machine, but I know that by default, TSME is disabled on this mainboard, and I have no reason to belive it has been enabled.

`Encrypted RAM: Unencrypted` was with fwupdmgr [1.7.4-32-g7fdf7c60e](https://github.com/fwupd/fwupd/tree/7fdf7c60ee6846e814bfb6e78b39edfb3640f1ac), which is the current HEAD of the main branch.

superm1, 2022-01-23:
Can you please share the output for

# fwupdtool security --force --verbose

hardfalcon, 2022-01-23:
Here you go, once with `mem_encrypt=off` and once with `mem_encrypt=on`:
[mem_encrypt_off.txt](https://github.com/fwupd/fwupd/files/7921064/mem_encrypt_off.txt)
[mem_encrypt_on.txt](https://github.com/fwupd/fwupd/files/7921065/mem_encrypt_on.txt)

superm1, 2022-02-01:
Looking at the kernel code some more, I think the documentation and the code don't match for SME.  The documentation makes it sounds that SME and SEV bits need to be at the same time, but the code shows that SME and SEV will be mutually exclusive.
There is a mask for SME to show the page reduction.  Can you please apply https://github.com/fwupd/fwupd/pull/4241 and share the debug output for your two cases again?  We can confirm if that mask is set at all times or only when SME is active.

If it's set at all times, I think we need to pull this code out of fwupd, we'll need to wait until a proper interface from the kernel to show SME status.

superm1, 2022-02-02:
I've checked a variety of hardware I have, and it does not set SME is enabled so I can't reproduce your result (or set `mem_encrypt=on` for that matter).  So I'll need to see your logs again in those two cases to make more progress.
This is what I see thus far:
```
04:20:41:0175 FuPluginMsr          SME/SEV check MSR: eax 01780f, ebx 00
04:20:41:0184 FuPluginMsr          PCI_MSR_AMD64_SYSCFG: 0740000, sme_is_enabled=0
04:20:41:0184 FuPluginMsr          PCI_MSR_AMD64_SEV: 00, sev_is_enabled=0
```

commit 783e2e04 (2022-02-04): Drop the AMD SME check for security attributes (Fixes: #4176)

This method does not indicate whether the kernel has enabled encryption.
That information is only visible from the kernel log at this time.

Continue to read the MSRs though, because if/when we get a method for the
kernel to export it we may use these as a basis of "support" but not a
indication of it being enabled.

encrypted:
```FuPluginMsr          SME/SEV check MSR: eax 01000f, ebx 016f
FuPluginMsr          PCI_MSR_AMD64_SYSCFG: 0f40000, sme_is_enabled=1
FuPluginMsr          PCI_MSR_AMD64_SEV: 00, sev_is_enabled=0
```

not_encr [... cut]
closed 2022-02-04 by superm1