fwupd/fwupd #4180: Vaydeer Multimedia Console doesn't work when fwupd.service is running

rjahanbakhshi, 2022-01-16:
**Describe the bug**
If I connect my Vaydeer Multimedia Console when fwupd.service is running, the device driver is correctly bound and immediately removed. If fwupd.service is not running, the driver stays bound and the device works correctly. 
Here's the systemd journal output when the problem occurs:
```Jan 16 13:03:54 rj-pc kernel: usb 1-6: new full-speed USB device number 11 using xhci_hcd
Jan 16 13:03:55 rj-pc kernel: usb 1-6: New USB device found, idVendor=0483, idProduct=5750, bcdDevice= 2.00
Jan 16 13:03:55 rj-pc kernel: usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 16 13:03:55 rj-pc kernel: usb 1-6: Product: Vaydeer Multimedia Console
Jan 16 13:03:55 rj-pc kernel: usb 1-6: Manufacturer: Vaydeer
Jan 16 13:03:55 rj-pc kernel: usb 1-6: SerialNumber: 4EA2335F5000
Jan 16 13:03:55 rj-pc kernel: input: Vaydeer Vaydeer Multimedia Console Keyboard as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001D/input/input50
Jan 16 13:03:55 rj-pc mtp-probe[456400]: checking bus 1, device 11: "/sys/devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6"
Jan 16 13:03:55 rj-pc kernel: hid-generic 0003:0483:5750.001D: input,hidraw19: USB HID v1.11 Keyboard [Vaydeer Vaydeer Multimedia Console] on usb-0000:2a:00.1-6/input0
Jan 16 13:03:55 rj-pc mtp-probe[456400]: bus: 1, device: 11 was not an MTP device
Jan 16 13:03:55 rj-pc fwupd[456230]: 12:03:55:0481 FuEngine             failed to add device usb:01:00:06: failed to send to device on ep 0x01: endpoint stalled or request not supported
Jan 16 13:03:55 rj-pc systemd-logind[981]: Failed to open /dev/input/event263: No such file or directory
Jan 16 13:03:55 rj-pc mtp-probe[456420]: checking bus 1, device 11: "/sys/devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6"
Jan 16 13:03:55 rj-pc mtp-probe[456420]: bus: 1, device: 11 was not an MTP device
```
udev monitor output:
```
KERNEL[87182.589726] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6 (usb)
KERNEL[87182.603621] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0 (usb)
KERNEL[87182.611642] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B (hid)
KERNEL[87182.611711] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48 (input)
KERNEL[87182.667398] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::numlock (leds)
KERNEL[87182.667424] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::numlock (leds)
KERNEL[87182.667430] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::capslock (leds)
KERNEL[87182.667435] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::capslock (leds)
KERNEL[87182.667440] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::scrolllock (leds)
KERNEL[87182.667449] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::scrolllock (leds)
KERNEL[87182.667454] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::compose (leds)
KERNEL[87182.667459] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::kana (leds)
KERNEL[87182.667466] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::kana (leds)
KERNEL[87182.667474] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/event263 (input)
KERNEL[87182.667497] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/hidraw/hidraw19 (hidraw)
KERNEL[87182.667525] bind     /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B (hid)
KERNEL[87182.667537] bind     /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0 (usb)
KERNEL[87182.667558] bind     /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6 (usb)
UDEV  [87182.681236] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6 (usb)
UDEV  [87182.682419] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0 (usb)
UDEV  [87182.683178] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B (hid)
UDEV  [87182.684367] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48 (input)
UDEV  [87182.685787] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::scrolllock (leds)
UDEV  [87182.685798] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::capslock (leds)
UDEV  [87182.685807] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::numlock (leds)
UDEV  [87182.686554] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::capslock (leds)
UDEV  [87182.686563] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::kana (leds)
UDEV  [87182.686673] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::compose (leds)
UDEV  [87182.686701] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::numlock (leds)
UDEV  [87182.686840] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::scrolllock (leds)
UDEV  [87182.687187] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::kana (leds)
UDEV  [87182.687363] add      /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/hidraw/hidraw19 (hidraw)
KERNEL[87182.847467] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::numlock (leds)
KERNEL[87182.847492] remove   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::numlock (leds)
KERNEL[87182.887583] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::capslock (leds)
KERNEL[87182.887599] remove   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::capslock (leds)
KERNEL[87182.914181] change   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::scrolllock (leds)
KERNEL[87182.914194] remove   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::scrolllock (leds)
KERNEL[87182.914208] remove   /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.1/usb1/1-6/1-6:1.0/0003:0483:5750.001B/input/input48/input48::compose (leds)
[... cut ...]
      Vendor:             Western Digital (ATA:0x101C, OUI:0014ee)
      GUIDs:              64ab0c82-9dc1-5abb-8c35-42e5a4a66d79
                          bafc7053-ac67-53d0-95e5-908cea0f11a3
                          cd0a921e-7b2c-5687-b49f-ea3077a42020
      Device Flags:       • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
```

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2022-01-16:
> Jan 16 13:03:55 rj-pc kernel: usb 1-6: New USB device found, idVendor=0483, idProduct=5750, bcdDevice= 2.00
> Jan 16 13:03:55 rj-pc kernel: usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> Jan 16 13:03:55 rj-pc kernel: usb 1-6: Product: Vaydeer Multimedia Console

It might say _Vaydeer_, but it's using the Vendor ID assigned to STMicroelectronics... more specifically a _mini LED display_: https://devicehunt.com/view/type/usb/vendor/0483/device/5750

rjahanbakhshi, 2022-01-17:
@hughsie, 
Thanks for your comment. True, the device is using an incorrect vendor and device ID. Is it technically possible not to cause the driver unbinding when the operation is not supported by the endpoint? I'm asking because, without fwupd, the device is working perfectly fine. If it's not possible, is there any configuration for fwupd for ignoring a specific device?

closed 2022-01-17 by hughsie

m4gr4th34, 2022-12-10:
I have the exact same issue with the exact same device on debian 11 testing.  however, even when i stop and disable fwupd.service the multimedia controller still does not function