fwupd/fwupd #4187: fwupdtool auto updates for retimer nvm update

krankun, 2022-01-17:
**Describe the bug**

filing the issue to understand the auto update flow for retimer nvm update , seems there is disconnect on how the retimer nvm updates should be hosted at boot time.

What steps will reproduce the problem?
1.place signed .cab files in /usr/share/fwupd/remotes.d/vendor/firmware  { vendor pushed new nvm updates for retimer } 
2.place the attached 91-fwupdtool-retimer.rules for in /lib/udev/rules.d/

# udev rules to start firmware update on the retimer.
SUBSYSTEM=="thunderbolt", ENV{DEVTYPE}=="thunderbolt_retimer", \
ATTR{vendor}=="xxxx", ATTR{device}=="xxxx", ACTION=="add", \
RUN+="/sbin/initctl emit fwupdtool-update 'PLUGIN=thunderbolt' GUID=<GUID of retimer>"

What is the expected output?
the first time you reboot the rule will trigger and will mark the retimer for pending update by creating a file under /var/lib/fwupd/pending and second reboot will trigger the auto update

What do you see instead?
errors emitting the auto update , attached dmesg logs
 3.007532] init: fwupdtool-update (xxxxxxxx-false) pre-start process (298) terminated with status 1
[    3.007793] udevd[268]: Process '/sbin/initctl emit fwupdtool-update 'PLUGIN=thunderbolt' GUID=xxxxxx' failed with exit code 1.

How frequently does this problem reproduce? (Always, sometimes, hard to
reproduce?)
always

What is the impact to the user, and is there a workaround? If so, what is
it?
cannot update nvm for BB retimer at the boot time.

**Expected behavior**
A clear and concise description of what you expected to happen.

fwupdmgr get-devices --show-all-devices
```
localhost / # fwupdmgr get-devices
Voxel
?
??USB4 host controller:
      Device ID:          881badd61549eb2f3927991ccc3aeaa35d07c3a8
      Summary:            Unmatched performance for high-speed I/O
      Update Error:       No vendor ID set
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ? TBT-fixed
      Device Flags:       ? Internal device
                          ? System requires external power source
                          ? Updatable
    
localhost / # fwupdmgr get-updates
Voxel
?
??USB4 Retimer:
  ?   Device ID:          16d35c510618116e7899ae45e62605572872e713
  ?   Summary:            A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
  ?   Current version:    19.00
  ?   Vendor:             Intel (THUNDERBOLT:0x8087)
  ?   GUIDs:              f00257c8-0e09-5a32-bbb3-f49748ae6cc5 ? THUNDERBOLT\VEN_8087&DEV_15EE&REV_00
  ?                       e6f2e4ac-f76a-5fc4-b36a-6f9cb548689e ? THUNDERBOLT\VEN_8087&DEV_15EE
  ?                       6d71ac25-b368-5e7b-ac62-32df365fa640 ? TBT-808715ee-retimer0-0:1.1
  ?   Device Flags:       ? Internal device
  ?                       ? Updatable
  ?                       ? Supported on remote server
  ?                       ? Device stages updates
  ? 
  ??retimer:
        New version:      21.00
        Remote ID:        vendor-directory
        Summary:          Firmware for the thunderbolt retimer
        License:          Proprietary
        Urgency:          High
        Flags:            is-upgrade
        Description:      
        Release version: 21.00

- Is this a regression?
NO

logs will be provided  once we agree upon flows, ccing @hughsie  @djcampello @superm1

krankun, 2022-01-17:
@hughsie @djcampello @superm1  here is the new issue for auto update cases for retimer nvm updates

hughsie, 2022-01-18:
Can you share what fwupdtool-update actually does? I think that's ChromeOS specific perhaps.

krankun, 2022-01-18:
[auto-update_demon_log.txt](https://github.com/fwupd/fwupd/files/7890791/auto-update_demon_log.txt)

as we can see above output of fwupdmgr get-updates in issue description I have staged updates  for new nvm version.

at the boot time the upstart job should run the demon and install nvm "fwupdmgr  update <GUID>" 

while in this situation after boot I tried to demon { attached is log }, why the demon exits and complains "another service has claimed the dbus name org.freedesktop.fwupd" ?

hughsie, 2022-01-19:
> another service has claimed the dbus name org.freedesktop.fwupd

That just means the daemon is already running. Maybe try `sudo killall fwupd` and try again? I'm not really sure what @djcampello plan is for the daemon mode in ChromeOS right now; the long term plan is just to move everything away from `fwupdtool` I think.

djcampello, 2022-01-19:
The name fwupdtool-update is mostly legacy. Here is the source: https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/third_party/chromiumos-overlay/sys-apps/fwupd/files/fwupdtool-update.conf

It is an upstart script that handles the initctl event "fwupdtool-update" that can be emitted anywhere and at any time in the system (typically a udev rule firing or the startup script: https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/third_party/chromiumos-overlay/sys-apps/fwupd/files/fwupd-at-boot.sh )

The script does some sanity checking on the arguments passed with the event, and sets up a minijail to run fwupdmgr within.

At the moment of the execution of the script we expect the daemon to be already running as it is started during boot via a different upstart script https://source.chromium.org/chromiumos/chromiumos/codesearch/+/main:src/third_party/chromiumos-overlay/sys-apps/fwupd/files/fwupd.conf but if that is not the case we relay on dbus activation for the daemon to start.

djcampello, 2022-01-19:
@krankun as suggested by @hughsie if by any reason you need to run your own instance of the daemon you can just do `stop fwupd` to kill the system instance.

krankun, 2022-01-19:
@hughsie  @djcampello  
currently once the pending updates are hosted and after the boot I see the splash screen that "fw update is in progress" but after few seconds we don't get the nvm updated.

later I "stop fwupd && kilall fwupd" after boot and run the demon but it still complains "another service has claimed the dbus name org.freedesktop.fwupd"

I have a suspicion that in flow we are writing echo 1 > usb4_port1/offline twice { especially in auto update case) ,  doing so is disconnecting retimer (one way to avoid that is to check if the port is online (0) then only offline(1) ) , 

Also debugging auto update flow is little tricky , is there a similar case for matching flow post update that would help me in debug.

[   38.878972] thunderbolt 0-0:1.1: GPIO lookup for consumer wp
[   38.878974] thunderbolt 0-0:1.1: using lookup tables for GPIO lookup
[   38.878975] thunderbolt 0-0:1.1: No GPIO consumer wp found
[   38.879017] thunderbolt 0-0:1.1: GPIO lookup for consumer wp
[   38.879017] thunderbolt 0-0:1.1: using lookup tables for GPIO lookup
[   38.879018] thunderbolt 0-0:1.1: No GPIO consumer wp found
[   38.879035] thunderbolt 0-0:1.1: new retimer found, vendor=0x8087 device=0x15ee
[   44.100909] thunderbolt 0-0:1.1: retimer disconnected
[   44.109898] udevd[3970]: Process '/bin/chown -R fwupd:fwupd /sys//devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1/nvm_non_active0' failed with exit code 1.
[   44.109974] udevd[3971]: Process '/bin/chown -R fwupd:fwupd /sys//devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1/nvm_active0' failed with exit code 1.
[   44.157799] udevd[3970]: Process '/bin/chown -R fwupd:fwupd /sys//devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1' failed with exit code 1.

djcampello, 2022-01-19:
> later I "stop fwupd && kilall fwupd" after boot and run the demon but it still complains "another service has claimed the dbus name org.freedesktop.fwupd"

are you running the daemon with `start fwupd` or within the minijail on the terminal? the dbus interface is configured to be owned by the `fwupd` user, running as root would not let the daemon bind to it.

krankun, 2022-01-19:
> are you running the daemon with `start fwupd` or within the minijail on the terminal? the dbus interface is configured to be owned by the `fwupd` user, running as root would not let the daemon bind [...]
i was running minijail

krankun, 2022-01-19:
start fwupd , seems work and not minijail 

start fwupd after boot log 
[ 3141.462189] thunderbolt 0-0:1.1: GPIO lookup for consumer wp
[ 3141.462192] thunderbolt 0-0:1.1: using lookup tables for GPIO lookup
[ 3141.462194] thunderbolt 0-0:1.1: No GPIO consumer wp found
[ 3141.462257] thunderbolt 0-0:1.1: GPIO lookup for consumer wp
[ 3141.462258] thunderbolt 0-0:1.1: using lookup tables for GPIO lookup
[ 3141.462259] thunderbolt 0-0:1.1: No GPIO consumer wp found
[ 3141.462281] thunderbolt 0-0:1.1: new retimer found, vendor=0x8087 device=0x15ee
[ 3141.485261] init: fwupdtool-update (e6f2e4ac-f76a-5fc4-b36a-6f9cb548689e-false) pre-start process (13093) terminated with status 1
[ 3141.499442] udevd[13090]: Process '/sbin/initctl emit fwupdtool-update 'PLUGIN=thunderbolt' GUID=e6f2e4ac-f76a-5fc4-b36a-6f9cb548689e' failed with exit code 1.
[ 3147.106015] thunderbolt 0-0:1.1: retimer disconnected
[ 3147.115157] udevd[13131]: Process '/bin/chown -R fwupd:fwupd /sys//devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1/nvm_non_active0' failed with exit code 1.
[ 3147.115249] udevd[13132]: Process '/bin/chown -R fwupd:fwupd /sys//devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1/nvm_active0' failed with exit code 1.
[ 3147.167543] udevd[13131]: Process '/bin/chown -R fwupd:fwupd /sys//devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1' failed with exit code 1.

djcampello, 2022-01-19:
is your minijail invocation complete (compare it to the contents of fwupd.conf), it might be missing the `-u fwupd`

krankun, 2022-01-19:
> is your minijail invocation complete (compare it to the contents of fwupd.conf), it might be missing the `-u fwupd`

minijail0     --uts -l -p -N     -v -P /mnt/empty -b / -b /proc -t -r -b /dev,,1 -b /sys,,1     -k /var,/var,tmpfs -b /var/cache/fwupd,,1 -b /var/lib/fwupd,,1     -k run,/run,tmpfs -b /run/dbus -b /run/lock,,1 -b /run/shill     -b /run/udev/data,,1 -u fwupd -g fwupd -G -c cap_sys_rawio+e     -- /usr/libexec/fwupd/fwupd -v --plugin-verbose=thunderbolt

krankun, 2022-01-20:
Hi , I will be on short vacation and will be back 1/24 to resume debug on this. Thank you!

krankun, 2022-01-25:
[log.txt](https://github.com/fwupd/fwupd/files/7935698/log.txt)

the auto update is triggered at boot around 9 secs 
[    9.814759] chromeos-boot-alert: update_fwupd_firmware

and looks like demon is launched bit later 

[   37.613890] thunderbolt 0-0:1.1: GPIO lookup for consumer wp
[   37.613893] thunderbolt 0-0:1.1: using lookup tables for GPIO lookup
[   37.613895] thunderbolt 0-0:1.1: No GPIO consumer wp found
[   37.613958] thunderbolt 0-0:1.1: GPIO lookup for consumer wp
[   37.613959] thunderbolt 0-0:1.1: using lookup tables for GPIO lookup
[   37.613960] thunderbolt 0-0:1.1: No GPIO consumer wp found
[   37.613989] thunderbolt 0-0:1.1: new retimer found, vendor=0x8087 device=0x15ee
[   43.103222] thunderbolt 0-0:1.1: retimer disconnected
[   43.111698] udevd[3957]: Process '/bin/chown -R fwupd:fwupd /sys//devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1/nvm_non_active0' failed with exit code 1.
[   43.111780] udevd[3958]: Process '/bin/chown -R fwupd:fwupd /sys//devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1/nvm_active0' failed with exit code 1.
[   43.153783] udevd[3965]: Process '/bin/chown -R fwupd:fwupd /sys//devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1' failed with exit code 1.

i dont see /etc/fwupd/fwupd.conf in my chrome os build , instead /etc/fwupd/demon.conf ,

djcampello, 2022-01-25:
> and looks like demon is launched bit later
>
>[ 37.613890] thunderbolt 0-0:1.1

This looks like the thunderbolt device is enumerated late. Do we know if the daemon is actually not running? 
You could look at this instead:

```
localhost ~ # grep fwupd /var/log/messages
...
2022-01-25T18:05:02.120013Z INFO dbus-daemon[850]: [system] Activating service name='org.freedesktop.fwupd' requested by ':1.50' (uid=20130 pid=1261 comm="/usr/bin/fwupdmgr get-plugins " label="u:r:minijailed:s0") (using servicehelper)
2022-01-25T18:05:03.863747Z NOTICE boot-update-firmware[1808]: Update fwupd firmware.
2022-01-25T18:05:05.589999Z INFO dbus-daemon[850]: [system] Successfully activated service 'org.freedesktop.fwupd'
```

> i dont see /etc/fwupd/fwupd.conf in my chrome os build , instead /etc/fwupd/demon.conf

that would be /etc/init/fwupd.conf

krankun, 2022-01-25:
localhost ~ #  grep fwupd /var/log/messages
2022-01-25T19:41:46.231828Z NOTICE boot-update-firmware[1675]: Update fwupd firmware.
grep: /var/log/messages: binary file matches

some thing is strange still : 
auto update is triggered if I do 
localhost ~ # echo  1 > /sys/bus/thunderbolt/devices/0-0/usb4_port1/offline 
localhost ~ # echo  1 > /sys/bus/thunderbolt/devices/0-0/usb4_port1/rescan  
and reboot 

the normal reboot wont trigger auto update when retimer is NOT enumerated in sysfs , this looks like demon launch at boot configuration.

krankun, 2022-02-02:
@hughsie @djcampello @superm1 

testing on the latest source tree { fwupd main )
9de6adf41fcc (HEAD -> main_fwupd_feb_2, fwupd_main/main) trivial: msr: add more debugging data for register status
f3d71a18a504 redfish: Be more robust by retrying IPMI transactions

fwupdtool --version
client version: 1.7.5
compile-time dependency versions
        gusb:   0.3.5

as per the Richard  there is a new API to quit the service for demon
gdbus call --system --dest org.freedesktop.fwupd --object-path / --method org.freedesktop.fwupd.Quit

later after we start the demon
/usr/libexec/fwupd/fwupd --verbose  
we always encounter the error " another service has claimed the dbus name org.freedesktop.fwupd"

As per Daniel "dbus interface is configured to be owned by the fwupd user, running as root would not let the daemon bind to it"  ,

Q: how to configure the demon to be owner bu user on chrome , I tried to follow https://support.google.com/chrome/a/answer/6220366?hl=en  but that requires a .google.com admin account

later the normal update doesnt work as the fwupdmgr doesnt have permissions for the retimer directory

fwupdmgr install /home/root/retimer-2.21.cab
Decompressing?           [***************************************]
Authenticating?          [ -                                     ]
Writing?                 [          |                            ]could not write firmware to thunderbolt device at /sys/devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1: Error opening file ?/sys/devices/pci0000:00/0000:00:0d.2/domain0/0-0/usb4_port1/0-0:1.1/nvm_non_active0/nvmem?: Permission denied

krankun, 2022-02-14:
most of the auto update fixes has been identified, there is still one issue left where we end up in "ACPI: set online state _DSM failed with error -1" after nvm has been updated and we bring port back online, 

root causing this has been identified that after nvm is written to retimer we need to wait few seconds before the port is retrieved back online ( echo 0 > offline) 

the possible fix could be make fu_thunderbolt_controller_set_port_online_cb common and call this from fu_plugin_thunderbolt_composite_cleanup->fu_thunderbolt_retimer_set_parent_port_online in fu-plugin-thunderbolt.c

cc: @hughsie @djcampello @superm1

hughsie, 2022-02-15:
> retimer we need to wait few seconds

Although it feels like a workaround, we could just add a g_usleep() in the fwupd thunderbolt plugin.

superm1, 2022-02-15:
Does the kernel emit any event during this time you can go on?

krankun, 2022-02-15:
> Does the kernel emit any event during this time you can go on?

@superm1  looks like there are some kernel events which are emitted and I notified Raj he will respond with more details on this thread , probably I will look in enhancing this future development , 
I fixed other comments in https://github.com/fwupd/fwupd/pull/4286/commits/28eea1023a4b052dbfdeb9b77ecc4d74e01dde3c

djcampello, 2022-02-15:
Sorry for being late... These is in fact a workaround. I don;t think we should do this... If anything the kernel interface should block the write operation until it can process it instead of returning an I/O error and falling on an irrecoverable state. This is not a fwupd bug.

closed 2022-03-17 by krankun