fwupd/fwupd #4190: (Plugin Modem Manager) `fu_io_channel_new_file: assertion 'filename != NULL' failed` (SIGSEGV)

j5k, 2022-01-18:
Can't run `fwupd` because of **SIGSEGV**:

**Service status:**
``` shell
× fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: failed (Result: core-dump) since Tue 2022-01-18 21:33:39 CET; 5min ago
       Docs: https://fwupd.org/
    Process: 6127 ExecStart=/usr/lib/fwupd/fwupd (code=dumped, signal=SEGV)
   Main PID: 6127 (code=dumped, signal=SEGV)
        CPU: 93ms

Jan 18 21:33:39 xxx systemd[1]: Starting Firmware update daemon...
Jan 18 21:33:39 xxx fwupd[6127]: 20:33:39:0822 FuIOChannel          fu_io_channel_new_file: assertion 'filename != NULL' failed
Jan 18 21:33:39 xxx systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Jan 18 21:33:39 xxx systemd[1]: fwupd.service: Failed with result 'core-dump'.
Jan 18 21:33:39 xxx systemd[1]: Failed to start Firmware update daemon.
```
**gdb:**
``` shell
gdb --args /usr/lib/fwupd/fwupd --verbose
...
(gdb) r
...
20:41:25:0451 FuDeviceList         ::added 4bde70ba4e39b28f9eab1628f9dd6e6244c03027
20:41:25:0451 FuPlugin             coldplug(modem_manager)
20:41:25:0452 FuPluginMm           ModemManager 1.18.4-1 is available
20:41:25:0452 FuPluginMm           added modem: /org/freedesktop/ModemManager1/Modem/0
20:41:25:0452 FuPluginMm           valid firmware update combination: mbim-qdu
20:41:25:0454 FuIOChannel          fu_io_channel_new_file: assertion 'filename != NULL' failed

Thread 1 "fwupd" received signal SIGSEGV, Segmentation fault.
0x00007fffefdd2d1b in ?? () from /usr/lib/fwupd-plugins-5/libfu_plugin_modem_manager.so
(gdb) bt
#0  0x00007fffefdd2d1b in  () at /usr/lib/fwupd-plugins-5/libfu_plugin_modem_manager.so
#1  0x00007fffefdd2ed1 in  () at /usr/lib/fwupd-plugins-5/libfu_plugin_modem_manager.so
#2  0x00007fffefdd3043 in  () at /usr/lib/fwupd-plugins-5/libfu_plugin_modem_manager.so
#3  0x00007ffff7f094dc in fu_plugin_runner_coldplug () at /usr/lib/libfwupdplugin.so.5
#4  0x000055555558cf3a in fu_engine_load ()
#5  0x000055555556adfc in main ()
(gdb) 
```
**Spec:**
```
fwupd:  1.7.4
Arch Linux 5.15.15-1-lts
ModemManager 1.18.4-1

Lenovo Thinkpad X1 Gen 9
Foxconn Qualcomm Snapdragon X55 5G modem
```

hughsie, 2022-01-18:
Can you get a backtrace with debug symbols installed please

benlorenz, 2022-01-19:
I got a similar crash on gentoo with fwupd 1.7.4 and ModemManager 1.18.4, using a Snapdragon X55 Modem.

The assert triggers here:
```
10:47:10:0160 FuPlugin             coldplug(modem_manager)
10:47:10:0161 FuPluginMm           ModemManager 1.18.4 is available
10:47:10:0161 FuPluginMm           added modem: /org/freedesktop/ModemManager1/Modem/2
10:47:10:0161 FuPluginMm           valid firmware update combination: mbim-qdu

Thread 1 "fwupd" hit Breakpoint 2, fu_io_channel_new_file (filename=0x0, error=0x7fffffffda30) at ../fwupd-1.7.4/libfwupdplugin/fu-io-channel.c:514
514		g_return_val_if_fail(filename != NULL, NULL);
(gdb) bt
#0  fu_io_channel_new_file (filename=0x0, error=0x7fffffffda30) at ../fwupd-1.7.4/libfwupdplugin/fu-io-channel.c:514
#1  0x00007ffff1497b9d in fu_mm_device_io_open (self=self@entry=0x5555557542a0, error=error@entry=0x7fffffffda30) at ../fwupd-1.7.4/plugins/modem-manager/fu-mm-device.c:699
#2  0x00007ffff7efb283 in fu_device_locker_new_full
    (device=device@entry=0x5555557542a0, open_func=open_func@entry=0x7ffff1497b90 <fu_mm_device_io_open>, close_func=close_func@entry=0x7ffff1497c20 <fu_mm_device_io_close>, error=error@entry=0x7fffffffda30)
    at ../fwupd-1.7.4/libfwupdplugin/fu-device-locker.c:202
#3  0x00007ffff149879d in fu_mm_device_setup (device=0x5555557542a0, error=0x7fffffffda30) at ../fwupd-1.7.4/plugins/modem-manager/fu-mm-device.c:1556
#4  0x00007ffff7f03dc0 in fu_device_setup (self=0x5555557542a0, error=error@entry=0x7fffffffda30) at ../fwupd-1.7.4/libfwupdplugin/fu-device.c:4284
#5  0x00007ffff14971d7 in fu_plugin_mm_device_add (plugin=plugin@entry=0x5555556b67e0, modem=modem@entry=0x555555688160) at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:230
#6  0x00007ffff149738e in fu_plugin_mm_setup_manager (plugin=0x5555556b67e0) at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:313
#7  fu_plugin_mm_name_owner_updated (plugin=0x5555556b67e0) at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:329
#8  0x00007ffff14974bb in fu_plugin_mm_coldplug (plugin=0x5555556b67e0, error=<optimized out>) at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:342
#9  0x00007ffff7f11cb7 in fu_plugin_runner_coldplug (self=self@entry=0x5555556b67e0, error=error@entry=0x7fffffffdba8) at ../fwupd-1.7.4/libfwupdplugin/fu-plugin.c:1023
#10 0x0000555555584851 in fu_engine_plugins_coldplug (self=0x5555555e8830) at ../fwupd-1.7.4/src/fu-engine.c:5591
#11 fu_engine_load (self=<optimized out>, flags=flags@entry=14, error=error@entry=0x7fffffffdc28) at ../fwupd-1.7.4/src/fu-engine.c:6993
#12 0x000055555556bd5d in main (argc=<optimized out>, argv=<optimized out>) at ../fwupd-1.7.4/src/fu-main.c:2112
(gdb) n
[Thread 0x7ffff25cf640 (LWP 1622) exited]
10:47:25:0053 FuIOChannel          fu_io_channel_new_file: assertion 'filename != NULL' failed
fu_mm_device_io_open (self=self@entry=0x5555557542a0, error=error@entry=0x7fffffffda30) at ../fwupd-1.7.4/plugins/modem-manager/fu-mm-device.c:700
700		if (self->io_channel == NULL)
```
`fu-mm-device.c:699` is assuming that `port_at` exists but the AT-port is not found / probed, the section that would assign the port is not even run:
```
Thread 1 "fwupd" hit Breakpoint 3, fu_mm_device_probe_default (error=0x7fffffffda30, device=0x5555557542a0) at ../fwupd-1.7.4/plugins/modem-manager/fu-mm-device.c:258
258		if (!mm_modem_get_ports(modem, &ports, &n_ports)) {
(gdb) n
265		if (self->update_methods & MM_MODEM_FIRMWARE_UPDATE_METHOD_FASTBOOT) {
(gdb) p ports
$1 = (MMModemPortInfo *) 0x555555750a10
(gdb) p n_ports
$2 = 2
(gdb) p ports[0]
$3 = {name = 0x55555574b310 "wwan0", type = MM_MODEM_PORT_TYPE_NET}
(gdb) p ports[1]
$4 = {name = 0x55555574b330 "wwan0mbim0", type = MM_MODEM_PORT_TYPE_MBIM}
(gdb) p self->update_methods
$5 = MM_MODEM_FIRMWARE_UPDATE_METHOD_MBIM_QDU
(gdb) l
260					    FWUPD_ERROR,
261					    FWUPD_ERROR_NOT_SUPPORTED,
262					    "failed to get port information");
263			return FALSE;
264		}
265		if (self->update_methods & MM_MODEM_FIRMWARE_UPDATE_METHOD_FASTBOOT) {
266			for (guint i = 0; i < n_ports; i++) {
267				if (ports[i].type == MM_MODEM_PORT_TYPE_AT) {
268					self->port_at = g_strdup_printf("/dev/%s", ports[i].name);
269					break;
```

I do have those:
```
crw------- 1 root root 238, 2 Jan 16 16:00 /dev/wwan0at0
crw------- 1 root root 238, 1 Jan 16 16:00 /dev/wwan0mbim0
crw------- 1 root root 238, 0 Jan 16 16:00 /dev/wwan0qcdm0
```
But it seems that ModemManager itself also doesn't show the AT port (at least I would expect it to be listed here):
```
[... 18 lines cut ...]
    at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:313
#2  fu_plugin_mm_name_owner_updated (plugin=0x5555556b67e0)
    at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:329
#3  0x00007ffff14974bb in fu_plugin_mm_coldplug (plugin=0x5555556b67e0, error=<optimized out>)
    at ../fwupd-1.7.4/plugins/modem-manager/fu-plugin-modem-manager.c:342
#4  0x00007ffff7f11cb7 in fu_plugin_runner_coldplug
    (self=self@entry=0x5555556b67e0, error=error@entry=0x7fffffffdba8)
    at ../fwupd-1.7.4/libfwupdplugin/fu-plugin.c:1023
#5  0x0000555555584851 in fu_engine_plugins_coldplug (self=0x5555555e8830)
    at ../fwupd-1.7.4/src/fu-engine.c:5591
#6  fu_engine_load (self=<optimized out>, flags=flags@entry=14, error=error@entry=0x7fffffffdc28)
    at ../fwupd-1.7.4/src/fu-engine.c:6993
#7  0x000055555556bd5d in main (argc=<optimized out>, argv=<optimized out>)
    at ../fwupd-1.7.4/src/fu-main.c:2112
```

With the patch I get:
```
11:07:11:0072 FuPlugin             coldplug(modem_manager)
11:07:11:0072 FuPluginMm           ModemManager 1.18.4 is available
11:07:11:0072 FuPluginMm           added modem: /org/freedesktop/ModemManager1/Modem/2
11:07:11:0072 FuPluginMm           valid firmware update combination: mbim-qdu
11:07:11:0073 FuPluginMm           failed to probe MM device: no AT port provided for filename
```
And the modem is not shown in the list of devices.

hughsie, 2022-01-19:
@aleksander0m is the expert here I guess. Do we need more things in the device node list perhaps?

benlorenz, 2022-01-19:
Looking at this a little bit more I think the ModemManager plugin runs into the branch recently introduced in #4133 which should maybe fail somewhat more graceful for this kind of modem without an AT port in ModemManager (and I guess that this modem should not need/have that AT port).

I added a patch to my build that reverts f0016cf11e8029fa112f5beed2a979fbf99f69d5. Then the error from the ModemManager plugin during startup is gone and I see the modem in the output again:
```
Devices with no available firmware updates: 
...
 • Qualcomm Snapdragon X55 5G
```

hughsie, 2022-01-19:
@benlorenz does this help? https://github.com/fwupd/fwupd/pull/4196

benlorenz, 2022-01-19:
> @benlorenz does this help? #4196

Yes that patch does help, no error from the plugin and the modem appears in the devicelist, thanks!

To make sure that I used the correct build I checked with once more with gdb:
```
18:24:32:0096 FuPlugin             coldplug(modem_manager)
18:24:32:0097 FuPluginMm           ModemManager 1.18.4 is available
18:24:32:0097 FuPluginMm           added modem: /org/freedesktop/ModemManager1/Modem/2
18:24:32:0097 FuPluginMm           valid firmware update combination: mbim-qdu

Thread 1 "fwupd" hit Breakpoint 1, fu_mm_device_setup (device=0x5555557522a0, error=0x7fffffffda20) at ../fwupd-1.7.4/plugins/modem-manager/fu-mm-device.c:1594
1594	{
(gdb) n
1595		FuMmDevice *self = FU_MM_DEVICE(device);
(gdb) 
1597		if (self->port_at != NULL) {
(gdb) 
1601		if (fu_device_get_branch(device) != NULL)
(gdb) 
1604			g_debug("using firmware branch: default");
(gdb) 
18:24:49:0451 FuPluginMm           using firmware branch: default
fu_device_setup (self=0x5555557522a0, error=error@entry=0x7fffffffda20) at ../fwupd-1.7.4/libfwupdplugin/fu-device.c:4289
4289		children = fu_device_get_children(self);
(gdb) c
Continuing.
18:25:07:0169 FuDevice             using bba54f76de66958e5d70964b6ddd2d1e6c7dae9e for /sys/devices/pci0000:00/0000:00:1c.0/0000:08:00.0
18:25:07:0169 FuPlugin             emit added from modem_manager: bba54f76de66958e5d70964b6ddd2d1e6c7dae9e
```

```
├─Qualcomm Snapdragon X55 5G:
│     Device ID:          bba54f76de66958e5d70964b6ddd2d1e6c7dae9e
│     Summary:            Mobile broadband device
│     Current version:    T99W175.F0.1.0.0.9.TF.008.055
│     Vendor:             foxconn (PCI:0x105B)
│     GUIDs:              ae673b81-b217-5b64-860d-b6ba059d34f1 ← PCI\VID_105B&PID_E0AB&REV_0000&CARRIER_TELEFONICA
│                         39fccad9-98bc-5221-83b2-ca390b2c954d ← PCI\VID_105B&PID_E0AB&REV_0000
│                         8e43be37-c679-5557-8309-4cc6ba927dc0 ← PCI\VID_105B&PID_E0AB
│                         c8591d06-63e4-55e7-9dec-3b0783b15e2c ← PCI\VID_105B
│     Device Flags:       • Updatable
│                         • System requires external power source
```

closed 2022-01-19 by hughsie

j5k, 2022-01-19:
One day later I come back and the problem seems to be fixed. I'm incredibly impressed! Thanks a lot!