fwupd/fwupd #4194: The meaning of, 'Devices with no available firmware updates', and of similar descriptions, is unclear

LinuxOnTheDesktop, 2022-01-19:
**Describe the bug**

fwupd can output each of the following strings (more specifically: strings that are something like titles).

- 'Devices with no available firmware updates'.
- 'Devices with the latest available firmware version'
- 'No updates available for remaining devices'

The difference in meaning between those titles is unclear.

**Steps to Reproduce**

Run the `get-updates` command. (Strictly: depending on the state of one's system, running that command can produce output that includes one or two or all three of the aforementioned strings.)

**Expected behavior**

One should be able to understand fwupd's output.

**fwupd version information**

client version:	1.7.2
compile-time dependency versions
gusb: 0.3.8

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): the problem applies to fwupd howsoever it was installed, I think.

**Additional questions**

- Operating system and version: Linux Mint 20.2
- Have you tried rebooting? Irrelevant
- Is this a regression? No.

closed 2022-04-27 by stale[bot]

kristof-mattei, 2023-02-04:
| Syntax      | Description | Description | 
| - | - | - |
| ``` Devices with no available firmware updates: ``` | message letting the user know no device upgrade available due to missing on LVFS | https://github.com/fwupd/fwupd/blob/255ca3d513590f66f7c3da48bd6f2a601fca82a0/src/fu-tool.c#L668-L670       |
| ```Devices with the latest available firmware version:``` | message letting the user know no device upgrade available | https://github.com/fwupd/fwupd/blob/255ca3d513590f66f7c3da48bd6f2a601fca82a0/src/fu-tool.c#L677-L678       |
| ```Devices with no available firmware updates:``` | (no direct explanation), but reference to `FWUPD_ERROR_NOTHING_TO_DO` | https://github.com/fwupd/fwupd/blob/255ca3d513590f66f7c3da48bd6f2a601fca82a0/src/fu-tool.c#L688-L690       |

LinuxOnTheDesktop, 2023-02-04:
@kristof-mattei 

&nbsp;&nbsp;&nbsp;&nbsp;&nbspThank you.

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;I remain unsure about the meaning of, `Devices with no available firmware updates`. Does it mean the following? Of the devices detected by fwupd on the user's system, none have updates available - because there is some sort of information missing from fwupd's database, where 'missing' means something like 'really should be there'.

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`Devices with the latest available firmware version`: I suppose that that text is clear enough, at least once one understands the message above.

&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`Devices with no available firmware updates`: 'no direct explanation' - quite.

kristof-mattei, 2023-02-04:
@LinuxOnTheDesktop 

> @kristof-mattei
> 
>      Thank you.
> 
>      I remain unsure about the meaning of, `Devices with no available firmware updates`. Does it mean the following? Of the devices detected by fwupd on the user's system, none have updates available - because there is some sort of information missing from fwupd's database, where 'missing' means something like 'really should be there'.
> 
>      `Devices with the latest available firmware version`: I suppose that that text is clear enough, at least once one understands the message above.
> 
>      `Devices with no available firmware updates`: 'no direct explanation' - quite.

`fwupd` seems to pull its updates from [https://fwupd.org/](LVFS)

| Syntax      | Description | 
| - | - | 
| ``` Devices with no available firmware updates: ``` | This device (or its manufacturer) was not found on LVFS. So we cannot ascertain whether the device is outdated or not. Example: You have a Samsung 980 Pro. Samsung is registered, but the 980 Pro isn't in the database. Or you have a WD Black, and WD isn't registered. |
| ```Devices with the latest available firmware version:``` | Device (and thus manufacturer) is registered on LVFS, butno updates available on LVFS for this device. Could be that manufacturer didn't upload latest firmware, or that device is actually up2date.  |
| ```Devices with no available firmware updates:``` | Indeed. |

LinuxOnTheDesktop, 2023-02-04:
@kristof-mattei

Right. Thank you again. What is meant now seems reasonably clear. Equally evident though is that the existing text - the text shown to the user - could be improved. I can try to think of (fairly pithy) formulations, if the developers want me to.

LinuxOnTheDesktop, 2023-02-06:
Here are my suggestions for rewording. 

- 'Devices with no available firmware updates' -> 'Devices unknown to the program'.

- 'Devices with the latest available firmware version' -> 'Devices with the latest known firmware'.

- 'No updates available for remaining devices' (a string seemingly missing from @kristof-mattei's tables) -> I don't know, because I still don't understand it.

It would be an idea also for the output in question to include a link to somewhere that has fuller explanations.

hughsie, 2023-02-06:
So, I think the confusion here is because we can have devices in two different states:

 * Enumerated, i.e. we know the firmware version, and we know the device is updatable, but we don't have the metadata about what new updates are available as the device is unknown to the LVFS.
 * The device has the newest version of firmware we know about, i.e. the device firmware version matches the newest archive on the LVFS.
 * The device has an older version than we know exists on the LVFS.

This is all handled here: https://github.com/fwupd/fwupd/blob/main/src/fu-util.c#L2113

LinuxOnTheDesktop, 2023-02-06:
@hughsie 

Thanks. Yet, in order to help you, _I need to be told, fairly plainly, what each heading_ ('Devices with no available firmware updates' etc.) _means_. Only then can I work on providing explanations - of those headings - that are plainer still and which can replace the headings that at present the user sees. Now, as I said, it seems to me that I am in a position - thanks to @kristof-mattei - to do the required rewriting, but only of _two_ of the three headings. I need to know what 'No updates available for remaining devices' means. Or are you saying, hughsie, that the explanations that I took myself to have in hand already - the respective explanations of the 'Devices with no available firmware updates' and ''Devices with the latest available firmware version' - are _wrong_?