fwupd/fwupd #4207: Lenovo T470: `fwupdmgr update` has effectively bricked my machine

yawpitch, 2022-01-20:
**Describe the bug**

Lenovo T470 failed to reboot after `fwupdmgr update`, bypassing the usual "Lenovo" boot screen and just going to black the first time. Waited over an hour just in case as the fan was going, but found no response to the power button; eventually tried a hardware reset via paperclip to the hardware reset button. This time I got the "Lenovo" screen, but after a few seconds "System Reset" appeared in the top left and again the machine froze, unresponsive once again to the power button, this time without fan speed. As per https://github.com/fwupd/firmware-lenovo/issues/140 only resort is to hardware reset again. On third attempt able to get to the bootloader and into Linux, however a fourth reboot has me back at a black boot screen. Attempted to boot into Windows to try the update there, but can no longer log into Windows as the machine again freezes up and becomes unresponsive to the power button as soon as the Windows 10 login splash becomes visible. Even tried downgrading to a previous version using gnome-firmware (couldn't find appropriate way to downgrade with fwupdmgr) and this appeared to work at first, but again got the black screen on reboot. Tried everything I found in #3819 and several open issues on https://github.com/fwupd/firmware-lenovo, to no avail.

- Secure Boot is disabled
- Boot Order Lock is disabled
- LUKS Disk encryption enabled on boot
- Tried both "UEFI only" boot and "both" (legacy and UEFI), same outcome

**Steps to Reproduce**
```shell
fwupdmgr refresh
fwupdmgr get-updates
fwupdmgr update
```

**Expected behavior**
The updates to complete and, ideally, to not have to keep a straightened paperclip handy at all times.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

```shell
client version:	1.7.4
compile-time dependency versions
	gusb:	0.3.10

daemon version:	1.7.4
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

```shell
> pacman -Syu fwupd
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

```shell
20HDCTO1WW
│
├─Core™ i7-7600U CPU @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ea
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         9ca69899-3716-5857-9fd3-882a5c73236f
│     Device Flags:       • Internal device
│
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.36
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUIDs:              4b37c365-ecb2-4d89-8e15-e93c27c6ce12
│                         897f3781-238c-56a0-95fa-e2612102db0e
│     Device Flags:       • Internal device
[... 703 lines cut ...]
│       ├── videotest.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── System Volume Information
│   ├── AadRecoveryPasswordDelete
│   └── ClientRecoveryPasswordRotation
└── vmlinuz-linux
```
**Additional questions**

- Operating system and version: Arch, fully updated
- Have you tried rebooting? YES ... many, many times. By paperclip.
- Is this a regression? Unknown.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? No.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? It is a Lenovo system, Boot Order Lock is disabled.

mrhpearson, 2022-01-21:
That sounds horrible :(
I've raised a ticket to get input from the FW team on recommendations to get your system back up and working. Ref LO-1531.
I'll update as soon as I hear from them with suggestions. 

Mark

ChiWei-Chen, 2022-01-21:
@yawpitch , I work with @mrhpearson and got the following questions from BIOS team.
Could you please check? 

> Can you confirm BIOS/EC version and MEFW version(including previous version and target version) for the fail system?
> Besides, I still want to know that whether MEFW can be updated successfully? Or ME was not successfully updated at all?

Thanks!
Chen

yawpitch, 2022-01-21:
Hi @mrhpearson and @ChiWei-Chen, thanks for looking into this.

 > Can you confirm BIOS/EC version and MEFW version(including previous version and target version) for the fail system?

After entering the boot setup, I see:

BIOS version: N1QET92W (1.67)
EC version: N1QHT54W (1.36)
ME version: 11.8.86.3909

Note that the original update that failed was for the BIOS and the ME, with the ME version being upgraded to being 11.8.86.3909. This has not changed in any attempt to downgrade the ME, which always ends in a "Reset System" and a hang on black and (eventually) a paperclip reset.

I then discard all changes and the machine continues to post (note that any time in troubleshooting this since the initial crash that I have made any change and saved that change the machine has immediately gone to a black screen and locked up, unresponsive to power button, requiring (eventually) a paperclip reset.

Once up in Arch I launch gnome-firmware and it reports:

BIOS (System Firmware) version: 0.1.6.7
EC version: 0.1.36
ME version: 184.77.3664 (this was my last attempt to downgrade ME)

 > Besides, I still want to know that whether MEFW can be updated successfully? Or ME was not successfully updated at all?

On the initial attempt that started all this I was asked if I wanted to restart, answered Y, the machine powered off and rebooted to Lenovo branded screen that eventually said "Reset system" then the screen died and it powered back up, terminating at a black screen that never went away and became unresponsive to the power button. So I believe it was never successfully updated. Eventually forced to cycle using the reset button, as per the Arch wiki, and after a second force reset was able to get into Arch. At that point `fwupdmgr get-history` showed "failed to run update on reboot" for ME version 3909.

I have made *NUMEROUS* attempts to downgrade and upgrade single items, current output of `fwupdmgr get-history` below appears to show the ME downgrade as successful with BIOS and EC as failed, though obviously the BIOS itself seems to disagree on all counts. 

```shell
20HDCTO1WW
│
├─THNSF5512GPUK TOSHIBA:
│ │   Device ID:          c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│ │   Previous version:   51025KLA
│ │   Update State:       Success
│ │   Last modified:      2020-03-19 13:28
│ │   GUID:               bfe8d16e-b612-5f26-9526-0123dd679e7d
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      51055KLA
│       License:          Unknown
│
├─UEFI Device Firmware:
│ │   Device ID:          efc88c8524a5d0ada629ba38b89ac722dc3d5efb
│ │   Previous version:   0.1.29
│ │   Update State:       Success
│ │   Last modified:      2020-03-19 13:33
│ │   GUID:               4b37c365-ecb2-4d89-8e15-e93c27c6ce12
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Reported to remote server
│ │                       • Device is usable for the duration of the update
│ │
│ └─  New version:      0.1.35
│       License:          Unknown
│
├─Intel Management Engine:
│ │   Device ID:          4718f2fc055b1c4de66633017326683a72592b4c
│ │   Previous version:   184.70.3626
│ │   Update State:       Success
│ │   Last modified:      2020-06-17 17:23
│ │   GUID:               bcff5fd8-0a99-48ad-b630-40e8eadb190e
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
[... 90 lines cut ...]
  │   GUID:               4e06f4ae-ab2e-4804-9a23-888e596a31e7
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │
  └─ThinkPad T470 / ThinkPad 25:
        New version:      0.1.63
        Remote ID:        lvfs
        Release ID:       5183
        Summary:          Lenovo ThinkPad T470 / 25 System Firmware
        License:          Proprietary
        Size:             9.4 MB
        Created:          2016-07-08
        Urgency:          High
        Vendor:           Lenovo
        Flags:            is-downgrade
        Description:
        Lenovo ThinkPad T470 / ThinkPad 25 System Firmware Version 1.63.Update includes a security fix.Updated the Diagnostics module to version 04.12.001.Fixed an issue where Force PXE boot by Intel AMT does not work.
``` 
The different ME version