fwupd/fwupd #4213: ThinkPad Thunderbolt 4 Docking Station - Ethernet crashes since fwupd to 1.7.4

fbatschi, 2022-01-21:
**Describe the bug**
Since updating fwupd to 1.7.4 there is apparently the possibility to do firmware updates for the Lenovo ThinkPad Thunderbolt 4 docking station.  After applying the firmware update for the docking station on my X1 Gen9 with the Thunderbolt 4 docking station there is a crash and the system freezes once the system tries to activate the Ethernet port of the docking station.

The following can be seen in journalctl

```
Jan 21 22:28:02 fedora kernel: igc: Failed to read reg 0xc030!
Jan 21 22:28:02 fedora kernel: WARNING: CPU: 7 PID: 2885 at drivers/net/ethernet/intel/igc/igc_main.c:6165 igc_rd32+0x7c/0x80 [igc]
Jan 21 22:28:02 fedora kernel: Modules linked in: uinput rfcomm snd_seq_dummy snd_hrtimer xt_conntrack xt_MASQUERADE nf_conntrack_netlink nft_counter xt_addrtype nft_compat br_netfilter bridge stp llc nls_utf8 cifs cifs_arc4 rdma_cm iw_cm ib_cm ib_core cifs_md4 dns_resolver fscache netfs nft_objref nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 overlay ip_set nf_tables nfnetlink qrtr ns bnep snd_ctl_led snd_soc_skl_hda_dsp snd_soc_intel_hda_dsp_common snd_soc_hdac_hdmi sunrpc snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic snd_soc_dmic iTCO_wdt snd_sof_pci_intel_tgl intel_pmc_bxt pmt_telemetry mei_wdt mei_hdcp iTCO_vendor_support pmt_class snd_sof_intel_hda_common intel_rapl_msr soundwire_intel soundwire_generic_allocation soundwire_cadence snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof iwlmvm
Jan 21 22:28:02 fedora kernel:  snd_soc_hdac_hda snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi soundwire_bus mac80211 snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine intel_tcc_cooling x86_pkg_temp_thermal intel_powerclamp snd_hda_intel coretemp snd_intel_dspcfg libarc4 intel_cstate snd_intel_sdw_acpi intel_uncore pcspkr snd_usb_audio snd_hda_codec iwlwifi snd_usbmidi_lib think_lmi snd_hda_core firmware_attributes_class snd_rawmidi vfat i2c_i801 wmi_bmof i2c_smbus uvcvideo snd_hwdep snd_seq fat btusb mei_me snd_seq_device cfg80211 mei videobuf2_vmalloc btrtl snd_pcm squashfs idma64 btbcm videobuf2_memops snd_timer loop videobuf2_v4l2 btintel videobuf2_common bluetooth joydev videodev processor_thermal_device_pci_legacy processor_thermal_device mc ecdh_generic thunderbolt intel_pmt processor_thermal_rfim processor_thermal_mbox processor_thermal_rapl intel_rapl_common intel_soc_dts_iosf igen6_edac thinkpad_acpi nxp_nci_i2c nxp_nci nci ledtrig_audio platform_profile nfc snd rfkill soundcore
Jan 21 22:28:02 fedora kernel:  int3403_thermal int340x_thermal_zone soc_button_array acpi_pad intel_hid int3400_thermal acpi_tad acpi_thermal_rel sparse_keymap zram ip_tables dm_crypt trusted asn1_encoder hid_logitech_hidpp hid_logitech_dj hid_sensor_hub intel_ishtp_loader intel_ishtp_hid i915 hid_multitouch i2c_algo_bit ttm drm_kms_helper cec crct10dif_pclmul crc32_pclmul crc32c_intel drm nvme igc serio_raw intel_ish_ipc ghash_clmulni_intel intel_ishtp nvme_core ucsi_acpi typec_ucsi typec i2c_hid_acpi wmi i2c_hid video pinctrl_tigerlake ipmi_devintf ipmi_msghandler fuse
Jan 21 22:28:02 fedora kernel: CPU: 7 PID: 2885 Comm: gnome-shell Not tainted 5.15.15-200.fc35.x86_64 #1
Jan 21 22:28:02 fedora kernel: Hardware name: LENOVO 20XWCTO1WW/20XWCTO1WW, BIOS N32ET75W (1.51 ) 12/02/2021
Jan 21 22:28:02 fedora kernel: RIP: 0010:igc_rd32+0x7c/0x80 [igc]
Jan 21 22:28:02 fedora kernel: Code: 48 c7 c6 58 e3 40 c0 e8 3a 20 85 c7 48 8b bb 30 ff ff ff e8 66 76 2e c7 84 c0 74 b2 89 ee 48 c7 c7 80 e3 40 c0 e8 a2 67 7f c7 <0f> 0b eb a0 0f 1f 44 00 00 41 56 41 55 41 54 55 48 89 f5 53 80 7e
Jan 21 22:28:02 fedora kernel: RSP: 0018:ffffa25cc5897c70 EFLAGS: 00010282
Jan 21 22:28:02 fedora kernel: RAX: 000000000000001f RBX: ffff9008cb518c10 RCX: 0000000000000027
Jan 21 22:28:02 fedora kernel: RDX: ffff900fff7e0a08 RSI: 0000000000000001 RDI: ffff900fff7e0a00
Jan 21 22:28:02 fedora kernel: RBP: 000000000000c030 R08: 0000000000000000 R09: ffffa25cc5897aa8
Jan 21 22:28:02 fedora kernel: R10: ffffa25cc5897aa0 R11: ffffffff88f46028 R12: 00000000ffffffff
Jan 21 22:28:02 fedora kernel: R13: ffff9008cb518000 R14: ffff9008cb3cdd40 R15: 000000000000c030
Jan 21 22:28:02 fedora kernel: FS:  00007f3e74c2bd80(0000) GS:ffff900fff7c0000(0000) knlGS:0000000000000000
Jan 21 22:28:02 fedora kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 21 22:28:02 fedora kernel: CR2: 000006ee3d0f4018 CR3: 0000000166320003 CR4: 0000000000770ee0
Jan 21 22:28:02 fedora kernel: PKRU: 55555554
Jan 21 22:28:02 fedora kernel: Call Trace:
Jan 21 22:28:02 fedora kernel:  <TASK>
Jan 21 22:28:02 fedora kernel:  igc_update_stats+0x72/0x690 [igc]
Jan 21 22:28:02 fedora kernel:  igc_get_stats64+0x7f/0x90 [igc]
Jan 21 22:28:02 fedora kernel:  dev_get_stats+0x59/0xc0
Jan 21 22:28:02 fedora kernel:  dev_seq_printf_stats+0x20/0xb0
Jan 21 22:28:02 fedora kernel:  dev_seq_show+0x10/0x30
Jan 21 22:28:02 fedora kernel:  seq_read_iter+0x2bf/0x4b0
Jan 21 22:28:02 fedora kernel:  seq_read+0xed/0x120
Jan 21 22:28:02 fedora kernel:  proc_reg_read+0x52/0xa0
Jan 21 22:28:02 fedora kernel:  vfs_read+0x92/0x190
Jan 21 22:28:02 fedora kernel:  ksys_read+0x4f/0xc0
Jan 21 22:28:02 fedora kernel:  do_syscall_64+0x38/0x90
Jan 21 22:28:02 fedora kernel:  entry_SYSCALL_64_after_hwframe+0x44/0xae
Jan 21 22:28:02 fedora kernel: RIP: 0033:0x7f3e7b0d978c
Jan 21 22:28:02 fedora kernel: Code: ec 28 48 89 54 24 18 48 89 74 24 10 89 7c 24 08 e8 39 89 f8 ff 48 8b 54 24 18 48 8b 74 24 10 41 89 c0 8b 7c 24 08 31 c0 0f 05 <48> 3d 00 f0 ff ff 77 34 44 89 c7 48 89 44 24 08 e8 7f 89 f8 ff 48
Jan 21 22:28:02 fedora kernel: RSP: 002b:00007fffbb0ee1e0 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
Jan 21 22:28:02 fedora kernel: RAX: ffffffffffffffda RBX: 000055ff98059220 RCX: 00007f3e7b0d978c
Jan 21 22:28:02 fedora kernel: RDX: 0000000000001000 RSI: 00007fffbb0ee2f0 RDI: 0000000000000042
Jan 21 22:28:02 fedora kernel: RBP: 00007f3e7b1d43a0 R08: 0000000000000000 R09: 0000000000000000
Jan 21 22:28:02 fedora kernel: R10: 0000000000001000 R11: 0000000000000246 R12: 00007fffbb0ee2f0
Jan 21 22:28:02 fedora kernel: R13: 0000000000000d68 R14: 00007f3e7b1d37a0 R15: 0000000000001000
Jan 21 22:28:02 fedora kernel:  </TASK>
Jan 21 22:28:02 fedora kernel: ---[ end trace 3bf280f689b0d891 ]---
Jan 21 22:28:02 fedora kernel: BUG: unable to handle page fault for address: 000000000000c030
Jan 21 22:28:02 fedora kernel: #PF: supervisor write access in kernel mode
Jan 21 22:28:02 fedora kernel: #PF: error_code(0x0002) - not-present page
Jan 21 22:28:02 fedora kernel: PGD 0 P4D 0 
Jan 21 22:28:03 fedora kernel: Oops: 0002 [#1] SMP NOPTI
Jan 21 22:28:03 fedora kernel: CPU: 7 PID: 2885 Comm: gnome-shell Tainted: G        W         5.15.15-200.fc35.x86_64 #1
Jan 21 22:28:03 fedora kernel: Hardware name: LENOVO 20XWCTO1WW/20XWCTO1WW, BIOS N32ET75W (1.51 ) 12/02/2021
Jan 21 22:28:03 fedora kernel: RIP: 0010:igc_update_stats+0x8c/0x690 [igc]
Jan 21 22:28:03 fedora kernel: Code: 89 0c 24 4e 8b b4 eb 88 00 00 00 e8 ae f1 ff ff 8b 93 fc 02 00 00 48 8b 0c 24 85 d2 74 0e 48 8b b3 98 02 00 00 31 d2 4c 01 fe <89> 16 85 c0 74 10 89 c0 49 01 86 80 00 00 00 48 01 83 40 02 00 00
Jan 21 22:28:03 fedora kernel: RSP: 0018:ffffa25cc5897c98 EFLAGS: 00010206
Jan 21 22:28:03 fedora kernel: RAX: 00000000ffffffff RBX: ffff9008cb518980 RCX: 0000000000000000
Jan 21 22:28:03 fedora kernel: RDX: 0000000000000000 RSI: 000000000000c030 RDI: ffff900fff7e0a00
Jan 21 22:28:03 fedora kernel: RBP: ffff9008cb518c10 R08: 0000000000000000 R09: ffffa25cc5897aa8
Jan 21 22:28:03 fedora kernel: R10: ffffa25cc5897aa0 R11: ffffffff88f46028 R12: 0000000000000000
Jan 21 22:28:03 fedora kernel: R13: 0000000000000000 R14: ffff9008cb3cdd40 R15: 000000000000c030
Jan 21 22:28:03 fedora kernel: FS:  00007f3e74c2bd80(0000) GS:ffff900fff7c0000(0000) knlGS:0000000000000000
Jan 21 22:28:03 fedora kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jan 21 22:28:03 fedora kernel: CR2: 000000000000c030 CR3: 0000000166320003 CR4: 0000000000770ee0
Jan 21 22:28:03 fedora kernel: PKRU: 55555554
Jan 21 22:28:03 fedora kernel: Call Trace:
Jan 21 22:28:03 fedora kernel:  <TASK>
[... 100 lines cut ...]
│     Device Flags:       • Updatable
│                         • System requires external power source
│                         • Device stages updates
│   
├─ThinkPad Thunderbolt 4 Dock:
[... cut ...]
│     GUIDs:              f1bdfa07-026c-5973-89f9-b7bd8e42ff9d
│                         dc3a1b4e-176f-589a-949c-d058324bb6a7
│                         350ec5bb-25c0-543e-8d4c-72c30d2b8df0
│                         2e927863-03be-5488-aeba-b0b9b3895fb5
│     Device Flags:       • System requires external power source
│                         • Device stages updates
│                         • Updatable
```

**Additional questions**

- Operating system and version:
Fedora 35

- Have you tried rebooting?
Yes

hughsie, 2022-01-22:
What firmware update version did you update to please. Was the package from the LVFS?

fbatschi, 2022-01-23:
I updated via LVFS, yes. 

I now updated the dock on a windows machine to the latest available firmware which is 1.0.07 for the dock and 35.00.06 for the TB4 and 1.73.1 for the I225 ethernet chip. The problem persist. As it only started after doing the update i assume it must be related to one of these firmwares.

Apparently not related to fwupd itself (although i am still not sure what the 
Update error means. How is the firmware update "inhibited" by the usi_dock plugin?

```
─ThinkPad Thunderbolt 4 Dock:
│     Device ID:          b927c17354471ef8df740c693f2fbba8092a9b11
│     Current version:    35.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0108, TBT:0x0108)
│     Update Error:       firmware update inhibited by [usi_dock] plugin
```

hughsie, 2022-01-24:
> How is the firmware update "inhibited" by the usi_dock plugin

Ohh, this just means that the device isn't to be updated using the thunderbolt mechanism, and instead it should be updated using the USI protocol instead. That bit is working as designed, but I think the TBT4 firmware has the issue. I'll reassign and poke some people.