fwupd/fwupd #4221: fwupd -- daemon seems to have trouble -- segfault at 8 (from dmesg log)

tswaehn, 2022-01-26:
**Describe the bug**
Failed to start Firmware update daemon.

**Steps to Reproduce**
Restart fwupd daemon

**Expected behavior**
Daemon can start

**fwupd version information**
```
fwupdmgr --version
client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5
```

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
installed with apt

**fwupd device information**
no devices found

```shell
fwupdmgr get-devices --show-all-devices
```
hangs and receives timeout

```
fwupdmgr get-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached

```

**Additional questions**

- Operating system and version: Linux debian64 5.15.0-2-amd64 #1 SMP Debian 5.15.5-2 (2021-12-18) x86_64 GNU/Linux
- Have you tried rebooting? - yes - multiple times
- Is this a regression?

hughsie, 2022-01-26:
what does `sudo fwupdtool get-devices --verbose` say?

tswaehn, 2022-01-26:
```
fwupdmgr get-devices --verbose
(fwupdmgr:15035): GLib-DEBUG: 16:55:23.481: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:15035): GLib-GIO-DEBUG: 16:55:23.490: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:15035): dconf-DEBUG: 16:55:23.490: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:15035): dconf-DEBUG: 16:55:23.490: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:15035): dconf-DEBUG: 16:55:23.490: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:15035): dconf-DEBUG: 16:55:23.490: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:15035): dconf-DEBUG: 16:55:23.490: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:15035): GLib-GIO-DEBUG: 16:55:23.490: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:15035): dconf-DEBUG: 16:55:23.491: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:15035): dconf-DEBUG: 16:55:23.491: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:15035): dconf-DEBUG: 16:55:23.491: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:15035): dconf-DEBUG: 16:55:23.491: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:15035): dconf-DEBUG: 16:55:23.491: watch_established: "/system/proxy/socks/" (establishing: 1)
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

hughsie, 2022-01-26:
`fwupdtool` not `fwupdmgr`

tswaehn, 2022-01-26:
sorry. sure. 

thats from journal

```
Jan 23 16:58:10 debian64 dbus-daemon[898]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service' requested by ':1.148' (uid=1000 pid=5398 comm="/usr/bin/gnome-software --gapplication-service ")
Jan 23 16:58:10 debian64 fwupd[5463]: 15:58:10:0634 GLib-GIO             g_file_new_for_path: assertion 'path != NULL' failed
Jan 23 16:58:10 debian64 fwupd[5463]: 15:58:10:0634 GLib-GIO             g_file_query_file_type: assertion 'G_IS_FILE(file)' failed
Jan 23 16:58:10 debian64 fwupd[5463]: 15:58:10:0634 GLib-GIO             g_file_monitor_file: assertion 'G_IS_FILE (file)' failed
Jan 23 16:58:10 debian64 kernel: fwupd[5463]: segfault at 8 ip 000055fadcf28109 sp 00007ffe82913720 error 4 in fwupd[55fadcf23000+26000]
Jan 23 16:58:10 debian64 systemd[1]: fwupd.service: Main process exited, code=killed, status=11/SEGV

```

```
sudo fwupdtool get-devices --verbose
15:58:01:0534 FuDebug              Verbose debugging enabled (on console 1)
15:58:01:0538 FuConfig             loading config values from /etc/fwupd/daemon.conf
15:58:01:0538 FuConfig             using autodetected max archive size 18.4 EB
15:58:01:0539 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
15:58:01:0539 FuRemoteList         loading /usr/share/fwupd/metainfo
15:58:01:0539 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:58:01:0539 XbSilo               file: c24eed9e-2b41-141a-3740-58c2ffc94728, current:4116e0d4-cf6c-79a3-41ae-92c2bcdd105e, cached: (null)
15:58:01:0543 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
15:58:01:0543 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
15:58:01:0543 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
15:58:01:0543 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
15:58:01:0543 GLib-GIO             g_file_new_for_path: assertion 'path != NULL' failed
15:58:01:0543 GLib-GIO             g_file_query_file_type: assertion 'G_IS_FILE(file)' failed
15:58:01:0543 GLib-GIO             g_file_monitor_file: assertion 'G_IS_FILE (file)' failed
Segmentation fault
```

tswaehn, 2022-01-26:
could be some file guessing that comes with a non-fullfilled path ?

tswaehn, 2022-01-26:
just updated

> installed with apt (using debian/bookworm [currently sid])

hughsie, 2022-01-26:
Can you get a backtrace with G_DEBUG=fatal-criticals set please?

tswaehn, 2022-01-28:
not sure if that is, what you mean? if not I am happy to follow ...

```
$ sudo gdb fwupdtool 
GNU gdb (Debian 10.1-2) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from fwupdtool...
(No debugging symbols found in fwupdtool)
(gdb) run get-devices --verbose G_DEBUG=fatal-criticals 
Starting program: /usr/bin/fwupdtool get-devices --verbose G_DEBUG=fatal-criticals
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff40f1640 (LWP 18881)]
15:22:28:0825 FuDebug              Verbose debugging enabled (on console 1)
[New Thread 0x7ffff38f0640 (LWP 18882)]
15:22:28:0830 FuConfig             loading config values from /etc/fwupd/daemon.conf
15:22:28:0830 FuConfig             using autodetected max archive size 18.4 EB
15:22:28:0838 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
15:22:28:0838 FuRemoteList         loading /usr/share/fwupd/metainfo
15:22:28:0838 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:22:28:0838 XbSilo               file: 4116e0d4-cf6c-79a3-41ae-92c2bcdd105e, current:4116e0d4-cf6c-79a3-41ae-92c2bcdd105e, cached: (null)
15:22:28:0838 XbSilo               loading silo with file contents
15:22:28:0839 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
15:22:28:0839 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
15:22:28:0839 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
15:22:28:0839 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
15:22:28:0839 GLib-GIO             g_file_new_for_path: assertion 'path != NULL' failed
15:22:28:0839 GLib-GIO             g_file_query_file_type: assertion 'G_IS_FILE(file)' failed
15:22:28:0839 GLib-GIO             g_file_monitor_file: assertion 'G_IS_FILE (file)' failed

Thread 1 "fwupdtool" received signal SIGSEGV, Segmentation fault.
0x000055555556cbba in main ()
(gdb) where
#0  0x000055555556cbba in main ()
(gdb) list
1	../sysdeps/unix/sysv/linux/dl-vdso-setup.c: No such file or directory.
(gdb) backtrace -full
#0  0x000055555556cbba in main ()

```

hughsie, 2022-01-31:
Okay, nearly there I think. If you do:

    sudo G_DEBUG=fatal-criticals gdb --args fwupdtool get-devices --verbose

then `r` to run, and `bt` to get backtrace, what does that say? Sorry for the delay, most of the family currently have COVID!

tswaehn, 2022-02-08:
here we go.

```
$ sudo G_DEBUG=fatal-criticals gdb --args fwupdtool get-devices --verbose
[sudo] password for tswaehn: 
GNU gdb (Debian 10.1-2) 10.1.90.20210103-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from fwupdtool...
(No debugging symbols found in fwupdtool)
(gdb) r
Starting program: /usr/bin/fwupdtool get-devices --verbose
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff40f1640 (LWP 10456)]
07:37:18:0652 FuDebug              Verbose debugging enabled (on console 1)
[New Thread 0x7ffff38f0640 (LWP 10457)]
07:37:18:0662 FuConfig             loading config values from /etc/fwupd/daemon.conf
07:37:18:0662 FuConfig             using autodetected max archive size 18.4 EB
07:37:18:0671 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
07:37:18:0671 FuRemoteList         loading /usr/share/fwupd/metainfo
07:37:18:0672 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
07:37:18:0672 XbSilo               file: c24eed9e-2b41-141a-3740-58c2ffc94728, current:4116e0d4-cf6c-79a3-41ae-92c2bcdd105e, cached: (null)
07:37:18:0676 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
07:37:18:0677 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
07:37:18:0677 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
07:37:18:0677 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
07:37:18:0677 GLib-GIO             g_file_new_for_path: assertion 'path != NULL' failed

Thread 1 "fwupdtool" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff7bdc4da in g_logv () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
(gdb) bt
#0  0x00007ffff7bdc4da in g_logv () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007ffff7bdc78f in g_log () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff7d776ba in g_file_new_for_path () at /lib/x86_64-linux-gnu/libgio-2.0.so.0
#3  0x000055555559237b in  ()
#4  0x0000555555592a91 in fu_remote_list_reload ()
#5  0x000055555559326d in fu_remote_list_load ()
#6  0x000055555558ad61 in fu_engine_load ()
#7  0x000055555556f1b7 in  ()
#8  0x000055555556fc91 in  ()
#9  0x0000555555574bbd in fu_util_cmd_array_run ()
#10 0x000055555556c9bd in main ()
(gdb) 
```

tswaehn, 2022-02-08:
Just a side note: I tried to build it from sources. Installed all dependencies (which are honestly massive huge). Meson went through - however compiling with ninja failed with missing files :( `Package `Json-1.0' not found in specified Vala API directories or GObject-Introspection GIR directories`

hughsie, 2022-02-08:
Can you install the fwupd-dbg package please and get me the backtrace again please; there are a few places that problem could come from and I don't see anything obvious.

> Installed all dependencies (which are honestly massive huge).

So honest question back; what dependencies were unnecessary in your opinion?

>  Couldn't find include Json-1.0.gir

Not sure about that; you're using `./contrib/setup` to install the deps?

tswaehn, 2022-02-18:
I did install the debug symbols. At the same time it did an update to a newer version

```
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-2-amd64
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.8

```

looks issue is gone now :D

closed 2022-02-18 by hughsie

elsandosgrande, 2022-02-23:
@hughsie So, I've run into this exact issue on Ubuntu Impish (21.10). Here's the backtrace with the debugging symbols package installed:
```
(gdb) backtrace
#0  0x00007ffff7bdad97 in g_logv () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007ffff7bdb053 in g_log () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff7d73b3a in g_file_new_for_path () at /lib/x86_64-linux-gnu/libgio-2.0.so.0
#3  0x00005555555907db in fu_remote_list_add_inotify (self=self@entry=0x5555555ed080, filename=<optimised out>, error=error@entry=0x7fffffffdd50) at ../src/fu-remote-list.c:88
#4  0x0000555555593b91 in fu_remote_list_add_for_path (error=0x7fffffffdd50, path=0x555555626370 "/etc/fwupd", self=0x5555555ed080) at ../src/fu-remote-list.c:197
#5  fu_remote_list_reload (self=self@entry=0x5555555ed080, error=error@entry=0x7fffffffdd50) at ../src/fu-remote-list.c:355
#6  0x0000555555594375 in fu_remote_list_load (self=0x5555555ed080, flags=FU_REMOTE_LIST_LOAD_FLAG_NONE, error=0x7fffffffdd50) at ../src/fu-remote-list.c:449
#7  0x000055555558ea51 in fu_engine_load (self=0x5555555f30c0, flags=14, error=0x7fffffffdd50) at ../src/fu-engine.c:6414
#8  0x0000555555573e80 in fu_util_start_engine (priv=0x5555555e56a0, flags=14, error=0x7fffffffdd50) at ../src/fu-tool.c:177
#9  0x0000555555575035 in fu_util_get_devices (priv=0x5555555e56a0, values=<optimised out>, error=0x7fffffffdd50) at ../src/fu-tool.c:666
#10 0x000055555556ff0d in fu_util_cmd_array_run (array=array@entry=0x5555555e5800, priv=priv@entry=0x5555555e56a0, command=0x7fffffffe59f "get-devices", values=0x7fffffffe298, error=error@entry=0x7fffffffdd50)
    at ../src/fu-util-common.c:551
#11 0x000055555556c9dd in main (argc=<optimised out>, argv=<optimised out>) at ../src/fu-tool.c:3186
```

For the record, this is the version that's available on Ubuntu, fully up to date:
```
$ apt search fwupd
Sorting... Done
Full Text Search... Done
fwupd/impish,now 1.5.11-0ubuntu2 amd64 [installed,automatic]
  Firmware update daemon
[…]
```

hughsie, 2022-02-23:
> this is the version that's available on Ubuntu

Not sure I can do much about them I'm afraid. There's a fwupd snap if that helps: https://github.com/fwupd/fwupd/wiki/fwupd-snap

troyrollo, 2023-05-05:
I had this problem and traced it to a different version of fwupd and its libraries being installed in /usr/local - once these were removed, fwupd started as expected.