fwupd/fwupd #4237: Remember window size

n1729m, 2022-01-31:
**Is your feature request related to a problem? Please describe.**

On opening the GUI firmware, the window size [right pan] doesn't display information completely.

[image]

So to display it properly, I resize it as below:

[image]

Upon relaunching the firmware app, the resized window size is not remembered and it defaults to display as seen in the first screenshot.

**Describe the solution you'd like**

- Possibility to remember the window size as resized by user.
- Dynamically resize window to display all info without user having to resize it.

**Additional context**

- OS: Arch Linux
- DE: Gnome 41.3 on Wayland.

hughsie, 2022-01-31:
Can you file this here please: https://gitlab.gnome.org/hughsie/gnome-firmware-updater

closed 2022-01-31 by hughsie

n1729m, 2022-01-31:
I apologize for opening issue at wrong place. Really appreciate for pointing me right direction.