fwupd/fwupd #4242: Error calling StartServiceByName for org.freedesktop.fwupd

Neo-Turak, 2022-02-02:
**Describe the bug**
Full message:Sorry something went wrong: Error calling StartServiceByName for org.freedektop.fwupd:Failed to activate service 'org.freedesktop.fwupd':timed out (service_start_timeout=25000ms)

but it did not effect any function of the software manager somehow.
this error occurs when I opening the Software Manager on Ubuntu.(however, it did not effect any function of the software manager somehow.)
OS Info below:
Ubuntu Jammy Jellyfish (development branch),41.1

**Steps to Reproduce**
open Ubuntu Software.

**Expected behavior**
it should not be show error message.

**fwupd version information**
fwupdmgr --version
client version:	1.7.1
compile-time dependency versions
	gusb:	0.3.8
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
(also this error occurs here)

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt.
**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
still shows error.
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

```

**Additional questions**

- Operating system and version:
- Linux  5.13.0-19-generic #19-Ubuntu SMP Thu Oct 7 21:58:00 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
- Have you tried rebooting?
- yes.

superm1, 2022-02-02:
Can you please share journal output for fwupd.service?

superm1, 2022-02-28:
Jammy has been upgraded to 1.7.5.  I have checked myself and didn't see this.  If you continue to see it, please re-open with the requested logs.

closed 2022-02-28 by superm1