fwupd/fwupd #4250: Segfault in fwupd service

wizeman, 2022-02-04:
**Describe the bug**
The fwupd service fails with a segfault after upgrading to NixOS 21.11. This happens 100% of the time on only 2 of my machines (both have Intel Atom CPUs with 2 GB of RAM), but not on any of my other ones (Raspberry Pis, AMD laptop, AMD desktop, AMD server and also a virtual machine with 2 GB of RAM).

Example log:
```syslog
2022-02-04T18:07:19.468127+00:00 atom2 daemon.info dbus-daemon[708]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service' requested by ':1.507' (uid=0 pid=30956 comm="/run/current-system/sw/bin/fwupdmgr refresh --forc" label="kernel")
2022-02-04T18:07:20.778116+00:00 atom2 kern.info kernel: .fwupd-wrapped[31050]: segfault at 280000042 ip 000074539a7bfd56 sp 00007ffc5f22cb78 error 4 in libc-2.33.so[74539a74a000+143000]
2022-02-04T18:07:20.778260+00:00 atom2 kern.warning kernel: CPU: 2 PID: 31050 Comm: .fwupd-wrapped Tainted: G S              T 5.15.18 #1-NixOS
2022-02-04T18:07:20.778644+00:00 atom2 daemon.info fwupd[31050]: 18:07:20:0775 XbValueBindings      xb_value_bindings_bind_str: assertion 'str != NULL' failed
2022-02-04T18:07:22.227174+00:00 atom2 daemon.warning systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
2022-02-04T18:07:22.230111+00:00 atom2 daemon.warning systemd[1]: fwupd.service: Failed with result 'core-dump'.
2022-02-04T18:07:44.482629+00:00 atom2 auth.notice dbus-daemon[708]: [system] Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)2022-02-04T18:07:19.468127+00:00 atom2 daemon.info dbus-daemon[708]: [system] Activating via systemd: service name='org.freedesktop.fwupd' unit='fwupd.service' requested by ':1.507' (uid=0 pid=30956 comm="/run/current-system/sw/bin/fwupdmgr refresh --forc" label="kernel")
2022-02-04T18:07:20.778116+00:00 atom2 kern.info kernel: .fwupd-wrapped[31050]: segfault at 280000042 ip 000074539a7bfd56 sp 00007ffc5f22cb78 error 4 in libc-2.33.so[74539a74a000+143000]
2022-02-04T18:07:20.778260+00:00 atom2 kern.warning kernel: CPU: 2 PID: 31050 Comm: .fwupd-wrapped Tainted: G S              T 5.15.18 #1-NixOS
2022-02-04T18:07:20.778644+00:00 atom2 daemon.info fwupd[31050]: 18:07:20:0775 XbValueBindings      xb_value_bindings_bind_str: assertion 'str != NULL' failed
2022-02-04T18:07:22.227174+00:00 atom2 daemon.warning systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
2022-02-04T18:07:22.230111+00:00 atom2 daemon.warning systemd[1]: fwupd.service: Failed with result 'core-dump'.
2022-02-04T18:07:44.482629+00:00 atom2 auth.notice dbus-daemon[708]: [system] Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

**Steps to Reproduce**

Run `fwupdmgr --refresh --force`.
But it also seems to fail with other commands.

**Expected behavior**

No segfault.

**fwupd version information**

```shell
$ fwupdmgr --version
client version: 1.7.2
compile-time dependency versions
        gusb:   0.3.7

Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

Installed on NixOS via the Nix package manager.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

**Additional questions**

- Operating system and version: NixOS 21.11
- Have you tried rebooting? Yes
- Is this a regression? Yes

hughsie, 2022-02-04:
> fwupd[31050]: 18:07:20:0775 XbValueBindings      xb_value_bindings_bind_str: assertion 'str != NULL' failed

Can you do:

    sudo G_DEBUG=fatal-criticals gdb --args fwupdtool get-devices
    (gdb) r
    <then wait for the crash>
    (gdb) bt

and then paste the backtrace please. Thanks!

wizeman, 2022-02-04:
```
(gdb) bt
#0  0x00007ffff7bdf087 in g_logv () from /nix/store/wvb73257jlqw35h96lp0r1x7n7jcaayj-glib-2.70.1/lib/libglib-2.0.so.0
#1  0x00007ffff7bdf30f in g_log () from /nix/store/wvb73257jlqw35h96lp0r1x7n7jcaayj-glib-2.70.1/lib/libglib-2.0.so.0
#2  0x00007ffff7f3c43a in fu_quirks_lookup_by_id (self=0x498c00, group=group@entry=0x53fa00 "", key=key@entry=0x44e42a "BatteryThreshold") at ../libfwupdplugin/fu-quirks.c:409
#3  0x00007ffff7f21404 in fu_context_lookup_quirk_by_id (self=self@entry=0x48a140, guid=guid@entry=0x53fa00 "", key=key@entry=0x44e42a "BatteryThreshold") at ../libfwupdplugin/fu-context.c:446
#4  0x000000000043e398 in fu_engine_context_set_battery_threshold (ctx=0x48a140) at ../src/fu-engine.c:6690
#5  fu_engine_load (self=0x495820, flags=flags@entry=14, error=error@entry=0x7fffffffced0) at ../src/fu-engine.c:6910
#6  0x0000000000420180 in fu_util_start_engine (priv=priv@entry=0x486f90, flags=flags@entry=14, error=error@entry=0x7fffffffced0) at ../src/fu-tool.c:247
#7  0x0000000000420fd1 in fu_util_get_devices (priv=0x486f90, values=<optimized out>, error=0x7fffffffced0) at ../src/fu-tool.c:748
#8  0x000000000042655d in fu_util_cmd_array_run (array=array@entry=0x487200, priv=priv@entry=0x486f90, command=0x7fffffffd7a9 "get-devices", values=0x7fffffffd3e8, error=error@entry=0x7fffffffced0) at ../src/fu-util-common.c:605
#9  0x000000000041d635 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-tool.c:3503
```

hughsie, 2022-02-04:
hmm, odd. what's the output of `sudo fwupdtool hwids` please

wizeman, 2022-02-04:
```shell
$ fwupdtool hwids
Computer Information
--------------------
BiosVendor: Intel Corp.
BiosVersion: MTCDT10N.86A.0165.2013.0114.1540
BiosMajorRelease: 0
BiosMinorRelease: 0
FirmwareMajorRelease: 00
FirmwareMinorRelease: 00
Manufacturer:
ProductName:
EnclosureKind: 3
BaseboardManufacturer: Intel Corporation
BaseboardProduct: DN2800MT

Hardware IDs
------------
not available as 'Family' unknown
not available as 'Family' unknown
{14c2cad6-3f14-5353-bb86-96f50fd4456e}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
not available as 'Family' unknown
not available as 'Family' unknown
not available as 'Family' unknown
not available as 'ProductSku' unknown
not available as 'ProductSku' unknown
{a75daac7-135f-5d1e-aaed-1d865311fb9b}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{34e7ec17-de26-5884-b775-61d25efde5bf}   <- Manufacturer + ProductName
not available as 'Family' unknown
not available as 'Family' unknown
{ec031666-69d7-5bf3-bb55-b3f68a59fedd}   <- Manufacturer + EnclosureKind
{c652fde9-dcdd-5ec5-9261-cc7a467c1faf}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
no GUIDs in data
```

wizeman, 2022-02-04:
Running the same command on the other machine that fails (just in case it helps):
```shell
# fwupdtool hwids
Computer Information
--------------------
BiosVendor:
BiosVersion: 7/29/2009
BiosMajorRelease: 0
BiosMinorRelease: 0
FirmwareMajorRelease: 00
FirmwareMinorRelease: 00
Manufacturer:
ProductName:
EnclosureKind: 3
BaseboardManufacturer: Intel Corporation
BaseboardProduct: D945GCLF2

Hardware IDs
------------
not available as 'Family' unknown
not available as 'Family' unknown
{6475b18f-6948-5ed4-b8c9-d284722d0364}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
not available as 'Family' unknown
not available as 'Family' unknown
not available as 'Family' unknown
not available as 'ProductSku' unknown
not available as 'ProductSku' unknown
{ed3524dd-3a14-57fe-a615-a4639599938c}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{34e7ec17-de26-5884-b775-61d25efde5bf}   <- Manufacturer + ProductName
not available as 'Family' unknown
not available as 'Family' unknown
{ec031666-69d7-5bf3-bb55-b3f68a59fedd}   <- Manufacturer + EnclosureKind
{fc87e0d7-c913-57ae-9a13-9486f67ba71d}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
no GUIDs in data
```

hughsie, 2022-02-04:
Can you try the wip/hughsie/battery-mapping branch please.

closed 2022-02-04 by hughsie