fwupd/fwupd #4253: Failed to load SMBIOS: neither SMBIOS or DT found (Win build)

necros2k7, 2022-02-07:
**Describe the bug**
16:24:39:0605 FuContext            Failed to load SMBIOS: neither SMBIOS or DT found
Loading…                 [-                                      ]16:24:41:0105 FuEngine             failed to probe device usb:02:00:01:03:04: USB error on device 1131:1001 : Entity not found [-5]
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
No detected devices

**Steps to Reproduce**
fwupdtool get-devices --show-all-devices

**Expected behavior**
usual programme execution

**fwupd version information**
1.7.4

- Operating system and version:
- win11 x79 moob
- Have you tried rebooting?
yes

hughsie, 2022-02-07:
> neither SMBIOS or DT found

I don't know how to read the SMBIOS blob or DMI data when using win32 APIs. If you can help that would be awesome. e.g. I'm looking for this data:

	[hughsie@fedora ~]$ sudo fwupdtool hwids 
	Computer Information
	--------------------
	BiosVendor: LENOVO
	BiosVersion: N2VET35W (1.20 )
	BiosMajorRelease: 1
	BiosMinorRelease: 20
	FirmwareMajorRelease: 01
	FirmwareMinorRelease: 09
	Manufacturer: LENOVO
	Family: ThinkPad P1 Gen 3
	ProductName: 20TJS2F44A
	ProductSku: LENOVO_MT_20TJ_BU_Think_FM_ThinkPad P1 Gen 3
	EnclosureKind: a
	BaseboardManufacturer: LENOVO
	BaseboardProduct: 20TJS2F44A

necros2k7, 2022-02-07:
Maybe can be of help:
https://github.com/digitalocean/go-smbios
"Supported operating systems and their SMBIOS retrieval mechanisms include:

Windows (GetSystemFirmwareTable)"

https://docs.microsoft.com/en-us/windows/win32/api/sysinfoapi/nf-sysinfoapi-getsystemfirmwaretable
https://chen-jiao.github.io/articles/2016/11/09/How-to-access-SMBIOS-in-Windows.html
https://stackoverflow.com/questions/60700302/win32-api-to-get-machine-uuid

hughsie, 2022-02-07:
Thanks, those were super useful:

	[hughsie@fedora build-win32 (wip/hughsie/win32-smbios %)]$ ./src/fwupdtool.exe hwids
	Computer Information
	--------------------
	BiosVendor: LENOVO
	...

closed 2022-02-07 by hughsie

superm1, 2022-02-07:
Btw double check the codebase for win32 disabled code. I seem to recall we had something set up to behave differently from smbios not being readable on windows but I don't remember what (and I'm traveling right now without a checkout).

hughsie, 2022-02-07:
> set up to behave differently from smbios not being readable on windows

Yes, I remembered this as well, but couldn't see in the code what we did. Will poke some more.

hughsie, 2022-02-07:
> but that's reading a registry key

But I can fix that now too. Give me 15 mins.