fwupd/fwupd #4259: Vulnerable firmware can't get upgraded: "failed to get device after detach: failed to wait for detach replug: device did not come back" or reverts back to old version after restart

mYnDstrEAm, 2022-02-08:
**Describe the bug**
The issue is described in more detail and partly in chronological order in my question about it [here](https://unix.stackexchange.com/questions/689284/how-to-solve-failed-to-get-device-after-detach-failed-to-wait-for-detach-replu).

I was trying to upgrade the vulnerable Unifying Receiver firmware with `fwupdmgr update` but I'm getting this output on Debian11/KDE:
```
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
failed to get device after detach: failed to wait for detach replug: device {id} did not come back
```
I then installed `solaar` (and I was never prompted to install it on Debian which is probably another issue) and disconnected one of the two receivers and could run the updates (it was two updates) after doing many retries.

Even stranger than this error is that it displayed that it has the latest version afterwards but it's back to the old firmware after a restart.

After rebooting I had to run the command a few times until it got so far as to prompt me about the root password but I still get the same error or (far less often) this error instead: `Message recipient disconnected from message bus without replying`.

Maybe the problem is connected t[o how integrity is ensured for firmware updates](https://unix.stackexchange.com/q/689663/233262), but I suspect the cause to be something else. I'm having this problem with both of the UR dongles.

**Steps to Reproduce**
On Debian11/KDE with Wayland try to update the vulnerable Unifying Receiver firmware.

**Expected behavior**
It should successfully update at the first run of `fwupdmgr update`.

**fwupd version information**

```
client version: 1.5.7
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.7
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt (Apper)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

**Additional questions**

- Operating system and version: Debian 11 (stable; with KDE)
- Have you tried rebooting? Yes
- Is this a regression? No(?)

superm1, 2022-02-28:
Can you run the daemon in verbose mode instead?  You can add `--verbose` to the daemon systemd unit.

mYnDstrEAm, 2022-02-28:
Should have checked if there is a verbose mode. Here's the output:
```
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Fwupd-DEBUG: downloading https://fwupd.org/downloads/e28a4a19f6ffd7e75e755100ead935817290617a-Logitech-Unifying-RQR24.10_B0036.cab
Fwupd-DEBUG: Emitting ::status-changed() [downloading]
Downloading…             [ -                                     ](fwupdmgr:507355): Fwupd-DEBUG: download progress: 10%
Fwupd-DEBUG: download progress: 10%
Downloading…             [***                                    ](fwupdmgr:507355): Fwupd-DEBUG: download progress: 10%
Fwupd-DEBUG: download progress: 10%
[...]
Fwupd-DEBUG: download progress: 100%
Fwupd-DEBUG: Emitting ::status-changed() [idle]
Downloading…             [***************************************]
Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…           [***************************************](fwupdmgr:507355): Fwupd-DEBUG: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
Fwupd-DEBUG: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [***************************************]
Fwupd-DEBUG: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
Fwupd-DEBUG: Emitting ::device-removed([id1])
Fwupd-DEBUG: Emitting ::device-changed([id2])
FuMain-DEBUG: ignoring Unifying Receiver with status unknown
Fwupd-DEBUG: Emitting ::changed()
Fwupd-DEBUG: Emitting ::device-changed([id3])
FuMain-DEBUG: ignoring Unifying Receiver with status idle
failed to get device after detach: failed to wait for detach replug: device [id3] did not come back
```
In this case it did not prompt me for the root password. In earlier tries it did at least after trying it for a few times (haven't done so here). After running this the adapter doesn't work anymore so I need to plug it out and back in.

superm1, 2022-02-28:
That's the client verbose mode.  We need the daemon's verbose output.  Modify the systemd unit and add `--verbose`.

Alternatively you can reproduce it with `sudo fwupdtool update --verbose` which is "standalone" and doesn't use dbus.

mYnDstrEAm, 2022-02-28:
How to modify it? Is there a (full) command?

mYnDstrEAm, 2022-02-28:
And then the logs of `journalctl -n 10`?
It only has this:
```
username : TTY=pts/1 ; PWD=/home/username ; USER=root ; COMMAND=/usr/bin/fwupdtool --verbose
sudo[id]: pam_unix(sudo:session): session opened for user root(uid=0) by username(uid=1000)
sudo[id]: pam_unix(sudo:session): session closed for user root
```
And `sudo fwupdtool --verbose` (before running `fwupdmgr update`):
```
FuDebug              Verbose debugging enabled (on console 1)
Command not found

Use fwupdtool --help for help
```

superm1, 2022-02-28:
`fwupdtool` drops in to replace `fwupd` + `fwupdmgr` just without dbus - it's like an all in one tool.

`sudo fwupdtool update --verbose`

And then provide that output.

mYnDstrEAm, 2022-02-28:
Right, I missed `update` there. It's a long output, don't like to post the entire thing - here's the most relevant part and end of the log (only id1-3 show more than once):
```
already set GType to FuLogitechHidPpBootloaderTexas, ignoring FuLogitechHidPpBootloaderTexas
using [id1] for usb:01:00:05
autodetected HID interface of 0x00
changing version for [id1] : 3.1->RQR24.00_B0000
emit added from logitech_hidpp: [id1] 
fu_plugin_device_registered(pci_bcr)
fu_plugin_device_registered(dell_dock)
fu_plugin_device_registered(thunderbolt)
fu_plugin_device_registered(msr)
fu_plugin_device_registered(uefi_capsule)
ignoring device [id1]  [logitech_hidpp:FuLogitechHidPpBootloaderTexas] existing device [id2] [logitech_hidpp:FuLogitechHidPpRuntime] already exists
ignoring for OR statement: indexed string '[id3]' was unfound
ignoring for OR statement: indexed string '[id4]' was unfound
requirement 1.2.7 ge 1.5.7 -> org.freedesktop.fwupd passed
ignoring Unifying Receiver with status unknown
ignoring Unifying Receiver with status unknown
ignoring for OR statement: indexed string '[id5]' was unfound
ignoring for OR statement: indexed string '[id7]' was unfound
ignoring for OR statement: indexed string '[id8]' was unfound
ignoring for OR statement: indexed string '[id9]' was unfound
ignoring for OR statement: indexed string '[id10]' was unfound
ignoring for OR statement: indexed string '[id11]' was unfound
ignoring for OR statement: indexed string '[id3]' was unfound
requirement 1.2.7 ge 1.5.7 -> org.freedesktop.fwupd passed
ignoring RQR24.06_B0030 == RQR24.06_B0030
ignoring RQR24.05_B0029 < RQR24.06_B0030
ignoring: USB error on device id:id : Entity not found [-5]
using [id3] for HIDRAW\VEN_ID&DEV_ID
ID,ID,ID has no updates, so ignoring device
ignoring Unifying Receiver with status idle
ignoring Unifying Receiver with status idle
modifying device Unifying Receiver [id2]
composite_cleanup(dell_dock)
uninhibiting: update
failed to get device after detach: failed to wait for detach replug: device [id2] did not come back
destroy(synaptics_mst)
destroy(logind)
destroy(upower)
destroy(linux_lockdown)
destroy(tpm_eventlog)
destroy(linux_swap)
destroy(dell)
destroy(flashrom)
destroy(linux_tainted)
destroy(redfish)
destroy(uefi_capsule)
```

superm1, 2022-02-28:
> Right, I missed `update` there. It's a long output, don't like to post the entire thing - here's the most relevant part and end of the log (only id1-3 show more than once):
>
> ```
> already set GType to [...]

can you put it in a gist?  Just in case anything else stands out?

This _looks_ like the problem to me though:

> ignoring: USB error on device id:id : Entity not found [-5]

Any errors in the kernel `dmesg` too around this time?

mYnDstrEAm, 2022-02-28:
Only this:
```
usb 1-5: USB disconnect, device number 7
usb 1-5: new full-speed USB device number 8 using xhci_hcd
usb 1-5: New USB device found, idVendor=046d, idProduct=id, bcdDevice= 3.01
usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
usb 1-5: Product: USB BootLoader
usb 1-5: Manufacturer: Logitech
hid-generic id2: hiddev0,hidraw0: USB HID v1.11 Device [Logitech USB BootLoader] on usb-0000:00:14.0-5/input0
hid-generic id3: hiddev0,hidraw0: USB HID v1.11 Device [Logitech USB BootLoader] on usb-0000:00:14.0-5/input0
usb 1-5: USB disconnect, device number 8
```

superm1, 2022-02-28:
Any chance you can turn on debian testing and pull fwupd from there and see if it happens in that version too?

mYnDstrEAm, 2022-02-28:
No, at least not for a long while (I'd use a virtual machine for that) and there's also no newer version in backports (it's 1.5.7-4).

superm1, 2022-02-28:
You can use apt pinning to pull it from testing can't you?  Anyway the reason I'm asking is because we did make changes after 1.5.7 to the logitech plugin timing.  This might be what you're hitting: https://github.com/fwupd/fwupd/commit/bb87a45af3cf1b8f69654ef5815d4aa3b40863a8

mYnDstrEAm, 2022-04-17:
This problem still occurs, it's an important security issue that isn't fixed on Debian11 and I won't try with apt pinning to prevent package/dependency versions inconsistencies and similar problems.

As explained above, I could upgrade it once but it was back to the old version after a reboot and this issue keeps many people vulnerable.

Once it's fixed it would make sense to [add fwupd and plasma-discover-backend-fwupd as default packages to install on KDE in Debian to prompt users for firmware updates to keep their firmware up-to-date and secure](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1005134) (note that this process also needs to be secure so updates don't introduce new vulnerabilities).

superm1, 2022-04-17:
> and I won't try with apt pinning to prevent package/dependency versions inconsistencies and similar problems.

There are other ways you can try the newer version.  You can get the snap, the flatpak, run it a chroot, or VM.

> As explained above, I could upgrade it once but it was back to the old version after a reboot and this issue keeps many people vulnerable.

This means it didn't really upgrade the first time.

> Once it's fixed, it would make sense to [add fwupd and plasma-discover-backend-fwupd as default packages to install on KDE in Debian to prompt users for firmware updates to keep their firmware [...]

This should be done regardless of this bug or not.

colttt, 2022-05-02:
I had the same issue and can confirm that it works with daemon version 1.8.0, I also have Debian 11 and used the snap version

closed 2022-05-02 by superm1

mYnDstrEAm, 2022-05-05:
Now this needs to get closed by Debian maintainers: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1010490

Rongronggg9, 2023-05-04:
The issue is still unresolved in v1.8.12. I can confirm that v1.5.8+ DID increase the timeout and the timeout is fairly enough (10s vs 1.2s). But fwupd seems unable to recognize the replugged `Logitech USB BootLoader`.

FYI: the M/N of the receiver is `C-U0012` (3mm Unifying Pico Receiver).

@superm1 Could you please reopen the issue before it is really fixed?

```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Unifying Receiver from RQR24.09_B0033 to RQR24.11_B0036?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This release addresses an unspecified encryption-key extraction              ║
║ vulnerability.                                                               ║
║                                                                              ║
║ Unifying Receiver and all connected devices may not be usable while          ║
║ updating.                                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
06:53:00.856 FuMain               locked /run/lock/fwupdtool
06:53:00.860 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Writing…                 [***************************************]06:53:00.860 FuProgress           percentage should not go down from 100 to 75: 0) ../src/fu-tool.c:1246 (16/17)

06:53:00.860 Fwupd                downloading https://fwupd.org/downloads/6a0134766c26c73cc08f3d30f0bb84d94d035b50f702e39506abdcab9a8f658f-Logitech-Unifying-RQR24.11_B0036.cab
06:53:00.860 Fwupd                Emitting ::status-changed() [downloading]
Downloading…             [                                -      ]06:53:02.140 Fwupd                download progress: 100%
06:53:02.140 Fwupd                download progress: 100%
06:53:02.140 Fwupd                download progress: 100%
Downloading…             [***************************************]06:53:03.204 Fwupd                download progress: 7%
06:53:03.204 Fwupd                download progress: 7%
06:53:03.204 Fwupd                download progress: 7%
06:53:03.204 Fwupd                download progress: 7%
Downloading…             [**                                     ]06:53:03.215 Fwupd                download progress: 10%
06:53:03.215 Fwupd                download progress: 10%
Downloading…             [***                                    ] Less than one minute remaining…06:53:03.225 Fwupd                download progress: 14%
06:53:03.225 Fwupd                download progress: 14%
Downloading…             [*****                                  ] Less than one minute remaining…06:53:03.233 Fwupd                download progress: 17%
06:53:03.233 Fwupd                download progress: 17%
Downloading…             [******                                 ] Less than one minute remaining…06:53:03.243 Fwupd                download progress: 21%
06:53:03.243 Fwupd                download progress: 21%
Downloading…             [********                               ]06:53:03.252 Fwupd                download progress: 24%
06:53:03.252 Fwupd                download progress: 24%
Downloading…             [*********                              ]06:53:03.261 Fwupd                download progress: 28%
06:53:03.261 Fwupd                download progress: 28%
Downloading…             [**********                             ]06:53:03.271 Fwupd                download progress: 31%
06:53:03.271 Fwupd                download progress: 31%
Downloading…             [************                           ]06:53:03.280 Fwupd                download progress: 35%
06:53:03.280 Fwupd                download progress: 35%
Downloading…             [*************                          ]06:53:03.289 Fwupd                download progress: 38%
06:53:03.289 Fwupd                download progress: 38%
Downloading…             [**************                         ]06:53:03.298 Fwupd                download progress: 42%
06:53:03.298 Fwupd                download progress: 42%
Downloading…             [****************                       ]06:53:03.306 Fwupd                download progress: 46%
06:53:03.306 Fwupd                download progress: 46%
Downloading…             [*****************                      ]06:53:03.315 Fwupd                download progress: 49%
06:53:03.315 Fwupd                download progress: 49%
Downloading…             [*******************                    ]06:53:03.456 Fwupd                download progress: 53%
06:53:03.456 Fwupd                download progress: 53%
Downloading…             [********************                   ]06:53:03.460 Fwupd                download progress: 56%
06:53:03.460 Fwupd                download progress: 56%
Downloading…             [*********************                  ]06:53:03.467 Fwupd                download progress: 60%
06:53:03.467 Fwupd                download progress: 60%
Downloading…             [***********************                ]06:53:03.474 Fwupd                download progress: 63%
06:53:03.474 Fwupd                download progress: 63%
Downloading…             [************************               ]06:53:03.481 Fwupd                download progress: 67%
06:53:03.481 Fwupd                download progress: 67%
[... cut ...]
[ 2207.066401] usb 1-2: USB disconnect, device number 25
[ 2208.272457] usb 1-2: new full-speed USB device number 26 using xhci_hcd
[ 2208.401194] usb 1-2: New USB device found, idVendor=046d, idProduct=aae5, bcdDevice= 3.02
[ 2208.401207] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2208.401211] usb 1-2: Product: USB BootLoader
[ 2208.401215] usb 1-2: Manufacturer: Logitech
[ 2208.404374] hid-generic 0003:046D:AAE5.0040: hiddev96,hidraw2: USB HID v1.11 Device [Logitech USB BootLoader] on usb-0000:00:14.0-2/input0
```

hughsie, 2023-05-05:
@Rongronggg9 why are you using fwupdtool? Does `fwupdmgr reinstall 279ed287-3607-549e-bacc-f873bb9838c4` work better?

Rongronggg9, 2023-05-06:
@hughsie I used `sudo fwupdtool update --verbose` as suggested in https://github.com/fwupd/fwupd/issues/4259#issuecomment-1054407345. Before that, I had tried `fwupdmgr update`, which threw the same error.

```console
$ fwupdmgr reinstall 279ed287-3607-549e-bacc-f873bb9838c4
Selected device: Unifying Receiver
Unable to locate release RQR24.09_B0033 for Unifying Receiver
$ fwupdmgr reinstall fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
Selected device: Unifying Receiver
Unable to locate release RQR24.09_B0033 for Unifying Receiver
$ fwupdmgr reinstall cc4cbfa9-bf9d-540b-b92b-172ce31013c1
Selected device: Unifying Receiver
Unable to locate release RQR24.09_B0033 for Unifying Receiver
```

hughsie, 2023-05-06:
You might need to do `fwupdmgr enable-remote lvfs-testing` to get the older firmware.

Rongronggg9, 2023-05-06:
@hughsie Uhh, the fact is that [`RQR24.11_B0036`](https://fwupd.org/lvfs/devices/com.logitech.Unifying.RQR24.signed.firmware) is only available in `lvfs-tesing` so another fact is I've always had it enabled... The older firmware was just never released to LVFS.

I have several `C-U0008` Unifying receivers (6mm nano) and they were produced with `RQR24.11_B0036` installed. I randomly picked one and try to reinstall the firmware, only resulting in another error occurring.

```console
$ sudo fwupdtool reinstall fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027 --verbose

<...omitted...>

16:37:44.404 Fwupd                Emitting ::status-changed() [idle]
16:37:44.404 Fwupd                status-code was 200
16:37:44.413 FuCommon             reading /root/.cache/fwupd/6a0134766c26c73cc08f3d30f0bb84d94d035b50f702e39506abdcab9a8f658f-Logitech-Unifying-RQR24.11_B0036.cab with 75384 bytes
Decompressing…           [ -                                     ]16:37:44.417 Jcat                 using gpgme v1.18.0
16:37:44.459 Jcat                 Using engine at /var/lib/fwupd/gnupg
16:37:44.465 Jcat                 Adding GnuPG public key /etc/pki/fwupd/GPG-KEY-Linux-Foundation-Firmware
16:37:44.473 Jcat                 importing key F2F0325648E1AE4956710198C6361787A0A849E1 [0] Success
16:37:44.473 Jcat                 ignoring LVFS-CA.pem as not GPG public key
16:37:44.473 Jcat                 Adding GnuPG public key /etc/pki/fwupd/GPG-KEY-Linux-Vendor-Firmware-Service
16:37:44.479 Jcat                 importing key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2 [0] Success
16:37:44.479 Jcat                 Adding GnuPG public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata
16:37:44.485 Jcat                 importing key CAE0420259890141E701156E0A6DCEF6B46F3DBF [0] Success
16:37:44.485 Jcat                 ignoring LVFS-CA.pem as not GPG public key
16:37:44.485 Jcat                 Adding GnuPG public key /etc/pki/fwupd-metadata/GPG-KEY-Linux-Vendor-Firmware-Service
16:37:44.490 Jcat                 importing key 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2 [0] Success
16:37:44.495 Jcat                 returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
16:37:44.495 Jcat                 verified: JcatResult:
  Timestamp:             2022-12-20T15:26:09Z
  Authority:             3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
  JcatGpgEngine:
    Kind:                gpg
    VerifyKind:          signature

16:37:44.496 Jcat                 got 1 PKCS7 signatures
16:37:44.496 Jcat                 verified: JcatResult:
  Timestamp:             2022-12-20T15:26:09Z
  Authority:             O=Linux Vendor Firmware Project,CN=LVFS CA
  JcatPkcs7Engine:
    Kind:                pkcs7
    VerifyKind:          signature

16:37:44.496 FuCabinet            verified metadata com.logitech.Unifying.RQR24.metainfo.xml: 4
16:37:44.496 FuCabinet            processing file: com.logitech.Unifying.RQR24.metainfo.xml
16:37:44.496 FuCabinet            processing release: RQR24.11_B0036
16:37:44.502 Jcat                 returned signature fingerprint 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
16:37:44.502 Jcat                 verified: JcatResult:
  Timestamp:             2022-12-20T15:26:09Z
  Authority:             3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
  JcatGpgEngine:
    Kind:                gpg
    VerifyKind:          signature

16:37:44.502 Jcat                 got 1 PKCS7 signatures
16:37:44.503 Jcat                 verified: JcatResult:
  Timestamp:             2022-12-20T15:26:09Z
  Authority:             O=Linux Vendor Firmware Project,CN=LVFS CA
  JcatPkcs7Engine:
    Kind:                pkcs7
    VerifyKind:          signature

16:37:44.503 FuCabinet            verified payload RQR24.11_B0036.shex: 4
16:37:44.503 FuMain               loading release failed on 59a11bc1795b6b83caa7eface68d0ad302cf5f86:com.logitech.Unifying.RQR24.signed.firmware failed: No supported devices found
16:37:44.503 FuMain               loading release failed on b0d4430dfa6bde9f0c22680df36dbc8c15c80753:com.logitech.Unifying.RQR24.signed.firmware failed: No supported devices found
16:37:44.504 FuMain               loading release failed on 5792b48846ce271fab11c4a545f7a3df0d36e00a:com.logitech.Unifying.RQR24.signed.firmware failed: No supported devices found
16:37:44.504 FuMain               loading release failed on 4bde70ba4e39b28f9eab1628f9dd6e6244c03027:com.logitech.Unifying.RQR24.signed.firmware failed: No supported devices found
16:37:44.504 FuMain               loading release failed on 04e17fcf7d3de91da49a163ffe4907855c3648be:com.logitech.Unifying.RQR24.signed.firmware failed: No supported devices found
16:37:44.504 FuMain               loading release failed on a45df35ac0e948ee180fe216a5f703f32dda163f:com.logitech.Unifying.RQR24.signed.firmware failed: No supported devices found
16:37:44.504 FuMain               loading release failed on c6a80ac3a22083423992a3cb15018989f37834d6:com.logitech.Unifying.RQR24.signed.firmware failed: No supported devices found
16:37:44.504 FuMain               loading release failed on 6924110cde4fa051bfdc600a60620dc7aa9d3c6a:com.logitech.Unifying.RQR24.signed.firmware failed: No supported devices found
16:37:44.505 FuMain               loading release failed on 362301da643102b9f38477387e2193e57abaa590:com.logitech.Unifying.RQR24.signed.firmware failed: No supported devices found
16:37:44.505 FuEngine             requirement 1.3.1 ge 1.8.12 -> org.freedesktop.fwupd passed
[... 232 lines cut ...]
[... cut ...]
[ 5321.186401] usb 1-1: Product: USB BootLoader
[ 5321.186405] usb 1-1: Manufacturer: Logitech
[ 5321.191116] hid-generic 0003:046D:AAAC.001E: hiddev96,hidraw2: USB HID v1.11 Device [Logitech USB BootLoader] on usb-0000:00:14.0-1/input0
[ 5321.802422] hid-generic 0003:046D:AAAC.001F: hiddev96,hidraw2: USB HID v1.11 Device [Logitech USB BootLoader] on usb-0000:00:14.0-1/input0
[ 5338.813830] hid-generic 0003:046D:AAAC.0020: hiddev96,hidraw2: USB HID v1.11 Device [Logitech USB BootLoader] on usb-0000:00:14.0-1/input0
```

hughsie, 2023-05-06:
Why are you using fwupdtool? Does fwupdmgr work?

Rongronggg9, 2023-05-07:
@hughsie I repeat:

> I used `sudo fwupdtool update --verbose` **as suggested in https://github.com/fwupd/fwupd/issues/4259#issuecomment-1054407345**. Before that, I had tried `fwupdmgr update`, **which threw the same error.**

In case you misunderstand me that I am in favor of `fwupdtool`, I declare:

I prefer to use `fwupdmgr` all the time. The reason for using `fwupdtool` is shown above. **All of my tests were done with both `fwupdmgr` and `fwupdtool`, which show no difference since they throw the same error for the same device.**

<hr>

I've just found the root cause of the bug: the bootloader of `C-U0012` is `046D:AAE5`, which cannot be recognized by fwupd.

hughsie, 2023-05-07:
> : the bootloader of C-U0012 is 046D:AAE5

But fwupd says it's USB\VID_046D&PID_AAAC

hughsie, 2023-05-07:
Can you try running `sudo /usr/libexec/fwupd/fwupd --verbose --verbose` in one terminal and `fwupdmgr reinstall` in the other please. Then attach the entire daemon log, thanks.

Rongronggg9, 2023-05-07:
> > : the bootloader of C-U0012 is 046D:AAE5
> 
> But fwupd says it's USB\VID_046D&PID_AAAC

In https://github.com/fwupd/fwupd/issues/4259#issuecomment-1537178661, I've said that it is `C-U0008` so of course its bootloader is 046D:AAAC.

In https://github.com/fwupd/fwupd/issues/4259#issuecomment-1534261701, I've said that it is `C-U0012` and from `dmesg` we can find that its bootloader is 046D:AAE5.

Rongronggg9, 2023-05-07:
> Can you try running `sudo /usr/libexec/fwupd/fwupd --verbose --verbose` in one terminal and `fwupdmgr reinstall` in the other please. Then attach the entire daemon log, thanks.

I think it is no further need to reinstall `RQR24.11_B0036` on my `C-U0008`. The bug in https://github.com/fwupd/fwupd/issues/4259#issuecomment-1537178661 occurs in ~50% of my tries (otherwise they just succeed). If failed, fwupd can still recognize the bootloader (`get-devices` shows the bootloader) so executing `fwupdmgr upgrade` or `sudo fwupdtool upgrade` can finish the reinstallation.

But the case of `C-U0012` is completely different, once fwupd makes it entered the DFU mode, the bootloader is not recognized and there is no Logitech device shown in `get-devices`.

hughsie, 2023-05-07:
Okay, no worries then. Please open a PR if you work out the fix.