fwupd/fwupd #4262: Dell WD19 upgrade issue

zsolt-donca, 2022-02-08:
**Describe the bug**
I tried to update the firmware of my system, and it fails to update of my Dell Thunderbolt controller with `device xxx did not come back`.

**Steps to Reproduce**
I run `sudo fwupdmgr update`:

```
❯ sudo fwupdmgr update
[sudo] password for zdonca: 
Devices with no available firmware updates: 
 • TPM 2.0
 • USB 2.0 BILLBOARD
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Package level of Dell dock from 01.00.21.00 to 01.00.21.01?          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release updated for the following changes:                       ║
║                                                                              ║
║ • Tool version updates to address security vulnerabilities.                  ║
║                                                                              ║
║ This firmware is provided by LVFS community members and is not provided (or  ║
║ supported) by the original hardware vendor.                                  ║
║                                                                              ║
║ Installing this update may also void any device warranty.                    ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall RTS5413 in Dell dock to 01.21?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release updated for the following changes:                       ║
║                                                                              ║
║ • Tool version updates to address security vulnerabilities.                  ║
║                                                                              ║
║ This firmware is provided by LVFS community members and is not provided (or  ║
║ supported) by the original hardware vendor.                                  ║
║                                                                              ║
║ Installing this update may also void any device warranty.                    ║
║                                                                              ║
║ RTS5413 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall RTS5487 in Dell dock to 01.47?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release updated for the following changes:                       ║
║                                                                              ║
║ • Tool version updates to address security vulnerabilities.                  ║
║                                                                              ║
║ This firmware is provided by LVFS community members and is not provided (or  ║
║ supported) by the original hardware vendor.                                  ║
║                                                                              ║
║ Installing this update may also void any device warranty.                    ║
║                                                                              ║
║ RTS5487 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall VMM5331 in Dell dock to 05.04.06?                                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release updated for the following changes:                       ║
║                                                                              ║
║ • Tool version updates to address security vulnerabilities.                  ║
║                                                                              ║
║ This firmware is provided by LVFS community members and is not provided (or  ║
║ supported) by the original hardware vendor.                                  ║
║                                                                              ║
[... 1501 lines cut ...]
Feb 08 16:19:48 ROC01-HYMT373 fwupd[3201]: 14:19:48:0021 FuMain               Emitting PropertyChanged('Percentage'='99%')
Feb 08 16:19:48 ROC01-HYMT373 fwupd[3201]: 14:19:48:0021 FuMain               Emitting PropertyChanged('Percentage'='100%')
Feb 08 16:19:55 ROC01-HYMT373 fwupd[3201]: 14:19:55:0271 FuHistory            modifying device Thunderbolt controller in Dell dock [ae5f488dbb89d6b056dae7ba9ee999cdf8140f76]
Feb 08 16:19:55 ROC01-HYMT373 fwupd[3201]: 14:19:55:0275 FuEngine             Emitting DeviceRequest('Message'='There is no change.')
Feb 08 16:19:55 ROC01-HYMT373 fwupd[3201]: 14:19:55:0275 FuHistory            modifying device Thunderbolt controller in Dell dock [ae5f488dbb89d6b056dae7ba9ee999cdf8140f76]
Feb 08 16:19:55 ROC01-HYMT373 fwupd[3201]: 14:19:55:0279 FuHistory            modifying device Thunderbolt controller in Dell dock [ae5f488dbb89d6b056dae7ba9ee999cdf8140f76]
Feb 08 16:19:55 ROC01-HYMT373 fwupd[3201]: 14:19:55:0283 FuHistory            modifying device Thunderbolt controller in Dell dock [ae5f488dbb89d6b056dae7ba9ee999cdf8140f76]
Feb 08 16:19:55 ROC01-HYMT373 fwupd[3201]: 14:19:55:0283 FuMain               Emitting PropertyChanged('Status'='device-restart')
Feb 08 16:19:55 ROC01-HYMT373 fwupd[3201]: 14:19:55:0283 FuDeviceList         plugin did not specify a remove delay, so guessing we should wait 10000ms for replug
Feb 08 16:20:05 ROC01-HYMT373 fwupd[3201]: 14:20:05:0284 FuHistory            modifying device Thunderbolt controller in Dell dock [ae5f488dbb89d6b056dae7ba9ee999cdf8140f76]
Feb 08 16:20:05 ROC01-HYMT373 fwupd[3201]: 14:20:05:0292 FuHistory            modifying device Thunderbolt controller in Dell dock [ae5f488dbb89d6b056dae7ba9ee999cdf8140f76]
Feb 08 16:20:05 ROC01-HYMT373 fwupd[3201]: 14:20:05:0299 FuHistory            modifying device Thunderbolt controller in Dell dock [ae5f488dbb89d6b056dae7ba9ee999cdf8140f76]
Feb 08 16:20:05 ROC01-HYMT373 fwupd[3201]: 14:20:05:0307 FuPlugin             composite_cleanup(dell_dock)
Feb 08 16:20:05 ROC01-HYMT373 fwupd[3201]: 14:20:05:0307 FuPluginDellDock     activating passive flow (6) for WD19TB
Feb 08 16:20:05 ROC01-HYMT373 fwupd[3201]: 14:20:05:0310 FuPlugin             composite_cleanup(thunderbolt)
Feb 08 16:20:05 ROC01-HYMT373 fwupd[3201]: 14:20:05:0310 FuIdle               uninhibiting: update

```

**Additional questions**

- Operating system and version: Arch Linux, updated just today.
- Have you tried unplugging the dock or any peripherals from your machine? A: Tried unplugging the monitor, resulting in pretty much the same behavior.
- Have you tried to power cycle the dock from the AC adapter? A: Yes, I unplugged and replugged the dock several times, no effect.
- Is this a regression? A: Not sure.

hughsie, 2022-02-08:
> plugin did not specify a remove delay, so guessing we should wait 10000ms for replug

Is this the official 1.7.5 tarball or some snapshot taken before the release? That looks very much like https://github.com/fwupd/fwupd/commit/915c4805f5790ad1b464f2f01ac9c8c40b6ad786 but that was included in the tarball.

zsolt-donca, 2022-02-09:
I am using the [Arch community package](https://archlinux.org/packages/community/x86_64/fwupd/), which is at version `1.7.5-1`. Its [PKGBUILD file](https://github.com/archlinux/svntogit-community/blob/packages/fwupd/trunk/PKGBUILD) downloads https://people.freedesktop.org/~hughsient/releases/fwupd-1.7.5.tar.xz, which seems to indeed contain https://github.com/fwupd/fwupd/commit/915c4805f5790ad1b464f2f01ac9c8c40b6ad786.

superm1, 2022-02-09:
> Is this the official 1.7.5 tarball or some snapshot taken before the release? That looks very much like https://github.com/fwupd/fwupd/commit/915c4805f5790ad1b464f2f01ac9c8c40b6ad786 but that was [...]

I don't think wd19 is supposed to replug at all. It's supposed to be a delayed activation.

CragW, 2022-02-09:
I think we don't need `FWUPD_DEVICE_FLAG_WAIT_FOR_REPLUG` for WD19, it was added in e03b8c4. The NVM prevents me to downgrade to an older version e.g.: from 60 to 43, but the `update` process was completed successfully.

superm1, 2022-02-09:
Can you please send up a PR to fix it up?

closed 2022-02-09 by hughsie

zsolt-donca, 2022-02-11:
I tried the latest development builds, and there is still something wrong with the update process, even though it produces no obvious error message (on it standard output). I get the same updates after I restart my computer, therefore I suspect that the the firmware update process doesn't do anything for me.

If told, I can open a separate issue to track this.

I am now using https://aur.archlinux.org/packages/fwupd-git, at the moment using a build based on https://github.com/fwupd/fwupd/commit/73377b2a530ec23e3712c1cf47f99e8885a18707 (which includes the above fix and some other commits as well).

There are messages starting with "failed to add device" below in the verbose daemon logs, I guess those are relevant.

The update process produces this output:
```
❯ sudo fwupdmgr update
[sudo] password for zdonca: 
Devices with no available firmware updates: 
 • TPM 2.0
 • USB 2.0 BILLBOARD
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Package level of Dell dock from 01.00.21.00 to 01.00.21.01?          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release updated for the following changes:                       ║
║                                                                              ║
║ • Tool version updates to address security vulnerabilities.                  ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall RTS5413 in Dell dock to 01.21?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release updated for the following changes:                       ║
║                                                                              ║
║ • Tool version updates to address security vulnerabilities.                  ║
║                                                                              ║
║ RTS5413 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall RTS5487 in Dell dock to 01.47?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release updated for the following changes:                       ║
║                                                                              ║
║ • Tool version updates to address security vulnerabilities.                  ║
║                                                                              ║
║ RTS5487 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall VMM5331 in Dell dock to 05.04.06?                                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release updated for the following changes:                       ║
║                                                                              ║
║ • Tool version updates to address security vulnerabilities.                  ║
║                                                                              ║
║ VMM5331 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall WD19TB to 01.00.01.01?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release updated for the following changes:                       ║
║                                                                              ║
║ • Tool version updates to address security vulnerabilities.                  ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
[... 1509 lines cut ...]
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0940 FuEngine             ignoring 01.21 == 01.21
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0941 FuMain               Called GetUpgrades(96969b82d977fdcbf607df084ef0dcf10119409b)
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0942 FuEngine             ignoring 01.47 == 01.47
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0942 FuEngine             ignoring 01.47 == 01.47
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0944 FuMain               Called GetUpgrades(228ac6846b64d532dd1ea1b7e651e8f5f55fbc34)
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0945 FuEngine             ignoring 05.04.06 == 05.04.06
[... cut ...]
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0952 FuEngine             ignoring 1.12.0 < 1.15.0
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0952 FuEngine             ignoring 1.11.0 < 1.15.0
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0953 FuMain               Called GetUpgrades(ae5f488dbb89d6b056dae7ba9ee999cdf8140f76)
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0954 FuMain               Called GetUpgrades(e14cf540d178c63765dd0c2e8ac94ae1df40ba48)
Feb 11 23:29:21 ROC01-HYMT373 fwupd[3367]: 21:29:21:0956 FuEngine             ignoring 46.00 == 46.00
```

superm1, 2022-02-11:
>  I get the same updates after I restart my computer, therefore I suspect that the the firmware update process doesn't do anything for me.

Have you unplugged the dock from your PC (not the wall) since you restarted?  It's supposed to apply the updates when unplugged.

zsolt-donca, 2022-02-14:
> Have you unplugged the dock from your PC (not the wall) since you restarted? It's supposed to apply the updates when unplugged.

You are right, that was it. After I unplugged the dock from my PC, it immediately spun up its fans for a couple of minutes. I left it like that for the weekend, and now it seems to have updated itself, there are no longer pending updates.

Would it be possible to print some kind of an informative message after the update to let the user know that they need to unplug the dock from their computer in order for the update to complete?

hughsie, 2022-02-14:
> Would it be possible to print some kind of an informative message after the update to let the user know that they need to unplug the dock from their computer

Ahh, we should have done that, but the update "failing" [from fwupd's point of view] meant we didn't show it. Could you try fwupd with this patch applied please? https://github.com/fwupd/fwupd/commit/f7b6fa2e829b5ab95ee1e37fd06ecce3bdf39928

zsolt-donca, 2022-02-14:
> Could you try fwupd with this patch applied please? https://github.com/fwupd/fwupd/commit/f7b6fa2e829b5ab95ee1e37fd06ecce3bdf39928

I'll try to downgrade the firmware and upgrade again. I'll let you know if I succeed and how it goes.

zsolt-donca, 2022-02-14:
I updated to https://github.com/fwupd/fwupd/commit/55bd4862d3ad8f0c7bf92dc6d7732afcad6e47a8. The downgrade was surprisingly easy. Both the downgrade and the update afterwards printed the following message:

```
Updates will complete after the dock is unplugged. The dock will be unavailable for approximately 1 minute.
```

So I guess this is settled, then.

Respect to the developers of `fwupd` for this amazing tool and for the support! Keep up the good work!

PR #4272 (2022-02-09): trivial: don't replug if thunderbolt device supports auth on disconnect
Fixes: #4262

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit f7b6fa2e (2022-02-09): trivial: don't replug if thunderbolt device supports auth on disconnect

Fixes: #4262