fwupd/fwupd #4267: [ThinkPad X1 Carbon Gen 9] failed to verify System Firmware: For System Firmware 0.1.51 expected ...

arno01, 2022-02-09:
**Describe the question**

```shell
root@x1gen9:~# fwupdmgr verify
Choose a device:
0.	Cancel
1.	0d5d05911800242bb1f35287012cdcbd9b381148 (Prometheus)
2.	dcd4118aa968110eaae58bf95432c9736be3a74e (System Firmware)
2
failed to verify System Firmware: For System Firmware 0.1.51 expected af3218af1484c0a9395fbc59a4190a181f6c7582|a5bc5b90f2b1117ee4c58d2a54a4eeaa0d5c90fe, got 46edb020ca3caadf14c878f6329f953f9b9aa478|76d832469d0239d7dd2dee918ce767add76c6fbf0b5171c1ee57b83b17f1cb99
```

**fwupd version information**

```shell
root@x1gen9:~# fwupdmgr --version
client version:	1.5.11
compile-time dependency versions
	gusb:	0.3.5

daemon version:	1.5.11

root@x1gen9:~# dpkg -l fwupd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version         Architecture Description
+++-==============-===============-============-=================================
ii  fwupd          1.5.11-0ubuntu2 amd64        Firmware update daemon
```

hughsie, 2022-02-09:
Can you attach the output of `fwupdmgr get-devices` please.

hughsie, 2022-02-09:
Lenovo specified the PCR0s of af3218af1484c0a9395fbc59a4190a181f6c7582 and a5bc5b90f2b1117ee4c58d2a54a4eeaa0d5c90fe for that firmware and model, but it appears your PCR0 isn't that for some reason. Also, (for Lenovo only) looking at https://fwupd.org/lvfs/components/10953/checksums it appears the PCR0 is not predictable *at all* -- @mrhpearson is there anything you can discover here? The PCR0 is supposed to be stable for a given system BIOS version...

arno01, 2022-02-09:
Oh, and the CPU voltage seem to have dropped from `1.0 V` down to `0.8 V` after the latest `fwupdmgr update`:

```
root@x1gen9:~# diff -Nur dmidecode.after.mobo.replacement.plus.fwupd-3rd-run dmidecode.after.mobo.replacement.plus.fwupd-4th-run 
--- dmidecode.after.mobo.replacement.plus.fwupd-3rd-run	2022-02-09 14:25:10.272641751 +0100
+++ dmidecode.after.mobo.replacement.plus.fwupd-4th-run	2022-02-09 14:40:28.843897855 +0100
@@ -485,7 +485,7 @@
 		TM (Thermal monitor supported)
 		PBE (Pending break enabled)
 	Version: 11th Gen Intel(R) Core(TM) i7-1185G7 @ 3.00GHz
-	Voltage: 1.0 V
+	Voltage: 0.8 V
 	External Clock: 100 MHz
 	Max Speed: 3000 MHz
 	Current Speed: 3000 MHz
@@ -740,7 +740,7 @@
 Handle 0x0035, DMI type 219, 106 bytes
 OEM-specific Type
 	Header and Data:
-		DB 6A 35 00 01 04 01 45 02 00 A4 06 81 85 38 30
+		DB 6A 35 00 01 04 01 45 02 00 90 06 81 85 3B 30
 		00 00 00 00 40 00 00 03 1F 00 00 C9 03 40 C4 02
 		FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
 		FF FF FF FF FF FF FF FF 03 00 00 00 80 00 00 00
@@ -863,8 +863,8 @@
 Handle 0x0045, DMI type 141, 30 bytes
 OEM-specific Type
 	Header and Data:
-		8D 1E 45 00 54 48 4E 4B 00 00 50 00 00 00 76 2F
-		4E 01 00 00 00 00 06 CE 85 04 00 00 00 00
+		8D 1E 45 00 54 48 4E 4B 00 00 50 00 00 00 DB 94
+		4E 01 00 00 00 00 DF DF 85 04 00 00 00 00
 
 Handle 0x0046, DMI type 140, 15 bytes
 ThinkPad Embedded Controller Program
```

Not sure if that's expected.

mrhpearson, 2022-02-14:
Sorry for the slow reply - inbox got the better of me.
I can reproduce the issue and have flagged it to the FW team for investigation. Internal ticket LO-1576
Thanks!
Mark