fwupd/fwupd #4281: No updates (or warnings) with lid is closed

iyanmv, 2022-02-14:
**Describe the bug**
Some updates cannot be installed when the lid is closed but no warnings or messages are shown to the user. I learned this the hard way, when I was about to open a bug report at the [lenovo firmware repo](https://github.com/fwupd/firmware-lenovo).

Most of the times I have my laptop with the lid closed and connected to a dock station, so running
```fwupdmgr refresh && fwupdmgr get-updates```
was not showing me anything.

Only when I did a `fwupdmgr get-devices` I saw a few items with the message `Update Error:       Cannot install update when the lid is closed`

**Steps to Reproduce**
With a Lenovo Thinkpad X1 Carbon Gen 9 / Yoga Gen 6 run `fwupdmgr refresh && fwupdmgr get-updates` with the lid closed

**Expected behavior**
Running `fwupdmgr refresh && fwupdmgr get-updates` should so all available updates, no matter if they cannot be installed at that time for whatever reason (no AC cord, lid closed, etc.) and get a `No updates available for remaining devices`. Open the lid, run the same command and get available updates.

**fwupd version information**
```
runtime   org.freedesktop.fwupd         1.7.5
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.16.8-arch1-1
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```
20XYCTO1WW
│
├─ELAN0672:00 04F3:3187:
│     Device ID:          0867c07f040e810fa5ea153981e63e324c29eb1f
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             HIDRAW:0x04F3
│     GUIDs:              940805ff-ca7a-56b2-af86-32e995004af9
│                         ab16bce2-e219-52f5-89a6-fd6b1c4244c0
│                         e4c6161e-2b1c-59cb-ad07-c8fb7feb155e
│                         4b808d27-c833-50cc-8177-9a401f5c08fe
│                         5b73c06e-3dfd-5e6c-b06a-24e3489206c2
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─WD19TB:
│ │   Device ID:          570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:            High performance dock
│ │   Current version:    01.00.01.01
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │ 
│ ├─Package level of Dell dock:
│ │     Device ID:        073624c16dd99abe01ba1da223a70321e4f29beb
│ │     Summary:          A representation of dock update status
│ │     Current version:  01.00.21.01
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 5 seconds
│ │     GUID:             8ceeeffd-51b6-580c-9b75-69143227aff8
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
[... 339 lines cut ...]
│   
├─USB4 host controller:
│     Device ID:          9af653bcb4168a85674ba52e65e741749a99a5d1
│     Summary:            Unmatched performance for high-speed I/O
│     Update Error:       No vendor ID set
│     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Updatable
│   
└─USB4 host controller:
      Device ID:          c75d9c46b6a28479934f3b6457240e29a771762b
      Summary:            Unmatched performance for high-speed I/O
      Update Error:       No vendor ID set
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c
      Device Flags:       • Internal device
                          • System requires external power source
                          • Updatable
```

**Additional questions**

- Operating system and version: Arch Linux.
- Have you tried rebooting? Yes.
- Is this a regression? I don't think so.

hughsie, 2022-02-14:
> fwupdmgr get-updates

Ya, this has confused *me* in the past too. Maybe we should show the devices with inhibits here too? Let me mock something up quickly.

iyanmv, 2022-02-14:
That was fast! Thanks!

iyanmv, 2022-02-15:
So with the last changes we will see when we have pending updates but are blocked for some reason, but why are they blocked in the first place? Why can't these updates be installed with the lid close?

hughsie, 2022-02-15:
> Why can't these updates be installed with the lid close

Some OEMs have flash programs that do the wrong thing when they are started with the lid closed and docked. e.g. going into suspend at the wrong time when the dock powers down, or just refusing to power the external screen back on after the update has started which lead people to think that start-up had hung and they forced rebooted -> bad!

iyanmv, 2022-02-15:
Oh, I see, thanks for the clarification.

evgeni, 2022-02-21:
I was just about to open a similar issue (but with AC-required updates, not lid-open ones).

Thanks for reporting and fixing!

iyanmv, 2022-03-24:
This is still a problem for me because I only get the full information with `get-devices` when update is allowed, not if it's available.

### What I expect:

No matter if lid is closed, if battery is low, if AC is not plugged, etc. I would like to know if there are any available updates for my device. They should be listed under the `Devices not currently updatable` list

### How it is right now:

If lid is closed I get
```
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB4 host controller
 • USB4 host controller
 • ELAN0672:00 04F3:3187
 • Integrated Camera
 • UEFI dbx
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19TB
 • MZVL2512HCJQ-00BL7
 • Thunderbolt controller in Dell dock
Devices not currently updatable:
 • Embedded Controller — Cannot install update when the lid is closed
 • Intel Management Engine — Cannot install update when the lid is closed
 • System Firmware — Cannot install update when the lid is closed
 • UEFI Device Firmware — Cannot install update when the lid is closed
No updates available
```
Notice the last sentence **No updates available** (which is a false statement).

And when I open the lid

```
Devices with no available firmware updates: 
 • USB4 host controller
 • USB4 host controller
 • ELAN0672:00 04F3:3187
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19TB
 • Embedded Controller
 • MZVL2512HCJQ-00BL7
 • System Firmware
 • Thunderbolt controller in Dell dock
 • UEFI Device Firmware
20XYCTO1WW
│
└─Intel Management Engine:
  │   Device ID:          a083ebc5138e5e071ef7270cc9a8280722cc7adf
  │   Summary:            UEFI ESRT device
  │   Current version:    240.23.1706
  │   Minimum Version:    0.0.1
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              03c27e00-8229-4c14-ba40-2afe2276555c
  │                       0b9fe299-7637-58b4-a175-556f8777bfd1 ← UEFI\RES_{03C27E00-8229-4C14-BA40-2AFE2276555C}
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Yoga Gen 6/ThinkPad X1 Carbon Gen 9:
        New version:      240.35.1951
        Remote ID:        lvfs
        Release ID:       11857
        Summary:          Lenovo ThinkPad X1 Yoga Gen 6/X1 Carbon Gen 9 Consumer ME Firmware
        License:          Proprietary
        Size:             4.0 MB
        Created:          2016-07-08
        Urgency:          High
        Details:          https://pcsupport.lenovo.com/de/en/search?query=N32RG24W
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        • 0 Intel Platform Update 2021.2 Release
        
        Version 15.0.35.1951 (LVFS: 240.35.1951)
        
        Problem Fixes
        
        • Mitigated the following security vulnerabilities under issues. Please see fixed issues for details.
        • Fixed issue where the console system could not connect to remote machine (client system AMT Web IP is 0.0.0.0) after restart system for vPro
        
        Note: Please suspend or disable any disk encryption tool prior to update.
```

iyanmv, 2022-03-24:
Should I open a new issue or can you reopen this one?

iyanmv, 2022-03-24:
The usecase is the following. Most of the times I use my laptop attached to a docking station with the lid closed. Using Discover (on Plasma) or manually running `fwupdmgr refresh && fwupdmgr get-updates` does not show any available updates. I have to remember to open the lid and re-run the previous command to see that, in fact, they are available updates.

Another usecase. Most people (I assume) will also use their laptops not directly attached to AC all the times. They will not realize updates are available until they use their laptop plugged.

In any of those scenarios, I would want to know that they are indeed available updates, and what I need to do to be able to install them (open lid, plug to AC, etc.)

Do you agree? At least, you have to agree that the message that is shown now (`No updates available`) is wrong.

superm1, 2022-03-24:
It almost sounds like we want an `inhibited updates` section in the `get-updates` command that explains why certain updates aren't allowed to me.

iyanmv, 2022-03-24:
Why is there such a distinction in the first place? Only `update` should care about if an update is inhibited or not, no?

heldchen, 2022-06-05:
came here to report / suggest the same thing. I don't mind that the (lenovo) system firmware is not upgradable with the lid closed, but what I'd like to see is at least some info/warning output in `get-updates` that there are upgrades waiting that can't be installed in the current state - my notebook lid is always closed as I'm using it on docking stations all the time.

iyanmv, 2022-06-14:
@heldchen probably you are not using their latest release. I can confirm that this issue is solved, and now when you run `get-updates` you get all available ones even if you cannot install them, together with the reason (lid close, low battery, etc.) in bold red.

iyanmv, 2023-05-10:
I'm sorry to write here again, but this is still an issue for me with the latest version. Is it okay to keep the conversation here or should I open an new issue?

With laptop not connected to AC:

```
$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • ELAN0672:00 04F3:3187
 • Integrated Camera
 • Internal SPI Controller
 • Prometheus IOTA Config
Devices with the latest available firmware version:
 • Battery
 • Embedded Controller
 • Intel Management Engine
 • MZVL2512HCJQ-00BL7
 • System Firmware
 • UEFI Device Firmware
 • Prometheus
No updates available

$ fwupdmgr update     
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • ELAN0672:00 04F3:3187
Devices with the latest available firmware version:
 • Battery
 • Embedded Controller
 • Integrated Camera
 • Intel Management Engine
 • Internal SPI Controller
 • MZVL2512HCJQ-00BL7
 • Prometheus IOTA Config
 • System Firmware
 • UEFI Device Firmware
 • Prometheus
```

And laptop connected to AC:

```
fwupdmgr get-updates
Devices with no available firmware updates: 
 • ELAN0672:00 04F3:3187
 • Integrated Camera
 • Internal SPI Controller
 • Prometheus IOTA Config
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Battery
 • Embedded Controller
 • Intel Management Engine
 • MZVL2512HCJQ-00BL7
 • Prometheus
 • UEFI Device Firmware
LENOVO 20XYCTO1WW
│
└─System Firmware:
  │   Device ID:          a083ebc5138e5e071ef7270cc9a8280722cc7adf
  │   Summary:            UEFI ESRT device
  │   Current version:    0.1.59
  │   Minimum Version:    0.1.39
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              14f3350e-cf63-4e68-a0d9-0af1d5389a17
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkPad X1 Carbon 9th / X1 Yoga 6th:
        New version:      0.1.60
        Remote ID:        lvfs
        Release ID:       29578
        Summary:          Lenovo ThinkPad X1 Carbon 9th / X1 Yoga 6th System Firmware
        License:          Proprietary
        Size:             18.3 MB
        Created:          2023-04-06
        Urgency:          High
        Tested by Lenovo:
          Tested:         2023-04-12
          Distribution:   debian 11
          Old version:    0.1.59
          Version[fwupd]: 1.5.7
        Tested by Lenovo:
          Tested:         2023-04-12
          Distribution:   ubuntu 22.04
          Old version:    0.1.59
          Version[fwupd]: 1.7.5
        Tested by Lenovo:
          Tested:         2023-04-12
          Distribution:   rhel 9.0
          Old version:    0.1.58
          Version[fwupd]: 1.7.4
        Tested by Lenovo:
          Tested:         2023-04-12
          Distribution:   sled 15.3
          Old version:    0.1.59
          Version[fwupd]: 1.5.8
        Vendor:           Lenovo
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        Lenovo System Firmware Version 1.60
        
        Important updates
        
        • Update includes a security fix.
        
        Problem fixes
        
        • Fixed an issue where NVMe passwords could not be erased properly.
        • Fixed an issue where abnormal shutdown may be logged when Thunderbolt device is attached or detached while system is in shutdown state.
```

I expect clear information that there is an available update but it is not possible to install it now because of this and that.

hughsie, 2023-05-10:
> with the latest version

Which latest version please?

closed 2023-05-24 by superm1

iyanmv, 2023-07-04:
I can confirm, now that I have received some new updates from Lenovo, that this works as expected.