fwupd/fwupd #4284: [drm] *ERROR* mstb 0000000057b5b857 port 1: DPCD read on addr 0x4b0 for 1 bytes NAKed

kdj0c, 2022-02-14:
**Describe the bug**

When fwupd is running, when I plug a monitor on my lenovo dock, I get the following kernel error on dmesg:
[drm] *ERROR* mstb 0000000057b5b857 port 1: DPCD read on addr 0x4b0 for 1 bytes NAKed

This is caused by the mst-synaptic plugin, which tries to read at 0x4b0 (REG_RC_CAP)

When looking at the code, https://github.com/fwupd/fwupd/blob/main/plugins/synaptics-mst/fu-synaptics-mst-device.c#L1299
It first read 1 byte at 0x4b0 and then check the vendor ID at 0x500 (REG_VENDOR_ID).

According to DP specification, 0x500 is the generic address for IEEE_OUI vendor identifier, and 0x4b0 is proprietary, and may not be implemented on all devices. (so may generate errors in kernel).

So I think the logic should be to first read at 0x500, and if the vendor is synaptic, then it is safe to read at 0x4b0.

The current logic is a bit weird because if the byte read at 0x4b0 (& 0x04), then it won't check the vendor id ?

**Steps to Reproduce**

Use a ThinkPad Thunderbolt 3 Dock Gen 2 , and connect anything on hdmi/display port.
(and I use the integrated Intel graphics of my laptop).

if you search on google for "DPCD read on addr 0x4b0", it is affecting a lot of people.

**Expected behavior**

Kernel should not complain on dmesg, when fwupd scan the devices.

**fwupd version information**

client version:	1.7.4
compile-time dependency versions
	gusb:	0.3.9

daemon version:	1.7.4

installed with dnf

**fwupd device information**

20TJS53A39
│
├─CometLake-H GT2 [UHD Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    05
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              07168636-0f3b-565c-8fe1-0f0a77d82cd8
│                         5fde5d20-db24-5f21-afdd-247c1bf1efa1
│                         6c3dbf6c-4e6f-5309-9954-c5ab7aca617e
│                         3777783a-3f83-56a5-95f4-533eb6a2bd19
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Core™ i7-10850H CPU @ 2.70GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000ec
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         a45b0522-5722-54bd-b802-86cd044262df
│                         7b9b6e8c-226c-5db6-86cb-ea3187578013
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.8
│     Minimum Version:    0.1.8
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     Update Error:       Cannot install update when the lid is closed
[... 269 lines cut ...]
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
└─UEFI Device Firmware:
      Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
      Summary:            UEFI ESRT device
      Current version:    22552
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      Update Error:       Cannot install update when the lid is closed
      GUIDs:              6255bb23-b2a8-469a-9161-bb9e1348ca23
                          8bb113aa-3089-57c1-9539-5c9c1b4470c6
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable

**Additional questions**

Fedora 35

hughsie, 2022-02-14:
>  [drm] ERROR mstb 0000000057b5b857 port 1: DPCD read on addr 0x4b0 for 1 bytes NAKed

It's pretty harmless tho, right? I'm using a ThinkPad Thunderbolt 3 Dock Gen 2 right now and it seems to work fine -- although I see the same error in dmesg.

> So I think the logic should be to first read at 0x500, and if the vendor is synaptic, then it is safe to read at 0x4b0.

Yes, that sounds sane. Would you be able to do a PR to make this change? Thanks!

hughsie, 2022-02-14:
@superm1 you might be able to remember some of the original logic for doing it this way perhaps?

kdj0c, 2022-02-14:
Yes the print is harmless, but it's still better to not have it ;)

I've looked a bit more at DP specification, and it says that even when reading proprietary section of 0x40C to 0x4FF, the device must ACK and return 0x00. So maybe the real fix is on Lenovo's firmware ?

I can make a PR, but I can't remove the read to 0x4b0 without potentially breaking some devices.

hughsie, 2022-02-14:
> the device must ACK and return 0x00

This is what I thought too, so I was surprised to see the error too. Maybe we need to open an issue (with a lot of explanation!) here https://github.com/fwupd/firmware-lenovo/issues so that @mrhpearson can open an internal issue.

kdj0c, 2022-02-15:
I've made a small program to read /dev/drm_dp_aux4, and reading to all addresses lead to this error. (even 0x00, 0x400 or 0x500).

So I think when nothing is connected to the MST port, the devices /dev/drm_dp_aux[456] should not be created by the kernel.
I will see if that make sense, and follow-up on dri-devel.

hughsie, 2022-02-15:
> and follow-up on dri-devel

Many thanks!

kdj0c, 2022-02-16:
I started a discussion on dri-devel:
https://lists.freedesktop.org/archives/dri-devel/2022-February/342423.html
As you can see, I also provided your proposal.

At this point, I'm closing this issue, because fwupd can't guess if the device is "readable" without attempting to read it.

Thanks,

closed 2022-02-16 by kdj0c

project-administrator, 2023-11-02:
I get same errors on a Dell laptop with a monitor connected through the Dell WD15 dock. Did not realize it's related to fwupd.

kikislater, 2024-07-17:
So what is the fix ? I have a WD15 with Thinkpad P14s gen 5 and it's the same !

hughsie, 2024-07-17:
> So what is the fix

I believe Jocelyn sent a kernel patch to fix this (basically, to downgrade the error). What kernel version are you using?

kikislater, 2024-07-17:
Ok thank you, I saw the issue in kernel report. I'm using 6.10