fwupd/fwupd #4285: crash (SIGSEGV) around 1-2 hours after start, with libusb-1.0 2:1.0.25-1

pabs3, 2022-02-15:
Forwarding [Debian fwupd#1005416](https://bugs.debian.org/1005416):

The upgrade of libusb-1.0 from 2:1.0.24-3 to 2:1.0.25-1 in Debian bookworm means fwupd (both 1.7.4-2 and 1.5.7-5) crashes with a SIGSEGV from one to two hours after it is started.

I am not sure if the fix should happen in fwupd or libusb so I have filed it against fwupd, since the tools in usbutils still work fine.

I have needrestart installed, which automatically restarts services including fwupd after libraries (eg libusb) used by them get upgraded.

I have included below the relevant apt upgrades, systemd journal entries and a sample gdb backtrace from the latest crash.

Here are the relevant apt upgrades since the issue started:

```
Start-Date: 2022-02-09  12:03:18
Commandline: /usr/bin/unattended-upgrade
Upgrade: libusb-1.0-0-dbgsym:amd64 (2:1.0.24-3, 2:1.0.25-1), libusb-1.0-0-dev:amd64 (2:1.0.24-3, 2:1.0.25-1), libusb-1.0-0:amd64 (2:1.0.24-3, 2:1.0.25-1)
End-Date: 2022-02-09  12:03:24
   
Start-Date: 2022-02-12  12:50:32
Requested-By: pabs (1000)
Install: libfwupdplugin5-dbgsym:amd64 (1.7.4-2), libfwupdplugin5:amd64 (1.7.4-2, automatic)
Upgrade: libfwupd2:amd64 (1.5.7-5, 1.7.4-2), fwupd-amd64-signed:amd64 (1.5.7+5, 1:1.2+3), fwupd-dbgsym:amd64 (1.5.7-5, 1.7.4-2), fwupd:amd64 (1.5.7-5, 1.7.4-2), libfwupd2-dbgsym:amd64 (1.5.7-5, 1.7.4-2)
Purge: libfwupdplugin1-dbgsym:amd64 (1.5.7-5), libfwupdplugin1:amd64 (1.5.7-5)
End-Date: 2022-02-12  12:50:45
```

Here are the fwupd related systemd journal entries:

```
$ sudo journalctl --no-hostname -u fwupd --since '2022-02-08 01:12:14' | vipe
Feb 08 01:12:14 systemd[1]: Starting Firmware update daemon...
Feb 08 01:12:14 systemd[1]: Started Firmware update daemon.
Feb 08 03:12:15 systemd[1]: fwupd.service: Deactivated successfully.
Feb 08 07:10:38 systemd[1]: Starting Firmware update daemon...
Feb 08 07:10:38 systemd[1]: Started Firmware update daemon.
Feb 08 09:10:39 systemd[1]: fwupd.service: Deactivated successfully.
Feb 08 14:14:48 systemd[1]: Starting Firmware update daemon...
Feb 08 14:14:49 systemd[1]: Started Firmware update daemon.
Feb 08 16:15:21 systemd[1]: fwupd.service: Deactivated successfully.
Feb 08 20:58:14 systemd[1]: Starting Firmware update daemon...
Feb 08 20:58:14 systemd[1]: Started Firmware update daemon.
Feb 08 22:58:16 systemd[1]: fwupd.service: Deactivated successfully.
Feb 08 22:58:16 systemd[1]: fwupd.service: Consumed 1.060s CPU time.
-- Boot <removed> --
Feb 09 10:04:07 systemd[1]: Starting Firmware update daemon...
Feb 09 10:04:09 systemd[1]: Started Firmware update daemon.
Feb 09 12:03:45 systemd[1]: Stopping Firmware update daemon...
Feb 09 12:03:45 systemd[1]: fwupd.service: Deactivated successfully.
Feb 09 12:03:45 systemd[1]: Stopped Firmware update daemon.
Feb 09 12:03:45 systemd[1]: Starting Firmware update daemon...
Feb 09 12:03:46 systemd[1]: Started Firmware update daemon.
Feb 09 14:03:46 fwupd[92678]: 06:03:46:0386 GLib-GObject         g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Feb 09 14:03:46 fwupd[92678]: 06:03:46:0386 GLib-GObject         g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Feb 09 14:03:46 fwupd[92678]: libusb: debug [libusb_unref_device] destroy device 1.2
Feb 09 14:03:46 systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Feb 09 14:03:46 systemd[1]: fwupd.service: Failed with result 'core-dump'.
Feb 09 23:34:34 systemd[1]: Starting Firmware update daemon...
Feb 09 23:34:34 systemd[1]: Started Firmware update daemon.
Feb 10 01:34:36 fwupd[398075]: 17:34:36:0406 GLib-GObject         g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Feb 10 01:34:36 fwupd[398075]: 17:34:36:0406 GLib-GObject         g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Feb 10 01:34:36 fwupd[398075]: libusb: debug [libusb_unref_device] destroy device 1.2
Feb 10 01:34:36 systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Feb 10 01:34:36 systemd[1]: fwupd.service: Failed with result 'core-dump'.
Feb 10 01:34:36 systemd[1]: fwupd.service: Consumed 1.057s CPU time.
Feb 10 08:57:46 systemd[1]: Starting Firmware update daemon...
Feb 10 08:57:47 systemd[1]: Started Firmware update daemon.
Feb 10 10:57:47 fwupd[624424]: 02:57:47:0336 GLib-GObject         g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Feb 10 10:57:47 fwupd[624424]: 02:57:47:0336 GLib-GObject         g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Feb 10 10:57:47 fwupd[624424]: libusb: debug [libusb_unref_device] destroy device 1.2
Feb 10 10:57:47 systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
[... 470 lines cut ...]
#22 glib_autoptr_clear_FuMainPrivate (_ptr=0x556f5036beb0) at ../src/fu-main.c:2022
  immediate_exit = 0
  timed_exit = 0
  options = {{long_name = 0x556f4efdf5f8 "timed-exit", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffcfba49474, description = 0x556f4efdf603 "Exit after a small delay", arg_description = 0x0}, {long_name = 0x556f4efdf61c "immediate-exit", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffcfba49470, description = 0x556f4efdfb20 "Exit after the engine has loaded", arg_description = 0x0}, {long_name = 0x0, short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x0, description = 0x0, arg_description = 0x0}}
  priv = 0x556f5036beb0
  error = 0x0
  argv0_file = 0x556f50360ec0
  context = 0x556f5036bb20
#23 glib_autoptr_cleanup_FuMainPrivate (_ptr=<synthetic pointer>) at ../src/fu-main.c:2022
  immediate_exit = 0
  timed_exit = 0
  options = {{long_name = 0x556f4efdf5f8 "timed-exit", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffcfba49474, description = 0x556f4efdf603 "Exit after a small delay", arg_description = 0x0}, {long_name = 0x556f4efdf61c "immediate-exit", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffcfba49470, description = 0x556f4efdfb20 "Exit after the engine has loaded", arg_description = 0x0}, {long_name = 0x0, short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x0, description = 0x0, arg_description = 0x0}}
  priv = 0x556f5036beb0
  error = 0x0
  argv0_file = 0x556f50360ec0
  context = 0x556f5036bb20
#24 main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:2055
  immediate_exit = 0
  timed_exit = 0
  options = {{long_name = 0x556f4efdf5f8 "timed-exit", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffcfba49474, description = 0x556f4efdf603 "Exit after a small delay", arg_description = 0x0}, {long_name = 0x556f4efdf61c "immediate-exit", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffcfba49470, description = 0x556f4efdfb20 "Exit after the engine has loaded", arg_description = 0x0}, {long_name = 0x0, short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x0, description = 0x0, arg_description = 0x0}}
  priv = 0x556f5036beb0
  error = 0x0
  argv0_file = 0x556f50360ec0
  context = 0x556f5036bb20
```

hughsie, 2022-02-15:
This is fixed by https://github.com/hughsie/libgusb/commit/24934619a2ad3467b98142dfab2039985afc970d and Debian needs to update libgusb to 0.3.9 or newer to prevent this happening.

closed 2022-02-15 by hughsie

pabs3, 2022-02-15:
Thanks for the info, I'll file a request about that.

superm1, 2022-02-15:
Fwiw I just uploaded 0.3.10 to unstable.