fwupd/fwupd #4290: `FuPluginTpm          failed to load eventlog: Failed to open file "/sys/kernel/security/tpm0/binary_bios_measurements": No such file or directory`

paulmenzel, 2022-02-16:
On the Dell Precision 3540/0M14W7 (system firmware 1.10.1) with Debian sid/unstable and Linux 5.16.7, fwupd looks for a non-existing file.

    $ fwupdmgr --version
    runtime   org.freedesktop.fwupd         1.7.5
    compile   org.freedesktop.gusb          0.3.10
    runtime   com.dell.libsmbios            2.4
    runtime   org.kernel                    5.16.0-1-amd64
    compile   org.freedesktop.fwupd         1.7.5
    runtime   org.freedesktop.gusb          0.3.10

    $ journalctl -u fwupd.service -b
    […]
    Feb 16 14:28:28 morley systemd[1]: Starting Firmware update daemon...
    Feb 16 14:28:29 morley fwupd[9846]: 13:28:29:0101 FuPluginTpm          failed to load eventlog: Failed to open file "/sys/kernel/security/tpm0/binary_bios_measurements": No such file or directory
    Feb 16 14:28:29 morley systemd[1]: Started Firmware update daemon.

hughsie, 2022-02-16:
@superm1 any idea why this file might not exist? @paulmenzel does /dev/tpm0 exist for you?

paulmenzel, 2022-02-16:
No,  the TPM device does not exist.

    $ ls /dev/tmp0
    ls: cannot access '/dev/tmp0': No such file or directory

    $ sudo dmesg | grep -i tpm
    [    0.830269] ima: No TPM chip found, activating TPM-bypass!

superm1, 2022-02-16:
TPM disabled in BIOS setup perhaps?

hughsie, 2022-02-16:
Did you expect the TPM device to exist?

paulmenzel, 2022-02-16:
> Did you expect the TPM device to exist?

No, I just saw the error in the log. (I do not have access to the device, but I guess the TPM is indeed disabled in the system firmware.)

superm1, 2022-02-16:
If you don't know whether it's enabled in BIOS setup, you may be able to use `/sys/class/firmware_attributes` to determine this.