fwupd/fwupd #4291: fwupd logs TPM related errors

paulmenzel, 2022-02-16:
On the Dell Latitude E7250, fwupd logs several errors.

```
○ fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static)
     Active: inactive (dead)
       Docs: https://fwupd.org/

Feb 15 20:20:05 ersatz fwupd[27270]: 19:20:05:0741 FuEngine             no GUIDs for device TPM [dfb1a4db3787a1799167fb68010ef732c71041f9]
Feb 15 20:20:07 ersatz fwupd[27270]: ERROR:sys:src/tss2-sys/api/Tss2_Sys_Execute.c:113:Tss2_Sys_ExecuteFinish() Unsupported device. The device is a TPM 1.2
Feb 15 20:20:07 ersatz fwupd[27270]: ERROR:esys:src/tss2-esys/api/Esys_Startup.c:216:Esys_Startup_Finish() Received a non-TPM Error
Feb 15 20:20:07 ersatz fwupd[27270]: ERROR:esys:src/tss2-esys/api/Esys_Startup.c:78:Esys_Startup() Esys Finish ErrorCode (0x00080001)
Feb 15 20:20:07 ersatz fwupd[27270]: 19:20:07:0250 FuEngine             failed to add device /sys/devices/pnp0/00:08/tpm/tpm0: failed to initialize TPM
Feb 15 20:20:07 ersatz systemd[1]: Started Firmware update daemon.
Feb 15 20:20:07 ersatz fwupd[27270]: 19:20:07:0282 FuPluginPciMei       ME family not supported for 0:10.0.56.3002
Feb 16 06:34:16 ersatz fwupd[27270]: 05:34:16:0729 FuPluginPciMei       ME family not supported for 0:10.0.56.3002
Feb 16 06:39:44 ersatz systemd[1]: fwupd.service: Deactivated successfully.
Feb 16 06:39:44 ersatz systemd[1]: fwupd.service: Consumed 1.192s CPU time.
```

Maybe one error message enough, if the TPM version is too low?

```
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.17.0-rc3-amd64
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10
```

superm1, 2022-02-16:
I'm sorry what message are you talking about?  The TSS2 messages come from the TSS2 library not fwupd, and they are all expected when you are using TPM 1.2 not TPM 2.0.

paulmenzel, 2022-02-16:
I would have though to only see fwupd messages in the output of fwupd service, and fwupd to suppress them or so.

commit 66e3d991 (2022-02-16): Don't show TSS2 error messages by default (Fixes: #4291)
closed 2022-02-16 by superm1