fwupd/fwupd #4294: mtd: no version reported

DylanVanAssche, 2022-02-16:
**Describe the bug**

The MTD plugin does not report any version which makes fwupd unhappy:

```
Pine64 PinePhonePro
│
└─spi0.0:
  │   Device ID:          c483d3f8b5d38f125bdc73f32d990dbea9c32768
  │   Description:        
  │   Updating your platform firmware may introduce new platform firmware features and bug fixes. Upgrading at your own risk!
  │   Update Error:       Device spi0.0 [c483d3f8b5d38f125bdc73f32d990dbea9c32768] has no firmware version <--- ISSUE
  │   GUID:               9101d2aa-f9e7-5b49-8720-16fc71fe9e69
  │   Device Flags:       • Internal device
  │                       • Needs a reboot after installation
  │                       • Updatable
  │ 
  └─Tow Boot SPI platform firmware:
        New version:      2021.10-004
        Branch:           community
        Summary:          Tow-Boot SPI platform firmware for the Pine64 PinePhone Pro.
        License:          MIT
        Size:             1.6 MB
        Urgency:          High
        Duration:         2 minutes
        Description:      
        Improvements:
        
        • 2021.10 U-boot
        • SPI support
        • Initial LVFS release
        
        This firmware is provided by LVFS community members and is not provided (or supported) by the original hardware vendor.
        
        Installing this update may also void any device warranty.
```

**Steps to Reproduce**

1. Get a device with an SPI flash
2. Ask info about an update: `fwupdmgr get-details $FILE`

**Expected behavior**

fwupd knows the version of the firmware on the SPI flash.

**fwupd version information**
1.7.5

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apk, from Alpine repos

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
Pine64 PinePhonePro
│
├─QUECTEL Mobile Broadband Module:
│     Device ID:          27730920c2119d128e431bbd20a8f225a2013d3a
│     Summary:            Quectel EG25-G modem
│     Current version:    0.5.7
│     Vendor:             QUALCOMM INCORPORATED (USB:0x2C7C)
│     Release Branch:     community
│     GUIDs:              db379a33-254f-5140-b37e-d36ae7e5c039
│                         d18f31f1-a3fa-55a2-b4ed-decfbc1e004d
│                         1a2996cb-f86e-5583-a464-e1b96e1c6ae9
│                         587bf468-6859-5522-93a7-6cce552a0aa3
│                         22ae45db-f68e-5c55-9c02-4557dca238ec
│     Device Flags:       • Updatable
│                         • System requires external power source
│   
├─SLD128:
│     Device ID:          46dad0f891e0142d68b298b33140ecd4555e7d75
│     Current version:    0x0202020000000000
│     Vendor:             EMMC:0x000088
│     GUIDs:              5c27e6e1-c17b-570e-ac9e-18999b45e157
│                         ad97b253-47fc-5e08-8860-4425f748c80e
│                         a5c6e4d5-600a-52f0-b7de-d07225340f04
│     Device Flags:       • Internal device
│                         • Updatable
│   
└─spi0.0:
      Device ID:          c483d3f8b5d38f125bdc73f32d990dbea9c32768
      Summary:            Memory Technology Device
      Vendor:             DMI:pine64
      GUID:               9101d2aa-f9e7-5b49-8720-16fc71fe9e69
      Device Flags:       • Internal device
                          • Updatable
                          • Needs a reboot after installation
```

**Additional questions**

- Operating system and version: postmarketOS edge, Pine64 PinePhone Pro
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2022-02-16:
> this device stores the version at offset 0x30

Right, agree. If this is the case then we can certainly quirk the MTD devices with the correct offset and size.

DylanVanAssche, 2022-02-16:
@hughsie @superm1 

I'm afraid there's not like a standard way to do it...
For U-Boot & Tow-Boot it can be done by just reading the content on the MTD device: https://stackoverflow.com/questions/5781738/getting-u-boots-version-from-userspace
This would require a regex, but allows not only to get the version but also if it is U-Boot or Tow-Boot (an U-Boot distribution https://tow-boot.org)

However, other platform firmware residing on MTD devices may not work.

hughsie, 2022-02-16:
And also a `sudo fwupdtool get-devices --verbose --plugins mtd` please.

DylanVanAssche, 2022-02-16:
> got a public link to an image I can read?

[2021.10-004.zip](https://github.com/fwupd/fwupd/files/8081972/2021.10-004.zip)

^^^ My metainfo file & Tow-Boot binary for SPI MTD devices.

hughsie, 2022-02-16:
> Tow-Boot binary for SPI MTD devices

So that would be `U-Boot TPL 2021.10 (Oct 04 2021 - 15:09:26)` and you'd expect fwupd to show a version of `2021.10` for that?

> 17:16:14:0703 FuHwids              smbios property Manufacturer=pine64

Bleh, fallback. What do you get from `sudo fwupdtool --verbose hwids` please.

DylanVanAssche, 2022-02-16:
> Tow-Boot binary for SPI MTD devices
> So that would be U-Boot TPL 2021.10 (Oct 04 2021 - 15:09:26) and you'd expect fwupd to show a version of 2021.10 for that?

Yes, `2021.10` is the version number of U-Boot, but Tow-Boot appends some more:

```
[dylan@dell-laptop 2021.10-004]$ strings Tow-Boot.spi.bin | grep '^Tow-Boot [[:digit:]]\{4\}\.[[:digit:]]\{2\}'
Tow-Boot 2021.10
Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)004-pre [variant: spi]
```
Where `004` is the Tow-Boot version. So something like `$UBOOT-$TOWBOOT` --> `2021.10-004`.

> Bleh, fallback. What do you get from sudo fwupdtool --verbose hwids please.

```
pine64-pinephonepro:~$ sudo fwupdtool --verbose hwids
[sudo] password for dylan: 
17:24:04:0657 FuDebug              Verbose debugging enabled (on console 1)
17:24:04:0659 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
17:24:04:0659 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
17:24:04:0660 FuHwids              smbios property Manufacturer=pine64
17:24:04:0660 FuHwids              smbios property EnclosureKind=8
17:24:04:0661 FuHwids              ignoring Family: offset bigger than size 6
17:24:04:0661 FuHwids              smbios property ProductName=Pine64 PinePhonePro
17:24:04:0661 FuHwids              ignoring ProductSku: offset bigger than size 6
17:24:04:0661 FuHwids              ignoring BiosVendor: offset bigger than size 0
17:24:04:0661 FuHwids              ignoring BiosVersion: offset bigger than size 0
17:24:04:0661 FuHwids              ignoring BiosMajorRelease: offset bigger than size 0
17:24:04:0661 FuHwids              ignoring BiosMinorRelease: offset bigger than size 0
17:24:04:0661 FuHwids              ignoring FirmwareMajorRelease: offset bigger than size 0
17:24:04:0662 FuHwids              ignoring FirmwareMinorRelease: offset bigger than size 0
17:24:04:0662 FuHwids              ignoring BaseboardManufacturer: offset bigger than size 0
17:24:04:0662 FuHwids              ignoring BaseboardProduct: offset bigger than size 0
17:24:04:0662 FuHwids              HardwareID-0 is not available, not available as 'Family' unknown
17:24:04:0662 FuHwids              HardwareID-1 is not available, not available as 'Family' unknown
17:24:04:0662 FuHwids              HardwareID-2 is not available, not available as 'BiosVendor' unknown
17:24:04:0662 FuHwids              HardwareID-3 is not available, not available as 'Family' unknown
17:24:04:0663 FuHwids              HardwareID-4 is not available, not available as 'Family' unknown
17:24:04:0663 FuHwids              HardwareID-5 is not available, not available as 'Family' unknown
17:24:04:0663 FuHwids              HardwareID-6 is not available, not available as 'ProductSku' unknown
17:24:04:0663 FuHwids              HardwareID-7 is not available, not available as 'ProductSku' unknown
17:24:04:0663 FuHwids              HardwareID-8 is not available, not available as 'BaseboardManufacturer' unknown
17:24:04:0663 FuHwids              HardwareID-10 is not available, not available as 'Family' unknown
17:24:04:0664 FuHwids              HardwareID-11 is not available, not available as 'Family' unknown
17:24:04:0664 FuHwids              HardwareID-13 is not available, not available as 'BaseboardManufacturer' unknown
Computer Information
--------------------
Manufacturer: pine64
ProductName: Pine64 PinePhonePro
EnclosureKind: 8

Hardware IDs
------------
not available as 'Family' unknown
not available as 'Family' unknown
not available as 'BiosVendor' unknown
not available as 'Family' unknown
not available as 'Family' unknown
not available as 'Family' unknown
not available as 'ProductSku' unknown
not available as 'ProductSku' unknown
not available as 'BaseboardManufacturer' unknown
{8900df57-13ac-5052-b4ea-af59e5377bc8}   <- Manufacturer + ProductName
not available as 'Family' unknown
not available as 'Family' unknown
{d868fc81-f35c-501b-95cc-0143f94baa7b}   <- Manufacturer + EnclosureKind
not available as 'BaseboardManufacturer' unknown
{adad873f-33d3-5477-bd5e-0870eca37444}   <- Manufacturer

```

hughsie, 2022-02-16:
> Tow-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)004-pre [variant: spi]

Hmm, that's not super easy to parse. Would `2021.10` do? Or could Tow-Book define a header somewhere `VERSION=2021.10.4\0` -- i.e. I'm not sure I want to get into regexes.

hughsie, 2022-03-31:
@DylanVanAssche I think when the Tow-Boot image contains a uSWID chunk somewhere we can use something like:

    [MTD\VENDOR_foo&PRODUCT_bar&NAME_baz]
    FirmwareGType = FuUswidFirmware

DylanVanAssche, 2022-03-31:
@hughsie Agreed! I just have to say, I haven't been able to look into this further. I hope I find time soon<sup>:tm: </sup>

DylanVanAssche, 2022-09-03:
@hughsie I generated some uSWID data with your tool: https://github.com/hughsie/python-uswid
This results into a binary with the data about Tow-Boot's release, but is there also a way to append it to the binary?

As far as I understand, we need to have this uSWID binary, appended to the Tow-Boot image, and then use `FuUswidFirmware`. The latter can extract the uSWID data from a firmware upgrade package (I think?). Can it also be used to get the same information from a Tow-Boot installation with uSWID on the MTD or a regular partition (3 possible locations: MTD, eMMC boot partitions or a regular partition with a specific UUID)

DylanVanAssche, 2022-09-04:
I managed to manually flash a Tow-Boot release with uswid data appended to it. Works fine for booting.
However, fwupd does not extract the version from it when using:

```
FirmwareGType = FuUswidFirmware
```

It reads the SPI flash though as it takes way longer at startup than without this quirk.
To confirm that the version is not properly set, I hardcoded a `fu_device_set_version(FU_DEVICE(self), "0.0.0");` in the MTD plugin. This makes the version appear as `0.0.0`. I tried flashing then but that doesn't do much:

```
pine64-pinephonepro:/home/dylan# /usr/libexec/fwupd/fwupd -v
16:09:41:0252 FuDebug              Verbose debugging enabled (on console 1)
16:09:41:0257 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
16:09:41:0257 FuCommon             reading /etc/fwupd/daemon.conf with 2081 bytes
16:09:41:0257 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
16:09:41:0258 FuConfig             using autodetected max archive size 18.4 EB
16:09:41:0261 FuRemoteList         loading /usr/share/fwupd/metainfo
16:09:41:0262 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
16:09:41:0263 XbSilo               file: 0ba02be1-d279-748b-343d-6eef4b27bd31, current:0ba02be1-d279-748b-343d-6eef4b27bd31, cached: (null)
16:09:41:0263 XbSilo               loading silo with file contents
16:09:41:0264 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
16:09:41:0265 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
16:09:41:0266 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/vendor-directory.conf.apk-new
16:09:41:0267 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
16:09:41:0269 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
16:09:41:0270 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/vendor.conf.apk-new
16:09:41:0270 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
16:09:41:0272 FuRemoteList         ordering lvfs-testing=lvfs+1
16:09:41:0272 FuRemoteList         ignoring unfound remote fwupd
16:09:41:0273 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
16:09:41:0275 Jcat                 loaded 1 certificates
16:09:41:0275 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:09:41:0275 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
16:09:41:0275 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
16:09:41:0276 Jcat                 loaded 1 certificates
16:09:41:0276 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:09:41:0277 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
16:09:41:0277 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
16:09:41:0277 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
16:09:41:0277 FuEngine             client certificate now exists: nothing to do
16:09:41:0277 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
16:09:41:0280 FuHistory            got schema version of 8
16:09:41:0282 FuPlugin             load(/usr/lib/fwupd-plugins-7/libfu_plugin_fastboot.so)
16:09:41:0302 FuPlugin             load(/usr/lib/fwupd-plugins-7/libfu_plugin_modem_manager.so)
16:09:41:0303 FuIdle               setting timeout to 7200s
16:09:41:0399 FuCommon             reading /proc/self/exe with 330472 bytes
16:09:41:0403 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
16:09:41:0404 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
16:09:41:0404 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
16:09:41:0405 XbSilo               file: 8d58fbdd-897a-8fc1-c9da-69c036644c9c, current:8d58fbdd-897a-8fc1-c9da-69c036644c9c, cached: (null)
16:09:41:0405 XbSilo               loading silo with file contents
16:09:41:0405 FuQuirks             invalid key names: CcgxDmcTriggerCode,CcgxFlashRowSize,CcgxFlashSize,CcgxImageKind,CorsairDeviceKind,CxaudioChipIdBase,CxaudioPatch1ValidAddr,CxaudioPatch2ValidAddr,DellDockBlobBuildOffset,DellDockBlobMajorOffset,DellDockBlobMinorOffset,DellDockBlobVersionOffset,DellDockBoardMin,DellDockHubVersionLowest,DellDockInstallDurationI2C,DellDockUnlockTarget,DellDockVersionLowest,DfuAltName,DfuForceTimeout,DfuForceVersion,ElantpI2cTargetAddress,ElantpIapPassword,ElantpIcPageCount,GenesysScalerCfiFlashId,GenesysScalerGpioEnableRegister,GenesysScalerGpioOutputRegister,GenesysScalerGpioValue,GenesysUsbhubReadRequest,GenesysUsbhubSwitchRequest,GenesysUsbhubWriteRequest,JabraMagic,LogitechHidppModelId,NordicHidBootloader,ParadeLspconAuxDeviceName,PciBcrAddr,RealtekMstDpAuxName,RealtekMstDrmCardKernelName,Rts54I2cSpeed,Rts54RegisterAddrLen,Rts54TargetAddr,SuperioAutoloadAction,SuperioControlPort,SuperioDataPort,SuperioGType,SuperioId,SuperioPort,SuperioTimeout,SynapticsMstDeviceKind,WacomI2cFlashBaseAddr,WacomI2cFlashBlockSize,WacomI2cFlashSize
16:09:41:0407 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
16:09:41:0407 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
16:09:41:0407 FuHwids              smbios property Manufacturer=PINE64
16:09:41:0407 FuHwids              smbios property EnclosureKind=3
[... cut ...]
	ID: 00 00 00 00 00 00 00 00
	Version: Unknown
	Voltage: Unknown
	External Clock: Unknown
	Max Speed: Unknown
	Current Speed: Unknown
	Status: Unpopulated
	Upgrade: None
	L1 Cache Handle: Not Provided
	L2 Cache Handle: Not Provided
	L3 Cache Handle: Not Provided
	Serial Number: Not Specified
	Asset Tag: Not Specified
	Part Number: Not Specified
	Characteristics: None

Handle 0x0005, DMI type 32, 11 bytes
System Boot Information
	Status: No errors detected

Handle 0x0006, DMI type 127, 4 bytes
End Of Table
```

hughsie, 2022-09-05:
> As far as I understand, we need to have this uSWID binary, appended to the Tow-Boot image

It doesn't have to be appended -- it can be anywhere in the image -- e.g. coreboot includes it as a file in the CBFS.

hughsie, 2022-09-05:
Can you grab me a:

    sudo fwupdtool get-devices --plugins mtd --verbose

with the `FirmwareGType = FuUswidFirmware` set please. Also if you can email me the image (i can keep private) I can try to reproduce myself with the mtdtest device.

DylanVanAssche, 2022-09-05:
@hughsie 

```
pine64-pinephonepro:/home/dylan# sudo fwupdtool get-devices --plugins mtd --verbose
16:55:23:0386 FuDebug              Verbose debugging enabled (on console 1)
16:55:23:0391 FuMain               locked /run/lock/fwupdtool
Loading…                 [-                                      ]16:55:23:0393 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
16:55:23:0393 FuCommon             reading /etc/fwupd/daemon.conf with 2081 bytes
16:55:23:0393 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
16:55:23:0393 FuConfig             using autodetected max archive size 18.4 EB
16:55:23:0394 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Loading…                 [ -                                     ]16:55:23:0397 FuRemoteList         loading /usr/share/fwupd/metainfo
16:55:23:0398 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
16:55:23:0398 XbSilo               file: 2e3e2745-08a9-ece0-51de-6d0c65d6b761, current:2e3e2745-08a9-ece0-51de-6d0c65d6b761, cached: (null)
16:55:23:0398 XbSilo               loading silo with file contents
16:55:23:0399 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
16:55:23:0400 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
16:55:23:0401 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/vendor-directory.conf.apk-new
16:55:23:0402 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
16:55:23:0404 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
16:55:23:0405 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/vendor.conf.apk-new
16:55:23:0405 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
16:55:23:0406 FuRemoteList         ordering lvfs-testing=lvfs+1
16:55:23:0407 FuRemoteList         ignoring unfound remote fwupd
Loading…                 [                                       ]16:55:23:0407 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
16:55:23:0408 Jcat                 loaded 1 certificates
16:55:23:0408 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:55:23:0408 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
16:55:23:0409 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
16:55:23:0410 Jcat                 loaded 1 certificates
16:55:23:0410 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:55:23:0410 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
16:55:23:0410 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
16:55:23:0410 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
16:55:23:0410 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]16:55:23:0410 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
16:55:23:0412 FuHistory            got schema version of 8
Loading…                 [*                                      ]16:55:23:0414 FuPlugin             load(/usr/lib/fwupd-plugins-7/libfu_plugin_fastboot.so)
16:55:23:0430 FuPlugin             load(/usr/lib/fwupd-plugins-7/libfu_plugin_modem_manager.so)
16:55:23:0511 FuCommon             reading /proc/self/exe with 461472 bytes
16:55:23:0515 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
16:55:23:0516 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
16:55:23:0516 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
16:55:23:0517 XbSilo               file: 933181a3-df7a-490d-6b80-50744708dbd8, current:013cfc73-68bc-9715-56ba-60beb1d3970d, cached: (null)
16:55:23:0596 FuQuirks             invalid key names: AndroidBootVersionProperty,CcgxDmcTriggerCode,CcgxFlashRowSize,CcgxFlashSize,CcgxImageKind,CorsairDeviceKind,CxaudioChipIdBase,CxaudioPatch1ValidAddr,CxaudioPatch2ValidAddr,DellDockBlobBuildOffset,DellDockBlobMajorOffset,DellDockBlobMinorOffset,DellDockBlobVersionOffset,DellDockBoardMin,DellDockHubVersionLowest,DellDockInstallDurationI2C,DellDockUnlockTarget,DellDockVersionLowest,DfuAltName,DfuForceTimeout,DfuForceVersion,ElantpI2cTargetAddress,ElantpIapPassword,ElantpIcPageCount,GenesysScalerCfiFlashId,GenesysScalerGpioEnableRegister,GenesysScalerGpioOutputRegister,GenesysScalerGpioValue,GenesysUsbhubReadRequest,GenesysUsbhubSwitchRequest,GenesysUsbhubWriteRequest,JabraMagic,LogitechHidppModelId,NordicHidBootloader,ParadeLspconAuxDeviceName,PciBcrAddr,plugin,RealtekMstDpAuxName,RealtekMstDrmCardKernelName,Rts54I2cSpeed,Rts54RegisterAddrLen,Rts54TargetAddr,SuperioAutoloadAction,SuperioControlPort,SuperioDataPort,SuperioGType,SuperioId,SuperioPort,SuperioTimeout,SynapticsMstDeviceKind,WacomI2cFlashBaseAddr,WacomI2cFlashBlockSize,WacomI2cFlashSize
Loading…                 [*                                      ]16:55:23:0598 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
16:55:23:0598 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
16:55:23:0598 FuHwids              smbios property Manufacturer=PINE64
16:55:23:0598 FuHwids              smbios property EnclosureKind=3
[... cut ...]
001856d0  1f 6c 70 6f 73 74 6d 61  72 6b 65 74 4f 53 18 20  |.lpostmarketOS. |
001856e0  70 70 6f 73 74 6d 61 72  6b 65 74 6f 73 2e 6f 72  |ppostmarketos.or|
001856f0  67 18 21 81 04 04 80 ff  ff ff ff ff ff ff ff ff  |g.!.............|
00185700  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

```

This is the Tow-Boot image with uSWID appended: 
[image.zip](https://github.com/fwupd/fwupd/files/9491369/image.zip)

It is all under a FOSS license (MIT) so no problem sharing anything ;)

hughsie, 2022-09-06:
Debugging now:

    $ sudo uswid --load /dev/mtd0 --verbose
    /dev/mtd0 has unknown extension, using uSWID
    Found USWID header at offset: 1594864
    Loaded:
    uSwidContainer([uSwidIdentity(acbd84ff-9898-4922-8ade-dd4bbe2e40ba,0,Tow-Boot.spi.bin,2021.10-005):
    uSwidEntity(Tow-Boot,tow-boot.org->TAG_CREATOR,LICENSOR,MAINTAINER,SOFTWARE_CREATOR)
    uSwidEntity(postmarketOS,postmarketos.org->DISTRIBUTOR)])

but:

    <firmware gtype="FuUswidFirmware">
      <flags>has-stored-size</flags>
      <offset>0x1855f0</offset>
      <size>0x107</size>
      <hdrver>0x1</hdrver>
    </firmware>

Debugging now.

closed 2022-09-07 by hughsie