fwupd/fwupd #4299: autopkgtests are failing in Ubuntu and Debian for 1.7.5

superm1, 2022-02-17:
It seems that the P2P client is failing.
```
Starting P2P client...
FuPlugin             add_security_attrs(acpi_dmar)
FuPlugin             add_security_attrs(acpi_facp)
FuPluginAcpiFacp     failed to load /sys/firmware/acpi/tables/FACP: Failed to open file "/sys/firmware/acpi/tables/FACP": No such file or directory
FuPlugin             add_security_attrs(iommu)
FuPlugin             add_security_attrs(linux_lockdown)
FuPlugin             add_security_attrs(linux_sleep)
FuPluginLinuxSleep   could not open /sys/power/mem_sleep: Error opening file /sys/power/mem_sleep: No such file or directory
FuPlugin             add_security_attrs(linux_swap)
FuPlugin             add_security_attrs(linux_tainted)
FuPlugin             add_security_attrs(pci_bcr)
FuPlugin             add_security_attrs(pci_mei)
FuPlugin             add_security_attrs(tpm)
FuPlugin             add_security_attrs(uefi_pk)
FAIL: fwupd/fwupdmgr-p2p.test (Child process exited with code 1)
FuMain               Called SetHints()
FuMain               got hint locale=C.UTF-8
[31m[1mWARNING:[0m The daemon has loaded 3rd party code and is no longer supported by the upstream developers!
FuMain               Called GetPlugins()
Mismatched daemon and client, use fwupdmgr instead
FuMain               client connection closed: Underlying GIOStream returned 0 bytes on an async read
```

Here is the full log.  https://ci.debian.net/data/autopkgtest/testing/amd64/f/fwupd/19232275/log.gz

superm1, 2022-02-17:
In the failure I specifically noticed:
```
Mismatched daemon and client, use fwupdmgr instead
```

Which makes me wonder if we just need to be using `--force` in the p2p test or detect `FWUPD_DBUS_SOCKET` and skip that test (someone would really have have to go out of their way to launch that way and probably doesn't need that infro).

superm1, 2022-02-17:
I also am surprised we aren't catching this in CI.

hughsie, 2022-02-17:
Not only that, but this:

> The daemon has loaded 3rd party code and is no longer supported by the upstream developers!

You able to get the daemon logs on a failing system?

superm1, 2022-02-17:
That's just the test plugin. Daemon logs are in the autopkgtest run. I'll try to see if this also is easy to reproduce in unstable in a container. I would think it is.

superm1, 2022-02-18:
@ycheng can you reproduce this?

superm1, 2022-02-18:
I tried in a jammy lxc container, it doesn't repro for me there, but maybe that's because I have real systemd in lxc but autopkgtest doesn't?

ycheng, 2022-02-18:
I can reproduce on focal host, the command to setup is:

$ sudo apt-get install autopkgtest lxc-utils lxc-template # once
$ sudo autopkgtest-build-lxc ubuntu jammy # once
$ cd fwupd
$ sudo LC_ALL=C autopkgtest -U --shell-fail -- lxc autopkgtest-jammy

If one test fails, it will give you a shell to debug.

superm1, 2022-02-18:
Backported the patches from the PR into 1.7.5-2 along with that container virtualization thing, but now it's failing because of that I think:
```
Executing: fwupd/fwupdmgr-p2p.test
Error: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

```
Executing: fwupd/fwupdmgr.test
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
Feb 18 21:11:11 ci-049-40846784 systemd[1]: Firmware update daemon was skipped because of a failed condition check (ConditionVirtualization=!container).
FAIL: fwupd/fwupdmgr.test (Child process exited with code 1)
```

https://ci.debian.net/data/autopkgtest/testing/amd64/f/fwupd/19347340/log.gz

@simondeziel I backported your fix since you initially filed it in debian I figured you wanted it sooner. I think your fix might have caused the failure though :/

superm1, 2022-02-19:
#4310 hopefully solves it.

closed 2022-02-19 by superm1

simondeziel, 2022-02-20:
> @simondeziel I backported your fix since you initially filed it in debian I figured you wanted it sooner. I think your fix might have caused the failure though :/

Oops, sorry for this and thanks for the autopkgtest workaround!

commit 3ea7f9e1 (2022-02-18): trivial: don't run systemd check when running as a socket

We don't want to be checking for a systemd daemon when we just told
you that it is a socketed daemon. (Fixes: #4299)
commit f301aabd (2022-02-18): trivial: don't run systemd check when running as a socket

We don't want to be checking for a systemd daemon when we just told
you that it is a socketed daemon. (Fixes: #4299)
PR #4310 (2022-02-19): trivial: allow fwupd.service to be run in containers for autopkgtest
Generally we don't want this - but for the autopkgtest case it's OK.
Fixes: #4299

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit cc19be59 (2022-02-19): trivial: allow fwupd.service to be run in containers for autopkgtest

Generally we don't want this - but for the autopkgtest case it's OK.
Fixes: #4299