fwupd/fwupd #4307: Dell WD19TB upgrade issue

d-xo, 2022-02-18:
**Describe the bug**

I am unable to update the thunderbolt controller in my wd19-tb from 0.43.0 to 0.60.0.

**Steps to Reproduce**

- run `fwupdmgr update`

**Expected behavior**

- the firmware on my dock is updated

**fwupd version information**

```shell
> fwupdmgr --version
client version:	1.7.2
compile-time dependency versions
	gusb:	0.3.7

daemon version:	1.7.2
```

Installed via nixpkgs (I use nixos)

**fwupd device information**

```shell
> fwupdmgr get-devices --filter=~internal
XPS 13 9370
│
└─Thunderbolt controller in Dell dock:
      Device ID:          1c6eb51e98f939edbaaa1f8ed253d5a0e29aff93
      Summary:            Thunderbolt controller
      Current version:    43.00
      Vendor:             Dell Inc. (THUNDERBOLT:0x00D4, TBT:0x00D4)
      Install Duration:   22 seconds
      Update State:       Failed
      Update Error:       Error reading from file: Input/output error
      Last modified:      2022-02-18 15:34
      GUIDs:              48711a14-0aaa-55b8-a17f-f7c494b318c6
                          08a8c886-2818-544c-b1a3-588eb07ae487
                          c94770ca-1773-592c-b20a-e87243bc7cd0
                          82cccc74-86f8-5bd6-b233-d342181ff40e
      Device Flags:       • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Install to parent device first
                          • Device stages updates
                          • Device is usable for the duration of the update
```

**Dock SKU**

- [ ] WD19 (Single-C)
- [x] WD19TB (Thunderbolt)
- [ ] WD19DC (Dual-C)

**Peripherals connected to the dock**

- 2x dell 1440p 27" screens
- an obins anne pro 2 keyboard
- a logitech usb web cam

**Verbose daemon logs**

https://gist.github.com/d-xo/78aaf9171b7fc5f98fbf3cf1e44b1d72

**Additional questions**

- Operating system and version:

nixos unstable

- Have you tried unplugging the dock or any peripherals from your machine?

yes

- Have you tried to power cycle the dock from the AC adapter?

yes

- Is this a regression?

not 100% sure, but iirc I have updated the firmware in this dock via fwupd before

superm1, 2022-02-18:
Any matching kernel logs at this time showing errors?

d-xo, 2022-02-18:
How would I get those?

superm1, 2022-02-18:
Look at your systemd journal if you use systemd, or dmesg.

d-xo, 2022-02-18:
I just reproduced the issue again and observed  both dmesg (`dmesg -w`) and the systemd journal (`journalctl -fk`) and no new messages were printed to either during the update attempt.

d-xo, 2022-02-18:
after connecting the dock to a different (non thunderbolt) usbc port, I was able to complete the firmware upgrade.

d-xo, 2022-02-18:
afaik it has two tb3 ports and one usb 3.1 type c:

[image]

superm1, 2022-02-18:
Did you have the device authorized? I noticed you were missing output from the Dell dock plugin in your gist.

If it's not authorized it won't prove properly (crag you might want to add a check for that attribute and inhibit updates...)

d-xo, 2022-02-18:
I have authorization disabled in the bios

superm1, 2022-02-18:
Hmm, then why wasn't the device tree showing up when connected to the thunderbolt port?  If you move it back over there, do you see more than the thunderbolt device?  Or still just that.

d-xo, 2022-02-20:
I'm guessing that one of my thunderbolt ports is bad? Or the two are different in some way...

Here is the output from the thunderbolt port where the update failed:

```
< fwupdmgr get-devices --filter=~internal
XPS 13 9370
│
└─Thunderbolt controller in Dell dock:
      Device ID:          1c6eb51e98f939edbaaa1f8ed253d5a0e29aff93
      Summary:            Thunderbolt controller
      Current version:    43.00
      Vendor:             Dell Inc. (THUNDERBOLT:0x00D4, TBT:0x00D4)
      Install Duration:   22 seconds
      GUIDs:              48711a14-0aaa-55b8-a17f-f7c494b318c6
                          08a8c886-2818-544c-b1a3-588eb07ae487
                          c94770ca-1773-592c-b20a-e87243bc7cd0
                          82cccc74-86f8-5bd6-b233-d342181ff40e
      Device Flags:       • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Install to parent device first
                          • Device stages updates
                          • Device is usable for the duration of the update
```

and here is the output when the dock is connected to the other thunderbolt port:

```
< fwupdmgr get-devices --filter=~internal
XPS 13 9370
│
└─WD19TB:
  │   Device ID:          275f399553a6ac1bbbe1e463791dfd12652efb6f
  │   Summary:            High performance dock
  │   Current version:    01.00.01.01
  │   Vendor:             Dell Inc. (USB:0x413C)
  │   Install Duration:   1 minute
  │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa
  │   Device Flags:       • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device can recover flash failures
  │                       • Device is usable for the duration of the update
  │
  ├─Package level of Dell dock:
  │     Device ID:        ff2ab17a0fda08f422afdb142e74368b20ed1cb7
  │     Summary:          A representation of dock update status
  │     Current version:  01.00.21.00
  │     Vendor:           Dell Inc. (USB:0x413C)
  │     Install Duration: 5 seconds
  │     GUID:             8ceeeffd-51b6-580c-9b75-69143227aff8
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device can recover flash failures
  │                       • Device is usable for the duration of the update
  │
  ├─RTS5413 in Dell dock:
  │     Device ID:        cadd482d8f2e6709ba268a8076df2c80f32088eb
  │     Summary:          USB 3.1 Generation 1 Hub
  │     Current version:  01.21
  │     Vendor:           Dell Inc. (USB:0x413C)
  │     Install Duration: 14 seconds
  │     GUIDs:            86fb40c0-8bf5-5a8b-a4ad-3156cf6bfaf4
  │                       b27d25f1-019d-5718-b41a-02ddaefe5577
  │                       ac5b774c-b49d-566b-9255-85f0f7f8a4ed
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device is usable for the duration of the update
[... 107 lines cut ...]
  │     Vendor:           Dell Inc. (USB:0x413C)
  │     Install Duration: 6 minutes
  │     GUID:             89fec0b6-6b76-5008-b82c-5e5c6c164007
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device is usable for the duration of the update
  │   
  └─Thunderbolt controller in Dell dock:
        Device ID:        1c6eb51e98f939edbaaa1f8ed253d5a0e29aff93
        Summary:          Thunderbolt controller
        Current version:  43.00
        Vendor:           Dell Inc. (THUNDERBOLT:0x00D4, TBT:0x00D4)
        Install Duration: 22 seconds
        GUIDs:            48711a14-0aaa-55b8-a17f-f7c494b318c6
                          08a8c886-2818-544c-b1a3-588eb07ae487
                          c94770ca-1773-592c-b20a-e87243bc7cd0
                          017a6b19-1f84-5be6-89f5-14ebd10a52de
        Device Flags:     • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Install to parent device first
                          • Device stages updates
                          • Device is usable for the duration of the update
```

d-xo, 2022-02-20:
Interestingly the thunderbolt controller fw version is once again 43.00? I ran the update with the dock plugged into the "good" thunderbolt port (i.e. where the full device tree is shown) and it seems to have succeeded, and is now showing 60.00 as it's version:

```
XPS 13 9370
│
└─WD19TB:
  │   Device ID:          275f399553a6ac1bbbe1e463791dfd12652efb6f
  │   Summary:            High performance dock
  │   Current version:    01.00.01.01
  │   Vendor:             Dell Inc. (USB:0x413C)
  │   Install Duration:   1 minute
  │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa
  │   Device Flags:       • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device can recover flash failures
  │                       • Device is usable for the duration of the update
  │ 
  ├─Package level of Dell dock:
  │     Device ID:        ff2ab17a0fda08f422afdb142e74368b20ed1cb7
  │     Summary:          A representation of dock update status
  │     Current version:  01.00.21.01
  │     Vendor:           Dell Inc. (USB:0x413C)
  │     Install Duration: 5 seconds
  │     GUID:             8ceeeffd-51b6-580c-9b75-69143227aff8
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device can recover flash failures
  │                       • Device is usable for the duration of the update
  │   
  ├─RTS5413 in Dell dock:
  │     Device ID:        cadd482d8f2e6709ba268a8076df2c80f32088eb
  │     Summary:          USB 3.1 Generation 1 Hub
  │     Current version:  01.21
  │     Vendor:           Dell Inc. (USB:0x413C)
  │     Install Duration: 14 seconds
  │     GUIDs:            86fb40c0-8bf5-5a8b-a4ad-3156cf6bfaf4
  │                       b27d25f1-019d-5718-b41a-02ddaefe5577
  │                       ac5b774c-b49d-566b-9255-85f0f7f8a4ed
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device is usable for the duration of the update
  │   
  ├─RTS5487 in Dell dock:
  │     Device ID:        c71b6a87faba01b30b2a765c65131f264b11cd74
  │     Summary:          USB 3.1 Generation 2 Hub
  │     Current version:  01.47
  │     Vendor:           Dell Inc. (USB:0x413C)
  │     Install Duration: 3 seconds
  │     GUIDs:            707c63d2-e597-5c40-84db-9b1bb4c48d96
  │                       acfcd89b-105d-55b9-b85b-08bf8508f38c
  │                       568ffa1e-a0db-5287-9ea3-872b60f7730b
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device is usable for the duration of the update
  │   
  ├─VMM5331 in Dell dock:
  │     Device ID:        68e822bd516207129427e07f5bb058ed2f801bc5
  │     Summary:          Multi Stream Transport controller
  │     Current version:  05.04.06
  │     Vendor:           Dell Inc. (USB:0x413C)
  │     Install Duration: 6 minutes
  │     GUID:             89fec0b6-6b76-5008-b82c-5e5c6c164007
  │     Device Flags:     • Updatable
  │                       • Supported on remote server
  │                       • Device stages updates
  │                       • Device is usable for the duration of the update
  │   
  └─Thunderbolt controller in Dell dock:
        Device ID:        1c6eb51e98f939edbaaa1f8ed253d5a0e29aff93
        Summary:          Thunderbolt controller
        Current version:  60.00
        Vendor:           Dell Inc. (THUNDERBOLT:0x00D4, TBT:0x00D4)
        Install Duration: 22 seconds
        GUIDs:            48711a14-0aaa-55b8-a17f-f7c494b318c6
                          08a8c886-2818-544c-b1a3-588eb07ae487
                          c94770ca-1773-592c-b20a-e87243bc7cd0
                          017a6b19-1f84-5be6-89f5-14ebd10a52de
        Device Flags:     • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Install to parent device first
                          • Device stages updates
                          • Device is usable for the duration of the update
```

superm1, 2022-02-21:
> I'm guessing that one of my thunderbolt ports is bad? Or the two are different in some way...

They're supposed to be identical.  I would suspect a hardware issue.
@CragW I think there is still a fwupd action though - you probably want to make sure the thunderbolt plugin doesn't expect a delayed activation if dell-dock didn't initialize.  I think that's the root of this original issue.

> Interestingly the thunderbolt controller fw version is once again 43.00? I ran the update with the dock plugged into the "good" thunderbolt port (i.e. where the full device tree is shown) and it seems to have succeeded, and is now showing 60.00 as it's version:

Did you unplug the dock from the wall by chance after your upgrade?

CragW, 2022-02-22:
> I think there is still a fwupd action though - you probably want to make sure the thunderbolt plugin doesn't expect a delayed activation if dell-dock didn't initialize. I think that's the root of this original issue.

That's good point, I will make a change to cancel the delayed activation if dell dock plugin not a runtime use.

closed 2022-02-24 by CragW

paco3346, 2022-04-04:
I hate to re-open this but it makes more sense than opening a new issue as it's the same thing.

I'd been unable to update my WD19TB via a thunderbolt port on any of my machines (all Arch) These include an XPS 7590, 9500, and a Thinkpad P15s G2. I tried updating via USB (via a USB-A to C adapter) and it worked perfectly first try. Unfortunately I don't have logs as I didn't think to preserve them before trying the update via USB.

dylanmtaylor, 2023-04-17:
> 

I figured out that you don't have to use a hub if you have a direct USB-C but NOT a Thunderbolt port on the host. Simply plugging the dock into the right side of my Precision 5570 where it is non-TB, I was able to upgrade successfully.

ponix-net, 2024-08-28:
If you don't have USB hub on your hand and your laptop has no non-thunderbolt port disabling Tunderbolt in BIOS for the time of upgrade also works.