fwupd/fwupd #4319: modem-manager: firmware branch not sometimes not reported

DylanVanAssche, 2022-02-22:
**Describe the bug**
In some cases the firmware branch is missing for devices using the Modem Manager plugin.
I think it is related to when the firmware branch is read. It is done in the setup phase, but devices can be detached and attached at any point in time. I noticed this after an upgrade, the version was updated, but the branch was not mentioned anymore when running `fwupdmgr get-devices`. Restarting the daemon or rebooting solves it of course since the daemon plugin will call the firmware branch AT command in the setup of the plugin. Same for when booting up, the device might be already booted, fwupd runs then, but the modem is slower so it gets added later on.

The AT command seems to timeout which may indicate that the modem isn't ready yet, should we retry this a few times with some delay? I saw in the plugin that this already happens for the version reporting.

```
pine64-pinephonepro:~$ sudo cat /var/log/fwupd 
[sudo] password for dylan: 
06:35:22:0940 FuDebug              Verbose debugging enabled (on console 0)
06:35:22:0978 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
06:35:22:0991 FuCommon             reading /etc/fwupd/daemon.conf with 1930 bytes
06:35:22:0991 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
06:35:22:0992 FuConfig             using autodetected max archive size 18.4 EB
06:35:23:0048 FuRemoteList         loading /usr/share/fwupd/metainfo
06:35:23:0136 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
06:35:23:0152 XbSilo               file: 01e2026d-49a5-4ce1-ffce-df820fb06d36, current:01e2026d-49a5-4ce1-ffce-df820fb06d36, cached: (null)
06:35:23:0152 XbSilo               loading silo with file contents
06:35:23:0173 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
06:35:23:0195 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
06:35:23:0239 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
06:35:23:0247 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
06:35:23:0275 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
06:35:23:0301 FuRemoteList         ordering lvfs-testing=lvfs+1
06:35:23:0301 FuRemoteList         ignoring unfound remote fwupd
06:35:23:0312 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
06:35:23:0312 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:35:23:0320 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
06:35:23:0332 Jcat                 loaded 1 certificates
06:35:23:0333 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
06:35:23:0333 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:35:23:0345 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
06:35:23:0347 Jcat                 loaded 1 certificates
06:35:23:0359 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
06:35:23:0360 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
06:35:23:0360 FuEngine             client certificate now exists: nothing to do
06:35:23:0361 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
06:35:23:0447 FuHistory            got schema version of 7
06:35:23:0448 FuIdle               setting timeout to 7200s
06:35:23:0449 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
06:35:23:0812 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
06:35:23:0818 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
06:35:23:0829 XbSilo               file: ddb43423-f8e3-a756-f96a-ced798f4dd95, current:ddb43423-f8e3-a756-f96a-ced798f4dd95, cached: (null)
06:35:23:0829 XbSilo               loading silo with file contents
06:35:23:0837 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
06:35:23:0838 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
06:35:23:0838 FuHwids              smbios property Manufacturer=pine64
06:35:23:0838 FuHwids              smbios property EnclosureKind=8
06:35:23:0838 FuHwids              ignoring Family: offset bigger than size 6
06:35:23:0838 FuHwids              smbios property ProductName=Pine64 PinePhonePro
06:35:23:0838 FuHwids              ignoring ProductSku: offset bigger than size 6
06:35:23:0838 FuHwids              ignoring BiosVendor: offset bigger than size 0
06:35:23:0839 FuHwids              ignoring BiosVersion: offset bigger than size 0
06:35:23:0839 FuHwids              ignoring BiosMajorRelease: offset bigger than size 0
06:35:23:0839 FuHwids              ignoring BiosMinorRelease: offset bigger than size 0
06:35:23:0839 FuHwids              ignoring FirmwareMajorRelease: offset bigger than size 0
06:35:23:0839 FuHwids              ignoring FirmwareMinorRelease: offset bigger than size 0
06:35:23:0839 FuHwids              ignoring BaseboardManufacturer: offset bigger than size 0
06:35:23:0839 FuHwids              ignoring BaseboardProduct: offset bigger than size 0
06:35:23:0840 FuHwids              HardwareID-0 is not available, not available as 'Family' unknown
06:35:23:0840 FuHwids              HardwareID-1 is not available, not available as 'Family' unknown
06:35:23:0840 FuHwids              HardwareID-2 is not available, not available as 'BiosVendor' unknown
06:35:23:0840 FuHwids              HardwareID-3 is not available, not available as 'Family' unknown
06:35:23:0840 FuHwids              HardwareID-4 is not available, not available as 'Family' unknown
06:35:23:0840 FuHwids              HardwareID-5 is not available, not available as 'Family' unknown
06:35:23:0840 FuHwids              HardwareID-6 is not available, not available as 'ProductSku' unknown
06:35:23:0840 FuHwids              HardwareID-7 is not available, not available as 'ProductSku' unknown
06:35:23:0840 FuHwids              HardwareID-8 is not available, not available as 'BaseboardManufacturer' unknown
06:35:23:0840 FuHwids              HardwareID-10 is not available, not available as 'Family' unknown
06:35:23:0840 FuHwids              HardwareID-11 is not available, not available as 'Family' unknown
[... 125 lines cut ...]
      GUIDs:              db379a33-254f-5140-b37e-d36ae7e5c039
                          d18f31f1-a3fa-55a2-b4ed-decfbc1e004d
                          1a2996cb-f86e-5583-a464-e1b96e1c6ae9
                          587bf468-6859-5522-93a7-6cce552a0aa3
                          22ae45db-f68e-5c55-9c02-4557dca238ec
      Device Flags:       • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Device supports switching to a different branch of firmware
    
pine64-pinephonepro:~$ 

# If you call the AT command manually, it just works of course ;)
pine64-pinephonepro:/home/dylan# echo "AT+GETFWBRANCH" | atinout - /dev/ttyUSB2 -

community

OK
```

**Additional questions**

- Operating system and version: postmarketOS edge, Pine64 PinePhone Pro
- Have you tried rebooting? Yes
- Is this a regression? Not sure, I haven't seen this at all when I was developing support for this.

hughsie, 2022-02-22:
> I noticed this after an upgrade, the version was updated, but the branch was not mentioned anymore

Hmm, we are using `klass_device->reload = fu_mm_device_setup` -- as reload is what's called when the update has completed. So we can implement the fu_mm_device_at_cmd() retry -- what's the GError->domain and GError->code when the modem is busy and we should retry? Thanks.

hughsie, 2022-02-22:
Also, which bit do we need to retry; the fu_device_locker_new_full() bit in fu_mm_device_setup_branch_at() or is doing the `fu_mm_device_at_cmd(self, self->branch_at)` bit over and over? And what kind of delay should we wait in between tries, e.g. 50ms, or 2s?

DylanVanAssche, 2022-02-22:
> what's the GError->domain and GError->code when the modem is busy and we should retry? Thanks.

The error is generated by https://github.com/fwupd/fwupd/blob/c77847b7ff18d81d6b0df8a7d657cd7c3fe1b610/plugins/modem-manager/fu-mm-device.c#L632 (write failure) and https://github.com/fwupd/fwupd/blob/c77847b7ff18d81d6b0df8a7d657cd7c3fe1b610/plugins/modem-manager/fu-mm-device.c#L648 (read failure).

I think this comes from https://github.com/fwupd/fwupd/blob/c77847b7ff18d81d6b0df8a7d657cd7c3fe1b610/libfwupdplugin/fu-io-channel.c#L336 

I guess we can retry on a write failure and a read failure.
I discussed this with the FOSS firmware dev for the modem and the difference is that the Pine64 PinePhone Pro has a way faster chip than the Pine64 PinePhone. Because of that, the modem was fully booted once fwupd did some probing for the branch.  I developed it on the PinePhone because I didn't have a Pro yet and the Pro's kernel is not fully stable yet.

> Also, which bit do we need to retry; the fu_device_locker_new_full() bit in fu_mm_device_setup_branch_at() or is doing the fu_mm_device_at_cmd(self, self->branch_at) bit over and over? And what kind of delay should we wait in between tries, e.g. 50ms, or 2s?

I would add retry to `fu_mm_device_at_cmd` method so it happens for all AT commands. 
ModemManager uses 3 retries with 3s between them.

hughsie, 2022-02-22:
Give me a few mins.

closed 2022-02-23 by hughsie