fwupd/fwupd #4329: Wacom Intous BT update completes sucessfully without updating

dsommers, 2022-02-23:
**Describe the bug**
When updating the firmware on a Wacom Intous BT (OTL-6100WL) via USB on RHEL8 using `fwupdmgr` it refuses to upgrade. When forcing an update, installed version numbers are still incorrect and `fwupdmgr` still wants to upgrade the firmware.

**Steps to Reproduce**
1. `fwupdmgr refresh`
2. `fwupdmgr get-updates`
3. `fwupdmgr update` which results in either `Firmware version formats were different, device was 'pair' and release is 'bcd'` or `Specified firmware is already installed '1.12'` (while reporting `Upgrade available for Intuos BT M from 1.7 to 1.11` a few lines higher up)
4. `fwupdmgr update --allow-branch-switch` (which seems to perform an update)
5. Goto 2.

**Expected behavior**
Firmware to be updated properly ;-)

**fwupd version information**
```shell
# fwupdmgr --version
client version:	1.5.9
compile-time dependency versions
	gusb:	0.3.0

daemon version:	1.5.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
It was part of the basic Red Hat Enterprise Linux 8 installation.

**fwupd device information**
```shell
# fwupdmgr get-devices --show-all-devices
20U9CTO1WW
│
├─Thunderbolt host controller:
│     Device ID:          4c3735d9756dc3b47bf51f258f74a5c1c524df71
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    46.00
│     Vendor:             Lenovo (THUNDERBOLT:0x0109|TBT:0x0109)
│     GUIDs:              32e299b7-9585-594c-833c-63ae28b14809 ← THUNDERBOLT\VEN_0109&DEV_1904&REV_00
│                         13a933fd-ab08-5b89-b95f-f0fbd1e736af ← THUNDERBOLT\VEN_0109&DEV_1904
│                         4117c99d-0660-5db6-bbfe-65ed036debe5 ← TBT-01091904-native
│                         bb3a7757-29b3-5950-9dba-71cc18c6fb86 ← TBT-01091904-native-controller0-0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Device stages updates
│   
├─CometLake-U GT2 [UHD Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              9a3b2da6-943a-5798-9478-9b01e85c49ee ← PCI\VEN_8086&DEV_9B41&SUBSYS_17AA22BE&REV_02
│                         29ee39cc-ad99-55d0-86c1-4f60814707bf ← PCI\VEN_8086&DEV_9B41&SUBSYS_17AA22BE
│                         9779d679-0f31-5dd5-b3c2-f6ff275a9ca9 ← PCI\VEN_8086&DEV_9B41&REV_02
│                         5589d313-88e5-54d3-8f4c-b2ee3882d4bc ← PCI\VEN_8086&DEV_9B41
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Embedded Controller:
│     Device ID:          0b3db288d2c013fcac57525c52aa6dc2c9934157
│     Current version:    0.1.13
│     Minimum Version:    0.1.13
│     Vendor:             DMI:LENOVO
│     GUIDs:              1ebe8ad3-3d40-4178-92ce-bbb3d0906332
│                         6dd2bd94-436c-5bb1-96a1-c8d0a87e4f09 ← UEFI\RES_{1EBE8AD3-3D40-4178-92CE-BBB3D0906332}
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
[... 123 lines cut ...]
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          1a71e0871d4136962e3280e153c5a48212ae4e43
      Current version:    24577
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      GUIDs:              798c0df8-2daa-4b93-a1c9-ded0bc7f231a
                          56f41448-c1e9-56d8-890b-945f7dac3428 ← UEFI\RES_{798C0DF8-2DAA-4B93-A1C9-DED0BC7F231A}
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```

**Additional questions**

- Operating system and version:  RHEL 8.5
- Have you tried rebooting? Yes
- Is this a regression? Dunno. Never tried upgrading this device before.

hughsie, 2022-02-23:
Hey @dsommers -- thanks for creating this issue. To make sure we've not already fixed this, can you try with the RHEL 8 packages here please? http://people.redhat.com/rhughes/packages/fwupd/el8/ if the output is the same then we can start digging.

dsommers, 2022-02-23:
@hughsie That version updated it perfectly.  Thanks!  I believe this can be closed now, as fixed in `fwupd-1.7.5-0.20220201.el8.x86_64`

hughsie, 2022-02-23:
Okay, thanks for testing. I think 1.7.something is on it's way to RHEL 8 officially, so fingers crossed for other people.

closed 2022-02-23 by hughsie

dsommers, 2022-02-23:
For reference, the relevant `fwupdmgr get-devices` section after the update.

```shell
├─Intuos BT M:
│ │   Device ID:          73ef80b60058b4f18549921520bfd94eaf18710a
│ │   Current version:    1.11
│ │   Vendor:             Wacom Co.,Ltd. (USB:0x056A)
│ │   Install Duration:   10 seconds
│ │   Update State:       Success
│ │   GUIDs:              0ae80a40-6d7b-52db-9a62-6fba5169e9e8 ← USB\VID_056A&PID_0378&REV_0111
│ │                       edf56833-dbe5-56ca-a651-734b01bb02ba ← USB\VID_056A&PID_0378
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │ 
│ ├─Intuos BT M [Legacy Bluetooth Module (ID6)]:
│ │     Device ID:        b759e23a21db52279e3473753aed535120ed18f8
│ │     Current version:  1.12
│ │     Vendor:           Wacom Co.,Ltd. (USB:0x056A)
│ │     Install Duration: 2 minutes
│ │     Update Error:     Device has been removed
│ │     GUID:             f316727e-0a60-5c63-a0b5-486c973bcaf9 ← USB\VID_056A&PID_0378-bluetooth-id6
│ │     Device Flags:     • Updatable
│ │   
│ └─Intuos BT M [Legacy Bluetooth Module]:
│       Device ID:        97785aabfcfc4eaa24eaed29d961a26989488cea
│       Current version:  1.12
│       Vendor:           Wacom Co.,Ltd. (USB:0x056A)
│       Install Duration: 30 seconds
│       Update Error:     Device has been removed
│       GUID:             230fb992-35c7-56b1-8236-7f5674a04153 ← USB\VID_056A&PID_0378-bluetooth
│       Device Flags:     • Updatable
│                         • Supported on remote server
```

The device works fine with both USB connected and BT connected after update.

What is different from before, there are now 2 entries with `Intuos BT M [Legacy Bluetooth Module]`, one tagged as `ID6`.  Otherwise all fine.

hughsie, 2022-02-23:
> one tagged as ID6

Ya, it's a legacy Wacom thing. I don't think we can avoid i unfortunately.