fwupd/fwupd #4345: Broken package after update

mrberry, 2022-02-28:
**Describe the bug**
After updating on Mint 20.2 fails with error:

E: /var/cache/apt/archives/libfwupdplugin5_1.7.7+f_amd64.deb: trying to overwrite '/usr/lib/x86_64-linux-gnu/libfwupdplugin.so.5.0.0', which is also in package libfwupdplugin4:amd64 1.7.4+f+3

Reading database ... 601521 files and directories currently installed.)
Preparing to unpack .../libfwupdplugin5_1.7.7+f_amd64.deb ...
Unpacking libfwupdplugin5:amd64 (1.7.7+f) ...
dpkg: error processing archive /var/cache/apt/archives/libfwupdplugin5_1.7.7+f_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libfwupdplugin.so.5.0.0', which is also in package libfwupdplugin4:amd64 1.7.4+f+3
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libfwupdplugin5_1.7.7+f_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
A package failed to install.  Trying to recover:
dpkg: dependency problems prevent configuration of fwupd:
 fwupd depends on libfwupdplugin5 (>= 1.7.7+f); however:
  Package libfwupdplugin5:amd64 is not installed.

dpkg: error processing package fwupd (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of fwupd-signed:
 fwupd-signed depends on fwupd (= 1.7.7+f); however:
  Package fwupd is not configured yet.

dpkg: error processing package fwupd-signed (--configure):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.31-0ubuntu9.2) ...
Errors were encountered while processing:
 fwupd

Unable to fix broken package.

**Steps to Reproduce**
Install update.

**Expected behavior**
Update without error.

**fwupd version information**

fwupd-gui/focal,focal,now 1 all [installed]
fwupd-signed/focal,now 1.62+f+1.7.7+f amd64 [installed]
fwupd-unsigned/focal,now 1.7.7+f amd64 [installed]
fwupd/focal,now 1.7.7+f amd64 [installed]
libfwupd2/focal,now 1.7.7+f amd64 [installed]
libfwupdplugin1/focal-updates,now 1.5.11-0ubuntu1~20.04.2 amd64 [installed]
libfwupdplugin4/focal,now 1.7.4+f+3 amd64 [installed]

```shell
fwupdmgr --version
```
fwupdmgr --version
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt (Synatptic package manager)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
fwupdmgr get-devices --show-all-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

**Additional questions**

- Operating system and version:

Mint 20.2

- Have you tried rebooting?

Yes

- Is this a regression?
Don't know,

superm1, 2022-02-28:
> 1.7.7

Woah - a time traveler! We just released 1.7.6.

Jesting aside, where did these packages come from?  `libfwupdplugin4` shouldn't contain the same libraries at `libfwupdplugin5`.  This looks like a downstream packaging problem.  I'm not aware of any of this happening in Debian or Ubuntu.

mrberry, 2022-02-28:
superm1, thank you for your reply. This came from the Mint update manager, has happened on  2 different installs. I am unable to fix because of the conflict. Do you have an suggestions how I might fix this?

superm1, 2022-02-28:
I think you need to figure out what APT repository is providing this package and report the problem to that package maintainer.
`apt policy libfwupdplugin5` and `apt policy libfwupdplugin4` might help.

mrberry, 2022-02-28:
Comes from Starlabs (the 2 laptops are this brand -https://starlabs.systems/ ). Would this be something that need to be taken up with them?

Thanks

# apt policy libfwupdplugin5
libfwupdplugin5:
  Installed: (none)
  Candidate: 1.7.7+f
  Version table:
     1.7.7+f 500
        500 http://ppa.launchpad.net/starlabs/ppa/ubuntu focal/main amd64 Packages

apt policy libfwupdplugin4
libfwupdplugin4:
  Installed: 1.7.4+f+3
  Candidate: 1.7.4+f+3
  Version table:
 *** 1.7.4+f+3 500
        500 http://ppa.launchpad.net/starlabs/ppa/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status

superm1, 2022-02-28:
Yup.

closed 2022-02-28 by superm1