fwupd/fwupd #4346: Dell XPS13-9370: Formerly updated TB controller no longer offered updates?

colemickens, 2022-02-28:
**Describe the question**

Years ago when I first got my XPS 13 9370, I noticed that fwupd was providing updates for both the Thunderbolt controller, as well as the "System Firmware" UEFI BIOS.

However, these days, `fwupd` seems to only know about my System Firmware:

```
cole@xeep ~
❯ sudo fwupdmgr update -v
(fwupdmgr:1744946): GLib-DEBUG: 12:51:05.529: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:1744946): GLib-GIO-DEBUG: 12:51:05.629: _g_io_module_get_default: Found default implementation dummy (GDummyProxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:1744946): Fwupd-DEBUG: 12:51:12.452: Emitting ::status-changed() [idle]
Devices with no available firmware updates:
 • USB2.0 Hub
 • USB3.0 Hub
 • KXG50ZNV1T02 NVMe TOSHIBA 1024GB
 • UEFI dbx
 • USB 2.0 BILLBOARD
Devices with the latest available firmware version:
 • System Firmware
(fwupdmgr:1744946): FuMain-DEBUG: 12:51:12.463: current version is 1.18.0: 1.18.0=same, 1.17.0=older, 1.16.0=older, 1.15.0=older, 1.14.3=older
```

Am I missing a kernel module, or some config, or did Dell remove support for some devices? I don't think its just my memory...

And thanks, `fwupd` is amazing great stuff, what an achievement!

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```shell
cole@xeep ~ 7s
❯ fwupdmgr --version
client version: 1.7.2
compile-time dependency versions
        gusb:   0.3.7

daemon version: 1.7.2
```

colemickens, 2022-02-28:
I'm trying to use bolt to force the TB controller on, but I'm not sure what to expect:

```
❯ sudo systemctl status bolt.service
● bolt.service - Thunderbolt system service
     Loaded: loaded (/etc/systemd/system/bolt.service; linked; vendor preset: enabled)
     Active: active (running) since Mon 2022-02-28 13:10:16 PST; 2min 5s ago
       Docs: man:boltd(8)
   Main PID: 1776267 (boltd)
     Status: "authmode: enabled, force-power: wait"
         IP: 0B in, 0B out
         IO: 0B read, 0B written
      Tasks: 3 (limit: 18801)
     Memory: 1.6M
        CPU: 29ms
     CGroup: /system.slice/bolt.service
             └─1776267 /nix/store/gpr6d4wzm36f4v2d5xy4yxc5r3qxv0qp-bolt-0.9.1/libexec/boltd

Feb 28 13:10:33 xeep boltd[1776267]: guard: got event for guard '3' (10)
Feb 28 13:10:33 xeep boltd[1776267]: power: guard '3' for 'boltctl' deactivated
Feb 28 13:10:33 xeep boltd[1776267]: power: shutdown scheduled (T-20.00s)
Feb 28 13:10:33 xeep boltd[1776267]: power: state changed: supported/wait
Feb 28 13:10:36 xeep boltd[1776267]: power: state changed: supported/on
Feb 28 13:10:36 xeep boltd[1776267]: power: guard '4' for 'boltctl' active
Feb 28 13:12:04 xeep boltd[1776267]: guard: got event for guard '4' (10)
Feb 28 13:12:04 xeep boltd[1776267]: power: guard '4' for 'boltctl' deactivated
Feb 28 13:12:04 xeep boltd[1776267]: power: shutdown scheduled (T-20.00s)
Feb 28 13:12:04 xeep boltd[1776267]: power: state changed: supported/wait

❯ sudo ./result/bin/boltctl power
acquired power guard (7)
```

and then in another terminal:

```
cole@xeep ~/code/nixcfg main* ⇡
❯ sudo fwupdmgr update --force
Devices with no available firmware updates:
 • USB2.0 Hub
 • USB3.0 Hub
 • KXG50ZNV1T02 NVMe TOSHIBA 1024GB
 • UEFI dbx
 • USB 2.0 BILLBOARD
Devices with the latest available firmware version:
 • System Firmware
```

superm1, 2022-02-28:
We intentionally made a change in upstream fwupd to remove the force power stuff which powers up the controller when nothing is plugged in because of some problems.  So this is expected behavior.

You can plug in a downstream device and it should power up.  When it's powered up you should see /sys/bus/thunderbolt get populated and fwupd should see it too.