fwupd/fwupd #4352: vli plugin probing prevents Targus USB3.0 SV Docking Station enumeration

djcampello, 2022-03-01:
The following entry in `plugins/vli/vli.quirk` prevents the enumeration of the `Targus USB3.0 SV Docking Station`:

```
GType = FuVliUsbhubDevice
[USB\VID_2109&PID_0812]
Plugin = vli
```

With the entry:

```
??USB3.0 Hub:
?     Device ID:          7c5b6dadf8126e3820c86cb56d9e1777084a98a8
?     Summary:            USB 2.x hub
?     Current version:    6.94
?     Vendor:             TARGUS (USB:0x2109)
?     Install Duration:   10 seconds
?     GUIDs:              039ee6a4-d140-5714-a480-a96b2e7cd638 ? USB\VID_2109&PID_0812&REV_0694
?                         7aaf809e-a3f1-5beb-8a28-09e44471534b ? USB\VID_2109&PID_0812
?                         08d0f0c8-f42d-55db-a971-b657bb7636ed ? USB\VID_2109&PID_0812&HUB_0002
?                         3e777b40-15fb-51cc-a3fe-ec2fcc5245f0 ? USB\VID_2109&PID_0812&SPI_C220&REV_0694
?                         cd3a7a5a-f5cf-546e-8f18-0a758327aa3e ? USB\VID_2109&PID_0812&SPI_C220
?     Device Flags:       ? Updatable
?                         ? Cryptographic hash verification is available
```

Without the problematic entry:

```
??Targus USB3.0 SV Docking Station:
?     Device ID:          0e399126586fa6c849149e941cb6c86b6f75377f
?     Current version:    30.39
?     Vendor:             DisplayLink (USB:0x17E9)
?     Serial Number:      10002159
?     GUIDs:              6a577e80-133e-54c8-bac9-771463481f5c ? USB\VID_17E9&PID_4306&REV_3039
?                         1d7be0f1-6d7e-5a29-a7d5-4290f6c4dc48 ? USB\VID_17E9&PID_4306
?     Device Flags:       ? Updatable
```

Expected kernel log entries (vli disabled):

```
2022-02-25T08:30:24.303143Z INFO kernel: [ 1569.605828] usb 2-3: new SuperSpeed USB device number 9 using xhci_hcd
2022-02-25T08:30:24.316104Z INFO kernel: [ 1569.618562] usb 2-3: New USB device found, idVendor=2109, idProduct=0812, bcdDevice= 6.94
2022-02-25T08:30:24.316155Z INFO kernel: [ 1569.618574] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
2022-02-25T08:30:24.316162Z INFO kernel: [ 1569.618581] usb 2-3: Product: USB3.0 Hub
2022-02-25T08:30:24.316168Z INFO kernel: [ 1569.618586] usb 2-3: Manufacturer: TARGUS
2022-02-25T08:30:24.318047Z INFO kernel: [ 1569.620516] hub 2-3:1.0: USB hub found
2022-02-25T08:30:24.318079Z INFO kernel: [ 1569.620685] hub 2-3:1.0: 4 ports detected
2022-02-25T08:30:24.357778Z ERR usb_bouncer[8937]: No active user session.
2022-02-25T08:30:24.467125Z INFO kernel: [ 1569.769401] usb 1-3: new high-speed USB device number 8 using xhci_hcd
2022-02-25T08:30:24.634125Z INFO kernel: [ 1569.936430] usb 1-3: New USB device found, idVendor=2109, idProduct=2812, bcdDevice= 6.90
2022-02-25T08:30:24.634178Z INFO kernel: [ 1569.936443] usb 1-3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
2022-02-25T08:30:24.634184Z INFO kernel: [ 1569.936450] usb 1-3: Product: USB2.0 Hub
2022-02-25T08:30:24.635037Z INFO kernel: [ 1569.937889] hub 1-3:1.0: USB hub found
2022-02-25T08:30:24.635071Z INFO kernel: [ 1569.938109] hub 1-3:1.0: 4 ports detected
2022-02-25T08:30:24.704735Z ERR usb_bouncer[8944]: No active user session.
2022-02-25T08:30:25.293140Z INFO kernel: [ 1570.595378] usb 2-3.1: new SuperSpeed USB device number 10 using xhci_hcd
2022-02-25T08:30:25.305062Z INFO kernel: [ 1570.607577] usb 2-3.1: New USB device found, idVendor=17e9, idProduct=4306, bcdDevice=30.39
2022-02-25T08:30:25.305110Z INFO kernel: [ 1570.607590] usb 2-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
2022-02-25T08:30:25.305119Z INFO kernel: [ 1570.607596] usb 2-3.1: Product: Targus USB3.0 SV Docking Station
2022-02-25T08:30:25.305124Z INFO kernel: [ 1570.607601] usb 2-3.1: Manufacturer: DisplayLink
2022-02-25T08:30:25.305129Z INFO kernel: [ 1570.607606] usb 2-3.1: SerialNumber: 10002159
```

With vli enabled (missing last usb device for the docking station):

```
2022-02-25T07:52:47.027159Z INFO kernel: [ 6793.169877] usb 2-3: new SuperSpeed USB device number 8 using xhci_hcd
2022-02-25T07:52:47.041073Z INFO kernel: [ 6793.183571] usb 2-3: New USB device found, idVendor=2109, idProduct=0812, bcdDevice= 6.94
2022-02-25T07:52:47.041133Z INFO kernel: [ 6793.183585] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
2022-02-25T07:52:47.041142Z INFO kernel: [ 6793.183592] usb 2-3: Product: USB3.0 Hub
2022-02-25T07:52:47.041148Z INFO kernel: [ 6793.183598] usb 2-3: Manufacturer: TARGUS
2022-02-25T07:52:47.043073Z INFO kernel: [ 6793.185602] hub 2-3:1.0: USB hub found
2022-02-25T07:52:47.043109Z INFO kernel: [ 6793.185722] hub 2-3:1.0: 4 ports detected
2022-02-25T07:52:47.076616Z ERR usb_bouncer[19877]: No active user session.

**_ 
2022-02-25T07:52:47.081724Z WARNING fwupd[3278]: 07:52:47:0081 FuDevice             already set GType to FuVliUsbhubDevice, ignoring FuVliUsbhubDevice
2022-02-25T07:52:47.083265Z WARNING fwupd[3278]: 07:52:47:0083 FuDevice             using 7c5b6dadf8126e3820c86cb56d9e1777084a98a8 for usb:02:00:03
2022-02-25T07:52:47.089559Z WARNING fwupd[3278]: 07:52:47:0089 FuPlugin             emit added from vli: 7c5b6dadf8126e3820c86cb56d9e1777084a98a8
2022-02-25T07:52:47.090278Z WARNING fwupd[3278]: 07:52:47:0090 FuPlugin             fu_plugin_device_registered(test)
2022-02-25T07:52:47.090278Z WARNING fwupd[3278]: 07:52:47:0090 FuPlugin             fu_plugin_device_registered(pci_bcr)
2022-02-25T07:52:47.090301Z WARNING fwupd[3278]: 07:52:47:0090 FuPlugin             fu_plugin_device_registered(dell_dock)
2022-02-25T07:52:47.090428Z WARNING fwupd[3278]: 07:52:47:0090 FuPlugin             fu_plugin_device_registered(msr)
2022-02-25T07:52:47.090518Z WARNING fwupd[3278]: 07:52:47:0090 FuDeviceList         ::added 7c5b6dadf8126e3820c86cb56d9e1777084a98a8
_**

2022-02-25T07:52:47.201122Z INFO kernel: [ 6793.343583] usb 1-3: new high-speed USB device number 9 using xhci_hcd
2022-02-25T07:52:47.367057Z INFO kernel: [ 6793.509854] usb 1-3: New USB device found, idVendor=2109, idProduct=2812, bcdDevice= 6.90
2022-02-25T07:52:47.367110Z INFO kernel: [ 6793.509868] usb 1-3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
2022-02-25T07:52:47.367118Z INFO kernel: [ 6793.509875] usb 1-3: Product: USB2.0 Hub
2022-02-25T07:52:47.368074Z INFO kernel: [ 6793.511336] hub 1-3:1.0: USB hub found
2022-02-25T07:52:47.368110Z INFO kernel: [ 6793.511526] hub 1-3:1.0: 4 ports detected
```

hughsie, 2022-03-01:
Maybe all of vli.quirk should go into vli-noinst.quirk and we only move over devices as-and-when they work? If we've had two problems in as many weeks then it's somewhat suggesting it's not safe to just assume all firmware is safe to probe...

djcampello, 2022-03-02:
I think that is a good alternative. We will lose being able to list some devices but since there is no available firmware to update, that might not be that bad.

superm1, 2022-03-02:
I think that's pragmatic for now.  If VLI can come back with some code changes to the plugin that allow more safely probing everything on the market then it can be restored.

closed 2022-03-02 by hughsie