fwupd/fwupd #4366: fu-self-test fails on cros_sdk because of missing /etc/os-release in sysroot

djcampello, 2022-03-08:
```
Bail out! ERROR:../fwupd-9999/src/fu-self-test.c:1913:fu_engine_multiple_rels_func: assertion failed (error == NULL): No os-release found (FwupdError, 5)
stderr:
**
ERROR:../fwupd-9999/src/fu-self-test.c:1913:fu_engine_multiple_rels_func: assertion failed (error == NULL): No os-release found (FwupdError, 5)
```

I was wondering can we fix the unit test in someway that it does not fail if no os-release is found... Error is coming from: https://github.com/fwupd/fwupd/blob/4ce9972ce54d3845d82170a83f848a9b65ddbabd/libfwupd/fwupd-common.c#L184

hughsie, 2022-03-08:
I thought the plan was to ship a /etc/os-release in Chrome OS?

djcampello, 2022-03-08:
It is the plan, but unit tests are run before the file is generated. The file is only added to the final image and is not present in the chroot when unit tests are run

closed 2022-03-09 by djcampello