fwupd/fwupd #4374: X1-Nano cannot update because no space left in the device?

jbuencuerpo, 2022-03-10:
**Describe the bug**
A clear and concise description of what the bug is.
I am trying to update :
**Steps to Reproduce**
Steps to reproduce the behavior.
update g
**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

hughsie, 2022-03-11:
I'm not completely sure how to debug this. For a start, can you attach the output of `du -h /sys/firmware/efi/efivars/*` please?

jbuencuerpo, 2022-03-11:
Hi, here it is
```shell
0	/sys/firmware/efi/efivars/AbtStatus-a0b1889e-00eb-445b-8ca9-e91ce43c907d
0	/sys/firmware/efi/efivars/Boot0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0002-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0010-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0011-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0012-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0013-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0014-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0015-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0016-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0017-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0018-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0019-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot001A-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot001B-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot001C-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot001D-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot001E-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot001F-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0020-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0021-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0022-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0023-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0024-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0025-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/Boot0026-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/BootCurrent-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/BootMediaInfo-5bd6b672-b6ea-4d6a-b590-18a932b78794
0	/sys/firmware/efi/efivars/BootOptionSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/BootOrder-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/BootOrderDefault-0b7646a4-6b44-4332-8588-c8998117f2ef
0	/sys/firmware/efi/efivars/BootState-60b5e939-0fcf-4227-ba83-6bbed45bc0e3
0	/sys/firmware/efi/efivars/BRDS-42780dd5-9a7d-404c-80e4-7f7094360394
0	/sys/firmware/efi/efivars/CapsuleLongModeBuffer-711c703f-c285-4b10-a3b0-36ecbd3c8be2
0	/sys/firmware/efi/efivars/certdb-d9bee56e-75dc-49d9-b4d7-b534210f637a
0	/sys/firmware/efi/efivars/certdbv-d9bee56e-75dc-49d9-b4d7-b534210f637a
0	/sys/firmware/efi/efivars/ConIn-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/ConInDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/ConOut-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/ConOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/CpuSetup-b08f97ff-e6e8-4193-a997-5e9e9b0adb32
0	/sys/firmware/efi/efivars/CpuSetupVolatileData-b08f97ff-e6e8-4193-a997-5e9e9b0adb32
0	/sys/firmware/efi/efivars/CurrentPolicy-77fa9abd-0359-4d32-bd60-28f4e78f784b
0	/sys/firmware/efi/efivars/db-d719b2cb-3d3a-4596-a3bc-dad00e67656f
0	/sys/firmware/efi/efivars/dbx-d719b2cb-3d3a-4596-a3bc-dad00e67656f
0	/sys/firmware/efi/efivars/DIAGSPLSHSCRN-a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380
0	/sys/firmware/efi/efivars/dump-type0-10-1-1643810461-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1643962698-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1644088398-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1644148502-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1644191447-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1644217484-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1644219025-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1644219193-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1644394141-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1644430513-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1644609440-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-1-1644689320-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-3-1643879743-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-3-1644090394-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-3-1644307738-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-3-1644565504-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-10-3-1644750027-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-11-1-1643810461-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-11-1-1643962698-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-11-1-1644088398-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
0	/sys/firmware/efi/efivars/dump-type0-11-1-1644148502-C-cfc8fc79-be2e-4ddc-97f0-9f98bfe298a0
[... cut ...]
0	/sys/firmware/efi/efivars/UnlockIDCopy-eaec226f-c9a3-477a-a826-ddc716cdc0e3
0	/sys/firmware/efi/efivars/VarErrorFlag-04b37fe8-f6ae-480b-bdd5-37d98c5e89aa
0	/sys/firmware/efi/efivars/VendorKeys-8be4df61-93ca-11d2-aa0d-00e098032b8c
0	/sys/firmware/efi/efivars/WAND-92daaf2f-c02b-455b-b2ec-f5a3594f4aea
0	/sys/firmware/efi/efivars/WGDS-92daaf2f-c02b-455b-b2ec-f5a3594f4aea
0	/sys/firmware/efi/efivars/WRDD-92daaf2f-c02b-455b-b2ec-f5a3594f4aea
0	/sys/firmware/efi/efivars/WRDS-92daaf2f-c02b-455b-b2ec-f5a3594f4aea
```

hughsie, 2022-03-11:
That looks like systemd is filling your NVRAM with logs -- e.g. see https://unix.stackexchange.com/questions/689158/disabling-writing-dump-files-to-efivars

I suspect deleting `/sys/firmware/efi/efivars/dump-type0*` will fix this, but you should probably find the root cause too.

jbuencuerpo, 2022-03-13:
Hi, thanks. I removed the dump-type0 files and I was able to upgrade the firmware. I suspect the issue is between TLP, acpi-call-dkms and a newer kernel version as here,

[https://gist.github.com/roadkell/9e98db6656e28fbbf1bf51082040f67](https://gist.github.com/roadkell/9e98db6656e28fbbf1bf51082040f67f)

I have removed those files, and I will report it on the distro so they can upgrade to the latest acpi_call.
I am closing this issue.
Thanks,

closed 2022-03-13 by jbuencuerpo