fwupd/fwupd #4377: Failing self-test 5/33 when pushing

memily, 2022-03-14:
**Describe the question**
We're wondering why we're failing self-test 5/33 when pushing a branch.

**fwupd version information**
Fedora 35, latest fwupd pull, no meson configs. We've attached both the terminal output and meson-logs testlog file.
Terminal output: [termOutput.txt](https://github.com/fwupd/fwupd/files/8243750/termOutput.txt)
meson-logs testlog: [testlog.txt](https://github.com/fwupd/fwupd/files/8243752/testlog.txt)

No rush! Thanks.

hughsie, 2022-03-14:
Is fwupd installed on the test system? does `fwupdmgr get-devices` work?

I think you can ignore it for a pull request, if that helps.

memily, 2022-03-14:
>Is fwupd installed on the test system? does fwupdmgr get-devices work?

Yes it is installed on the test machine, and yes running fwupd/fwupdmgr get-devices works

>I think you can ignore it for a pull request, if that helps.

Should I force push the branch?

memily, 2022-03-15:
>You shouldn't have to - maybe something like git push --set-upstream origin wip/memily/branchfoo 

I tried this initially and failed, we also tried to force push but failing 5/33 seems to be the main block to a successful push. When we setup fwupd we allowed the repo safety features to be installed that check for CI and ninja tests if that helps.

hughsie, 2022-03-15:
> setup fwupd we allowed the repo safety features to be installed

Maybe redo contrib/setup and say no this time? Does that work?

superm1, 2022-03-15:
`pre-commit uninstall -t pre-push` should probably work

memily, 2022-03-17:
>Maybe redo contrib/setup and say no this time? Does that work? 

There but for the grace of God go I... crashing fedora 35 yet again. 

>pre-commit uninstall -t pre-push should probably work 

After reinstalling fedora 35 OS and running the above command before pushing, success!

Reference PR: #4399

closed 2022-03-17 by memily